espressif/esp-idf #4478: error : invalid air mode: 255 while trying HFP API

Tarun-Narain, 2019-12-12:
i was working on HFP, whenever a call is recieved(smartphone) the following error comes up after the function `esp_hf_client_outgoing_data_ready()` is called: 
`bt_hf_client_sco_out_data invalid air mode: 255`

i registered the HF data callbacks as :
 ```
esp_hf_client_register_data_callback(esp_hf_client_incoming_data_cb,
                                                 esp_hf_client_outgoing_data_cb);
```
The AVRCP and A2DP are working perfectly, i can also see incoming calls and network state.
Code reference : a2dp_sink example Compiled using ESP-IDF 4.0
Thankful for the support. Help!

AbnerFederer, 2019-12-13:
Hi @Tarun-Narain,

Where do you place the `esp_hf_client_outgoing_data_ready();` API?

It shouldn't be placed in `esp_hf_client_incoming_data_cb`, please comment it out or place it just after the outgoing call data is ready, to test the incoming calls.

Thanks
Abner

Tarun-Narain, 2019-12-13:
Hi @AbnerFederer, 
i've commented the line in `esp_hf_client_incoming_data_cb` but the error still exists, i call the `esp_hf_client_outgoing_data_ready();` in 
```
void bt_app_hf_event(esp_hf_client_cb_event_t event, esp_hf_client_cb_param_t *param)
{
    switch (event) {

 case ESP_HF_CLIENT_CIND_CALL_EVT:
        ESP_LOGE(BT_AV_TAG, "--Call indicator %s %d",
                 c_call_str[param->call.status],param->call.status);
				 if(param->call.status==1)// 1=>call in progress
					 esp_hf_client_outgoing_data_ready();
        break;
}
```

AbnerFederer, 2019-12-13:
Hi @Tarun-Narain 

It should be used in your application where outgoing data is ready for each SCO packet.
For example, when i2s have saved enough data for the outgoing SCO packet.

Thanks
Abner

Tarun-Narain, 2019-12-13:
Thanks @alexey-morozov, 
Can you tell me why there is noise when i feed the audio To `i2s_write` or `write_ringbuf` functions?

closed 2019-12-13 by Tarun-Narain

AbnerFederer, 2019-12-13:
Hi @Tarun-Narain 

I have tested the HFP with A2DP coexist the incoming data I think is good enough.
Maybe you can control the frequency for `write_ringbuf` or `i2s_write`.
I will continue to dive into HFP and A2DP about SCO issues.
I will merge optimizations if any updates.

Thanks
Abner

Tarun-Narain, 2019-12-13:
Thanks for solving the issue within a day. 😃