espressif/esp-idf #4479: Release 3.3.1 decreased WiFi stability

vonnieda, 2019-12-12:
## Environment

- Development Kit:  none
- Module or chip used: ESP32-WROVER-B
- IDF version: v3.3.1
- Build System: Make
- Compiler version: 1.22.0-80-g6c4433a
- Operating System: macOS
- Power Supply: USB

## Problem Description

I'm seeing much more frequent WiFi disconnects and dropped packets since upgrading from 3.3.0 to 3.3.1 this morning. This is in a mature product that relies on WiFi.

Of note, I am seeing the following message constantly where I did not see it on v3.3.0.

```
I (458817) wifi: bcn_timout,ap_probe_send_start
I (512937) wifi: bcn_timout,ap_probe_send_start
I (528187) wifi: bcn_timout,ap_probe_send_start
I (695617) wifi: bcn_timout,ap_probe_send_start
I (705887) wifi: bcn_timout,ap_probe_send_start
I (768157) wifi: bcn_timout,ap_probe_send_start
I (778807) wifi: bcn_timout,ap_probe_send_start
I (837237) wifi: bcn_timout,ap_probe_send_start
I (862507) wifi: bcn_timout,ap_probe_send_start
```

Nothing else has changed between the upgrade from v3.3.0 to v3.3.1. The same device is in the same position on my desk, same distance from AP, etc. I just updated to 3.3.1 and re-flashed my firmware.

I realize this is somewhat vague. If you can recommend a specific WiFi connection quality test I can perform I'd be happy to compare 3.3.0 and 3.3.1.

Thanks,
Jason

vonnieda, 2019-12-12:
Some additional WiFi debug information that might be useful:

```
I (3765) wifi: new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
I (4535) wifi: state: init -> auth (b0)
I (4545) wifi: state: auth -> assoc (0)
I (4565) wifi: state: assoc -> run (10)
I (4585) wifi: connected with XXXXXXXXX, aid = 9, channel 1, BW20, bssid = fc:ec:da:xx:xx:xx
I (4585) wifi: security type: 3, phy: bgn, rssi: -51
I (4605) wifi: pm start, type: 1
I (4605) wifi: AP's beacon interval = 102400 us, DTIM period = 1
```

vonnieda, 2019-12-12:
Okay, I seem to have tracked this down to power saving. I have been using `esp_wifi_set_ps(WIFI_PS_MIN_MODEM)`, if I change to `esp_wifi_set_ps(WIFI_PS_NONE)` the problem goes away and the connection remains stable. I no longer get the `wifi: bcn_timout,ap_probe_send_start` messages and I don't lose my connection.

Additionally, I implemented a ping and here are results with power saving on:

```
ping www.google.com 500
60 bytes from 172.217.4.228: icmp_seq=0 time=72 ms
60 bytes from 172.217.4.228: icmp_seq=1 time=20 ms
60 bytes from 172.217.4.228: icmp_seq=2 time=24 ms
60 bytes from 172.217.4.228: icmp_seq=3 time=27 ms
60 bytes from 172.217.4.228: icmp_seq=4 time=15 ms
Request timeout for icmp_seq 5
Request timeout for icmp_seq 6
I (42784) wifi: bcn_timout,ap_probe_send_start
60 bytes from 172.217.4.228: icmp_seq=7 time=674 ms
60 bytes from 172.217.4.228: icmp_seq=8 time=18 ms
60 bytes from 172.217.4.228: icmp_seq=9 time=14 ms
60 bytes from 172.217.4.228: icmp_seq=10 time=18 ms
60 bytes from 172.217.4.228: icmp_seq=11 time=14 ms
60 bytes from 172.217.4.228: icmp_seq=12 time=9 ms
60 bytes from 172.217.4.228: icmp_seq=13 time=32 ms
60 bytes from 172.217.4.228: icmp_seq=14 time=24 ms
60 bytes from 172.217.4.228: icmp_seq=15 time=19 ms
60 bytes from 172.217.4.228: icmp_seq=16 time=72 ms
60 bytes from 172.217.4.228: icmp_seq=17 time=14 ms
60 bytes from 172.217.4.228: icmp_seq=18 time=14 ms
60 bytes from 172.217.4.228: icmp_seq=19 time=37 ms
60 bytes from 172.217.4.228: icmp_seq=20 time=23 ms
60 bytes from 172.217.4.228: icmp_seq=21 time=26 ms
60 bytes from 172.217.4.228: icmp_seq=22 time=24 ms
60 bytes from 172.217.4.228: icmp_seq=23 time=21 ms
60 bytes from 172.217.4.228: icmp_seq=24 time=64 ms
60 bytes from 172.217.4.228: icmp_seq=25 time=39 ms
60 bytes from 172.217.4.228: icmp_seq=26 time=115 ms
60 bytes from 172.217.4.228: icmp_seq=27 time=20 ms
60 bytes from 172.217.4.228: icmp_seq=28 time=23 ms
60 bytes from 172.217.4.228: icmp_seq=29 time=216 ms
60 bytes from 172.217.4.228: icmp_seq=30 time=27 ms
60 bytes from 172.217.4.228: icmp_seq=31 time=89 ms
60 bytes from 172.217.4.228: icmp_seq=32 time=20 ms
60 bytes from 172.217.4.228: icmp_seq=33 time=21 ms
60 bytes from 172.217.4.228: icmp_seq=34 time=22 ms
60 bytes from 172.217.4.228: icmp_seq=35 time=20 ms
60 bytes from 172.217.4.228: icmp_seq=36 time=20 ms
60 bytes from 172.217.4.228: icmp_seq=37 time=20 ms
60 bytes from 172.217.4.228: icmp_seq=38 time=19 ms
60 bytes from 172.217.4.228: icmp_seq=39 time=19 ms
60 bytes from 172.217.4.228: icmp_seq=40 time=73 ms
60 bytes from 172.217.4.228: icmp_seq=41 time=22 ms
60 bytes from 172.217.4.228: icmp_seq=42 time=49 ms
60 bytes from 172.217.4.228: icmp_seq=43 time=24 ms
60 bytes from 172.217.4.228: icmp_seq=44 time=80 ms
60 bytes from 172.217.4.228: icmp_seq=45 time=27 ms
Request timeout for icmp_seq 46
Request timeout for icmp_seq 47
Request timeout for icmp_seq 48
Request timeout for icmp_seq 49
I (91854) wifi: bcn_timout,ap_probe_send_start
I (92064) wifi: ampdu: ignore deleting tx BA0
Request timeout for icmp_seq 50
```

And with power saving off:

```
ping www.google.com 500
60 bytes from 172.217.4.228: icmp_seq=0 time=26 ms
60 bytes from 172.217.4.228: icmp_seq=1 time=26 ms
60 bytes from 172.217.4.228: icmp_seq=2 time=26 ms
60 bytes from 172.217.4.228: icmp_seq=3 time=18 ms
60 bytes from 172.217.4.228: icmp_seq=4 time=21 ms
60 bytes from 172.217.4.228: icmp_seq=5 time=18 ms
60 bytes from 172.217.4.228: icmp_seq=6 time=21 ms
60 bytes from 172.217.4.228: icmp_seq=7 time=18 ms
60 bytes from 172.217.4.228: icmp_seq=8 time=19 ms
60 bytes from 172.217.4.228: icmp_seq=9 time=40 ms
60 bytes from 172.217.4.228: icmp_seq=10 time=24 ms
60 bytes from 172.217.4.228: icmp_seq=11 time=23 ms
60 bytes from 172.217.4.228: icmp_seq=12 time=23 ms
60 bytes from 172.217.4.228: icmp_seq=13 time=46 ms
60 bytes from 172.217.4.228: icmp_seq=14 time=31 ms
60 bytes from 172.217.4.228: icmp_seq=15 time=18 ms
60 bytes from 172.217.4.228: icmp_seq=16 time=21 ms
60 bytes from 172.217.4.228: icmp_seq=17 time=35 ms
60 bytes from 172.217.4.228: icmp_seq=18 time=18 ms
60 bytes from 172.217.4.228: icmp_seq=19 time=18 ms
60 bytes from 172.217.4.228: icmp_seq=20 time=19 ms
60 bytes from 172.217.4.228: icmp_seq=21 time=21 ms
60 bytes from 172.217.4.228: icmp_seq=22 time=19 ms
60 bytes from 172.217.4.228: icmp_seq=23 time=19 ms
60 bytes from 172.217.4.228: icmp_seq=24 time=23 ms
60 bytes from 172.217.4.228: icmp_seq=25 time=20 ms
60 bytes from 172.217.4.228: icmp_seq=26 time=29 ms
60 bytes from 172.217.4.228: icmp_seq=27 time=21 ms
60 bytes from 172.217.4.228: icmp_seq=28 time=16 ms
60 bytes from 172.217.4.228: icmp_seq=29 time=18 ms
60 bytes from 172.217.4.228: icmp_seq=30 time=21 ms
60 bytes from 172.217.4.228: icmp_seq=31 time=19 ms
60 bytes from 172.217.4.228: icmp_seq=32 time=26 ms
60 bytes from 172.217.4.228: icmp_seq=33 time=23 ms
60 bytes from 172.217.4.228: icmp_seq=34 time=22 ms
60 bytes from 172.217.4.228: icmp_seq=35 time=17 ms
60 bytes from 172.217.4.228: icmp_seq=36 time=21 ms
60 bytes from 172.217.4.228: icmp_seq=37 time=16 ms
60 bytes from 172.217.4.228: icmp_seq=38 time=29 ms
60 bytes from 172.217.4.228: icmp_seq=39 time=17 ms
60 bytes from 172.217.4.228: icmp_seq=40 time=18 ms
60 bytes from 172.217.4.228: icmp_seq=41 time=31 ms
60 bytes from 172.217.4.228: icmp_seq=42 time=17 ms
60 bytes from 172.217.4.228: icmp_seq=43 time=18 ms
60 bytes from 172.217.4.228: icmp_seq=44 time=17 ms
60 bytes from 172.217.4.228: icmp_seq=45 time=21 ms
60 bytes from 172.217.4.228: icmp_seq=46 time=20 ms
60 bytes from 172.217.4.228: icmp_seq=47 time=14 ms
60 bytes from 172.217.4.228: icmp_seq=48 time=14 ms
60 bytes from 172.217.4.228: icmp_seq=49 time=13 ms
60 bytes from 172.217.4.228: icmp_seq=50 time=17 ms
```

vonnieda, 2019-12-12:
Sorry for the spam, last update on this:

Changing wifi/bluetooth coexistence control from software to hardware seems to also fix the issue. I had previously always used software.

So, final statement: WIFI_PS_MIN_MODEM + Hardware Coex seems to work fine in 3.3.1. In 3.3.0 I used WIFI_PS_MIN_MODEM + Software Coex and it worked fine, but in 3.3.1 it was unstable.

Alvin1Zhang, 2019-12-13:
@vonnieda Thanks for reporting.

TimXia, 2019-12-13:
@vonnieda What is the brand of the AP? What is Bluetooth doing when the problem happens?

vonnieda, 2019-12-13:
@TimXia It's a UniFi AP, AC Pro, I believe. I will get exact model and firmware version in the morning.

Bluetooth (BLE) is active and transmitting approx. 20 bytes per second, receiving about half that.

vonnieda, 2019-12-13:
Model: UniFi AP-AC-Pro
Version: 4.0.54.10625
Board Revision: 26

nopnop2002, 2020-01-18:
I have same issues.
I execute [this](https://github.com/espressif/esp-idf/tree/master/examples/wifi/getting_started/station) example.

My environment:

	freeRTOS version:V8.2.0
	NEWLIB version:3.0.0
	lwIP version:2-0-3-255
	ESP-IDF version:v4.0-dev-1186-gf16f4297c-dirty

Connected to Wifi with 4007 Tick and left for a while.
Wifi was disconnected at 944587Tick.

I (0) cpu_start: App cpu up.
I (372) heap_init: Initializing. RAM available for dynamic allocation:
I (379) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (385) heap_init: At 3FFB8FE8 len 00027018 (156 KiB): DRAM
I (391) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (398) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (404) heap_init: At 400904AC len 0000FB54 (62 KiB): IRAM
I (410) cpu_start: Pro cpu start user code
I (429) spi_flash: detected chip: generic
I (429) spi_flash: flash io: dio
I (429) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (497) wifi station: ESP_WIFI_MODE_STA
I (507) wifi: wifi driver task: 3ffc0b84, prio:23, stack:3584, core=0
I (507) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (507) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (537) wifi: wifi firmware version: 35d27bc
I (537) wifi: config NVS flash: enabled
I (537) wifi: config nano formating: disabled
I (537) wifi: Init dynamic tx buffer num: 32
I (547) wifi: Init data frame dynamic rx buffer num: 32
I (547) wifi: Init management frame dynamic rx buffer num: 32
I (557) wifi: Init management short buffer num: 32
I (557) wifi: Init static rx buffer size: 1600
I (557) wifi: Init static rx buffer num: 10
I (567) wifi: Init dynamic rx buffer num: 32
I (677) phy: phy_version: 4100, 2a5dd04, Jan 23 2019, 21:00:07, 0, 0
I (677) wifi: mode : sta (a4:cf:12:44:4f:00)
I (807) wifi: new:<3,0>, old:<1,0>, ap:<255,255>, sta:<3,0>, prof:1
I (1777) wifi: state: init -> auth (b0)
I (1787) wifi: state: auth -> assoc (0)
I (1787) wifi: state: assoc -> run (10)
I (1807) wifi: connected with aterm-e625c0-g, channel 3, HT20, bssid = c0:25:a2:ac:cb:ba
I (1807) wifi: pm start, type: 1

I (3997) tcpip_adapter: sta ip: 192.168.10.102, mask: 255.255.255.0, gw: 192.168.10.1
I (3997) wifi station: got ip:192.168.10.102
I (3997) wifi station: wifi_init_sta finished.
I (4007) wifi station: connect to ap SSID:aterm-e625c0-g
I (944587) wifi: state: run -> init (ec0)
I (944587) wifi: pm stop, total sleep time: 871377013 us / 942774188 us

I (944587) wifi: new:<3,0>, old:<3,0>, ap:<255,255>, sta:<3,0>, prof:1
I (944587) wifi station: retry to connect to the AP
I (944597) wifi station: connect to the AP fail
I (944707) wifi: new:<3,0>, old:<3,0>, ap:<255,255>, sta:<3,0>, prof:1
I (944717) wifi: state: init -> auth (b0)
I (945717) wifi: state: auth -> init (200)
I (945717) wifi: new:<3,0>, old:<3,0>, ap:<255,255>, sta:<3,0>, prof:1
I (945717) wifi station: retry to connect to the AP
I (945717) wifi station: connect to the AP fail
I (947767) wifi station: retry to connect to the AP
I (947767) wifi station: connect to the AP fail
I (947887) wifi: new:<3,0>, old:<3,0>, ap:<255,255>, sta:<3,0>, prof:1
I (947887) wifi: state: init -> auth (b0)
I (947897) wifi: state: auth -> assoc (0)
E (947897) wifi: Set status to INIT
I (947897) wifi: state: assoc -> init (c00)
I (947897) wifi: new:<3,0>, old:<3,0>, ap:<255,255>, sta:<3,0>, prof:1
I (947907) wifi station: retry to connect to the AP
I (947907) wifi station: connect to the AP fail
I (949957) wifi station: retry to connect to the AP
I (949957) wifi station: connect to the AP fail
I (950077) wifi: new:<3,0>, old:<3,0>, ap:<255,255>, sta:<3,0>, prof:1
I (950077) wifi: state: init -> auth (b0)
I (950087) wifi: state: auth -> assoc (0)
E (950097) wifi: Set status to INIT
I (950097) wifi: state: assoc -> init (c00)
I (950097) wifi: new:<3,0>, old:<3,0>, ap:<255,255>, sta:<3,0>, prof:1
I (950107) wifi station: connect to the AP fail

nopnop2002, 2020-01-21:
Still testing, but improved with the following changes:

1.Updating network connection code
https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/network/tcpip_adapter_migration.html

2.Add esp_wifi_set_ps().

    wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
    ESP_ERROR_CHECK(esp_wifi_init(&cfg));
    ESP_ERROR_CHECK(esp_wifi_set_ps(WIFI_PS_NONE));

nopnop2002, 2020-01-25:
I have replaced the Wifi router.
All problems are gone.
Wifi is connected for a long time.
In my case, the problem was caused by hardware.

wildwildwilliam, 2020-02-06:
@vonnieda I notice your log shows AP beacon interval and DTIM period. how do you get those values for log output? I'd like to do the same, but I struggle to find an API.

I (4605) wifi: AP's beacon interval = 102400 us, DTIM period = 1

vonnieda, 2020-02-06:
@wildwildwilliam those messages are coming from the ESP wifi code, not mine. I didn't do anything in particular to enable them.

wildwildwilliam, 2020-02-07:
@vonnieda What is your wifi firmware version? For example, mine is: I (625) wifi: wifi firmware version: c94f3e6

If ESP can open-source the wifi driver code, the community can probably help to debug a lot of the issues.

TimXia, 2020-12-15:
@vonnieda Some bugs that cause beacon timoeut have been fixed since IDFv3.3.2. Please test Wi-Fi and Bluetooth coexistence on the latest IDFv3.3/IDFv4.0/IDFv4.1/IDF/v4.2.

TimXia, 2020-12-15:
@wildwildwilliam The log that shows AP beacon interval and DTIM period is printed by Wi-Fi firmware since IDFv3.3.2.

AxelLin, 2021-11-13:
> @vonnieda Some bugs that cause beacon timoeut have been fixed since IDFv3.3.2. Please test Wi-Fi and Bluetooth coexistence on the latest IDFv3.3/IDFv4.0/IDFv4.1/IDF/v4.2.

@vonnieda 
Any chance to verify with above IDF versions?

jack0c, 2022-01-05:
This bug should be fix since IDFv3.3.2.

closed 2022-01-05 by jack0c