espressif/esp-idf #4480: Missing dependencies for fresh install on Raspberry Pi 3B+ with Buster Lite

AloyseTech, 2019-12-13:
----------------------------- Delete below -----------------------------

If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at esp32.com instead.

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [ ] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [ ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ ] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs on the monitor](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/idf-monitor.html#automatically-decoding-addresses) or the [coredump](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Provide more items under **Other items if possible** can help us better locate your problem.
5. Use markup (buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

----------------------------- Delete above -----------------------------

## Environment

- IDF version (run ``git describe --tags`` to find it): v4.1-dev-1086-g93a8603c5
- Build System:         CMake
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
- Operating System:     Linux (Raspbian Buster Lite
- Power Supply:         USB

## Problem Description

//Detailed problem description goes here.

### Expected Behavior
when running install.sh, the tools installation should succeed.

### Actual Behavior
Installation fails at 2 stages:
It first miss libffi, then opensslv.

To fix the missing dependencies I've run the following command :

> sudo apt-get install libffi-dev libssl-dev

After installing the 2 additional dependencies, install.sh succeed.

When proceeding to build the hello-world example, the compilation fails with segfault error.
To fix this, I updated the Raspberry Pi firmware using 

> sudo rpi-update

Building works successfully after the update.

### Steps to repropduce

Install a fresh Raspbian Buster Lite on a RPI3B+, follow esp-idf getting started for linux.

dobairoland, 2019-12-16:
Hi @AloyseTech. Did you install the prerequisites for Debian according to this page https://docs.espressif.com/projects/esp-idf/en/latest/get-started/linux-setup.html?

AloyseTech, 2019-12-16:
Hi @dobairoland ,
Yes, I followed the getting started guide for linux/debian and installed the following dependencies :
`sudo apt-get install git wget flex bison gperf python python-pip python-setuptools python-serial python-click python-cryptography python-future python-pyparsing python-pyelftools cmake ninja-build ccache`

dobairoland, 2019-12-16:
Thank you for your report. I'll update the documentation. Newer versions of `apt-get` doesn't install as many packages as older ones and we need to add more and more packages to our list manually. Thanks!

AloyseTech, 2019-12-16:
Please also note that without updating the Raspberry Pi firmware, the build fails with segfault. I'm not sure what cause the issue though (I tried the firmware update after reading this : https://github.com/esphome/issues/issues/677).

dobairoland, 2019-12-16:
We don't have recommendations in the documentation regarding to do system update for any operating systems. I just assume (hope) that everyone keeps up-to-date their systems with regular security and bugfix updates.

I'm guessing that `apt-get` installed the latest versions of these packages and the rest of the system was out-of-date and linked against different or older packages.

closed 2019-12-18 by espressif-bot