espressif/esp-idf #4484: How to get RSSI of my wifi network.

ROYMURLI, 2019-12-13:
Is there any function from which I can get my wifi RSSI directly without scanning the wifi?
Actually I am using the -
static wifi_ap_record_t *ap_records;
esp_wifi_scan_start(&scan_config, true);
esp_wifi_scan_get_ap_num(&ap_num);
ap_records = (wifi_ap_record_t *) malloc(sizeof(wifi_ap_record_t) * ap_num);
esp_wifi_scan_get_ap_records(&ap_num, ap_records);

after this I am priting the wifi ssid and rssi and auth.

strcpy(ssid_list[ap_index+1],(char *)ap_records[ap_index].ssid);
						strcpy(wifi_temp,(char *)ap_records[ap_index].ssid);
						//wifi_temp[14] = '.';
						wifi_temp[13] = '~';
						wifi_temp[14] = '\0';
						sprintf(ret_val, "%s<%d<%d", wifi_temp, ap_records[ap_index].rssi, ap_records[ap_index].authmode);	
						ap_index++;

Till here I am able to do everything and also able to connect to wifi and I am writing my ssid and password in NVS.

But what I want to know later on if I need to/print my wifi rssi how can I do it.
I used this function:-
static wifi_ap_record_t curr_wifi;
esp_wifi_scan_get_ap_records(1, curr_wifi);
// here curr_wifi is my stored SSID in NVS which I chosed to connect.
But this did not work.

vonnieda, 2019-12-13:
```
wifi_ap_record_t ap;
esp_wifi_sta_get_ap_info(&ap);
printf("%d\n", ap.rssi);
```

See https://docs.espressif.com/projects/esp-idf/en/stable/api-reference/network/esp_wifi.html#_CPPv424esp_wifi_sta_get_ap_infoP16wifi_ap_record_t

ROYMURLI, 2019-12-14:
Thanks, It worked. 👍

closed 2019-12-14 by ROYMURLI