espressif/esp-idf #4485: lwIP assert triggering when compiling firmware with log disabled

KaeLL, 2019-12-13:
----------------------------- Delete below -----------------------------

If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at esp32.com instead.

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [ ] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [ ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ ] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs on the monitor](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/idf-monitor.html#automatically-decoding-addresses) or the [coredump](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Provide more items under **Other items if possible** can help us better locate your problem.
5. Use markup (buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:      [none]
- Module or chip used:  [ESP32-WROOM-32D]
- IDF version (run ``git describe --tags`` to find it): v4.1-dev-1086-g93a8603c5
- Build System:         [Make]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): esp-2019r2) 8.2.0
- Operating System:     [Linux]
- Power Supply:         [external 5V]

## Problem Description
The following assertion keeps triggering when the firmware is compiled with all logs disabled:
https://github.com/espressif/esp-lwip/blob/c9e3b53c6f04052943f97210b858cec805cc98d9/src/api/tcpip.c#L455

### Expected Behavior
Not triggering it?!
### Actual Behavior
Triggers it. 
### Steps to repropduce
I tried reproducing it on the http-client example, but it worked correctly.

## Debug Logs

```
assertion "Invalid mbox" failed: file "esp-idf/components/lwip/lwip/src/api/tcpip.c", line 455, function: tcpip_send_msg_wait_sem
abort() was called at PC 0x4010ae17 on core 0
0x4010ae17: __assert_func at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdlib/assert.c:62 (discriminator 8)

ELF file SHA256: 0b5c4522e3671a2cb8ed9f79cf363bb764d77be8a802fce7fb7ab002de5feb3f

Backtrace: 0x40094918:0x3ffbd250 0x40095099:0x3ffbd270 0x4010ae17:0x3ffbd290 0x400f9d13:0x3ffbd2c0 0x401091b5:0x3ffbd2f0 0x40132321:0x3ffbd330 0x4017d96d:0x3ffbd380 0x4017dce5:0x3ffbd3e0 0x4018036b:0x3ffbd400 0x401825a9:0x3ffbd420 0x4017e8b4:0x3ffbd440 0x4017ee75:0x3ffbd460 0x4017d8bb:0x3ffbd480 0x40130901:0x3ffbd510 0x4013096d:0x3ffbd550 0x401200e1:0x3ffbd590 0x4011f8f7:0x3ffbd620 0x40127b92:0x3ffbd650 0x40127cba:0x3ffbd670 0x4010aa71:0x3ffbd6b0 0x4010aabf:0x3ffbd6f0 0x400d1154:0x3ffbd710
0x40094918: invoke_abort at esp-idf/components/esp32/panic.c:159

0x40095099: abort at esp-idf/components/esp32/panic.c:174

0x4010ae17: __assert_func at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdlib/assert.c:62 (discriminator 8)

0x400f9d13: tcpip_send_msg_wait_sem at esp-idf/components/lwip/lwip/src/api/tcpip.c:455 (discriminator 1)

0x401091b5: netconn_gethostbyname_addrtype at esp-idf/components/lwip/lwip/src/api/api_lib.c:1325 (discriminator 2)

0x40132321: lwip_getaddrinfo at esp-idf/components/lwip/lwip/src/api/netdb.c:342

0x4017d96d: resolve_host_name at esp-idf/components/esp-tls/esp_tls.c:136
 (inlined by) esp_tcp_connect at esp-idf/components/esp-tls/esp_tls.c:155
 (inlined by) esp_tls_low_level_conn at esp-idf/components/esp-tls/esp_tls.c:234

0x4017dce5: esp_tls_conn_new_sync at esp-idf/components/esp-tls/esp_tls.c:332

0x4018036b: ssl_connect at esp-idf/components/tcp_transport/transport_ssl.c:74

0x401825a9: esp_transport_connect at esp-idf/components/tcp_transport/transport.c:165

0x4017e8b4: esp_http_client_connect at esp-idf/components/esp_http_client/esp_http_client.c:1016

0x4017ee75: esp_http_client_perform at esp-idf/components/esp_http_client/esp_http_client.c:890

0x4017d8bb: esp_http_ar_make_request at test/components/esp_http_client_ar/esp_http_client_ar.c:202

0x4010aabf: app_main at test/main/main.c:276

0x400d1154: main_task at esp-idf/components/esp32/cpu_start.c:569```

## Other items if possible
https://gist.github.com/KaeLL/c8d01de4547d284e81b4cf7f3a2457ae

freakyxue, 2020-03-04:
hi @KaeLL 
 Which option do you disable to display this assert? How can I reproduce it

KaeLL, 2020-03-04:
I should've closed this issue. If I recall correctly, this assert was triggering because I had forgotten to initialize the wifi stack before performing a request.

seregega, 2020-04-24:
Correct, it's done after init wi-fi stack first.

closed 2020-05-27 by Alvin1Zhang

Mcurti12, 2021-05-27:
@KaeLL Hey! Any chance you'd be able to go into more detail about how you fixed this? I am having the same issue trying to use WiFi and HTTP and am getting the same assertion "Invalid mbox" failed: as you. Any help on what you did would be amazing :)

jimmieclark3, 2021-06-03:
> I should've closed this issue. If I recall correctly, this assert was triggering because I had forgotten to initialize the wifi stack before performing a request.

I literally did the same thing.  THANK YOU for the note.  Completely missed it.

KaeLL, 2021-06-03:
@Mcurti12 make sure you got the [init](https://github.com/espressif/esp-idf/blob/master/examples/common_components/protocol_examples_common/connect.c#L275-L310) right.

Kaushal-maniyar, 2022-04-07:
> I should've closed this issue. If I recall correctly, this assert was triggering because I had forgotten to initialize the wifi stack before performing a request.

Thank you, it helped a lot.

vmashchenko, 2023-04-26:
I had the same issue and solved it in such a way:

```

// added at the top
#include "esp_wifi.h" 
#include "nvs_flash.h" 

void app_main(void)       
{
    // Initialize NVS
    esp_err_t ret = nvs_flash_init();
    if (ret == ESP_ERR_NVS_NO_FREE_PAGES || ret == ESP_ERR_NVS_NEW_VERSION_FOUND) {
      ESP_ERROR_CHECK(nvs_flash_erase());
      ret = nvs_flash_init();
    }
    ESP_ERROR_CHECK(ret);

    // Initialize TCP/IP stack
    tcpip_adapter_init();

    // Initialize WiFi
    wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
    ESP_ERROR_CHECK(esp_wifi_init(&cfg));

    // Set WiFi mode
    ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_STA));

    // Start WiFi
    ESP_ERROR_CHECK(esp_wifi_start());
}

void setup() 
{
	
  app_main(); 
   ...                             // the rest my code
 }
```