espressif/esp-idf #4487: FLASH Access Errors (corruption?) on 16MB ESP32-WROOM if JTAG connected during serial programming

bgintz-rsc, 2019-12-14:
## Environment

- Development Kit:     none
- Kit version none
- Module or chip used:  8 MB ESP32-WROOM-32D and 16 MB ESP32-WROOM-32U
- IDF version
    - v4.0-beta2-174-g99fb9a3f7 but 
    - also replicated on stable release 3.2 and above 

- Build System:        
  - CMake
- Compiler version:
   - xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
- Operating System:    
  - OSX 10.14.6
- Power Supply:        
  -  external 3.3V

## Problem Description

FLASH access errors and reboot error message displayed on 16MB ESP32-WROOM-32U which are not present in 8MB ESP32-WROOM-32D running identical code on the same hardware.

### Issues

1. flash read err, 1000 always reported on boot of 16MB ESP32-WROOM-32U (program runs).  
1. FLASH access errors occur on 16MB ESP32-WROOM-32U when a JTAG debug probe is connected to the device during flashing with the serial port using an ESP-PROG board.  Disconnecting the JTAG resolves these issues which include:  

    --  SPIFFS cannot access FLASH to mount the filesystem
    -- Attempts to access NVS partitions fail (not related to addition to SPIFFS)

NOTES:
    -- These issues  can be easily replicated using the spiffs example in the ESP-IDF and appears to be unrelated to compiler or IDF version but do occur **only** if there is a JTAG probe connected to the board on power up.
    -- This appears to be an initialization issue with configuring JTAG access of the FLASH  following a firmware load via the UART. 
   -- The flash was erased prior to starting each test 

### Expected Behavior

FLASH operations on a 16MB ESP32-WROOM-32U should not fail when a JTAG probe is connected connected on power up. Also there should be no "flash read err, 1000" reported on boot.   Instead the behavior should be the same as the  8MB ESP32-WROOM-32D 

### Actual Behavior

#### See above

### Steps to reproduce

1. Connect an ESP-PROG board to the 16MB ESP32-WROOM-32U module (JTAG and serial interfaces)
2. Build the spiffs sample code: 
    "idf.py build"
3.  flash the firmware using the command line provided by the build system:  
    $ ../../../components/esptool_py/esptool/esptool.py -p /dev/tty.usbserial-1463301 -b 460800 --after hard_reset write_flash --flash_mode dio --flash_size detect --flash_freq 40m 0x1000 build/bootloader/bootloader.bin 0x8000 build/partition_table/partition-table.bin 0x10000 build/spiffs.bin"
4. Open up a monitor to view the serial port of the device -  observe the failure
5. Disconnect the JTAG probe and reset the device - observe the code runs successfully (but still issues a flash error on boot)
6. Replace the 16MB module with the 8MB module and repeat the above test with JTAG connected and observe no failures.

## Debug Logs (Important: bold text indicates errors)

### Test 1:  Boot log:  16MB Module with JTAG and UART connected after programming the device:
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
**flash read err, 1000**
ets_main.c 371 
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6884
[... 238 lines cut ...]
I (219) cpu_start: App version:      v4.0-beta2-174-g99fb9a3f7
I (226) cpu_start: Compile time:     Dec 14 2019 11:52:46
I (232) cpu_start: ELF file SHA256:  5735ff317e5ee814...
I (238) cpu_start: ESP-IDF:          v4.0-beta2-174-g99fb9a3f7
I (244) cpu_start: Starting app cpu, entry point is 0x40080fc0
I (0) cpu_start: App cpu up.
I (255) heap_init: Initializing. RAM available for dynamic allocation:
I (262) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (268) heap_init: At 3FFB3068 len 0002CF98 (179 KiB): DRAM
I (274) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (280) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (287) heap_init: At 40089B08 len 000164F8 (89 KiB): IRAM
I (293) cpu_start: Pro cpu start user code
I (311) spi_flash: detected chip: generic
I (312) spi_flash: flash io: dio
I (312) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (320) example: Initializing SPIFFS
I (410) example: Partition size: total: 896321, used: 502
I (410) example: Opening file
I (510) example: File written
I (520) example: Renaming file
I (570) example: Reading file
I (570) example: Read from file: 'Hello World!'
I (570) example: SPIFFS unmounted

igrr, 2019-12-14:
Please check https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/jtag-debugging/tips-and-quirks.html#why-to-set-spi-flash-voltage-in-openocd-configuration — this may explain the behavior you are seeing. Please check that MTDI pin state at reset results in the correct strapping of VDD_SDIO voltage.

bgintz-rsc, 2019-12-17:
Thank you for for the response.

1. MTDI is only connected to the ESP-PROG board as shown:

[image]

2. Openocd is not running during any of the above tests (the JTAG is passively connected) this is identical on both the 16MB module which fails and the 8 MB module which does not fail.

3. Does the 16MB part require a different configuration than the 8 MB part for VDD_SDIO?  If so, where is thus documented.  Again - the problem can be replicated on the 16MB and not on the 8MB.

igrr, 2019-12-17:
The problem seems to be related to strapping:

```
rst:0x10 (RTCWDT_RTC_RESET),boot:0x33
```

In the first log where you are observing the SPIFFS issue the strapping value is 0x33, whereas in other cases it is 0x13. This value is equal to {10'b0, MTDI, GPIO0, GPIO2, GPIO4, MTDO, GPIO5}. So when the value is 0x33, it means that MTDI was sampled "high" at reset. The reason could be lack of a pull-down resistor on this pin — at least I don't see one in the part of the schematic you have shared. If there is no pull-down resistor and the pin is connected to the JTAG cable (which ends with a hi-Z output of the adapter) then the pin is effectively "floating" and random noise may be induced. Try adding a pull-down resistor on this pin to see if the problem persists.

I can't explain why this issue is not observed with the 8MB module; possibly something between the two boards is different enough to not result in the same amount of noise induced on MTDI pin.

bgintz-rsc, 2019-12-18:
Hello Ivan,.

Thanks for the suggestion.  I tried this with both a 10K and 5K pull down to ground.  The boot pin strapping still shows as 0x33 and the flash error message and failures persist.  Here is a log:

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6896
ho 0 tail 12 room 4
load:0x40078000,len:14076
load:0x40080400,len:4304
entry 0x400806e8
I (75) boot: Chip Revision: 1
I (75) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (41) boot: ESP-IDF v4.0-beta2-174-g99fb9a3f7 2nd stage bootloader
I (41) boot: compile time 13:34:03
I (41) boot: Enabling RNG early entropy source...
I (47) boot: SPI Speed      : 40MHz
I (52) boot: SPI Mode       : DIO
I (56) boot: SPI Flash Size : 2MB
I (60) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (70) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (78) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (85) boot:  2 factory          factory app      00 00 00010000 00100000
I (93) boot:  3 storage          Unknown data     01 82 00110000 000f0000
I (100) boot: End of partition table
I (105) boot_comm: chip revision: 1, min. application chip revision: 0
I (112) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x05ea4 ( 24228) map
I (130) esp_image: segment 1: paddr=0x00015ecc vaddr=0x3ffb0000 size=0x02070 (  8304) load
I (133) esp_image: segment 2: paddr=0x00017f44 vaddr=0x40080000 size=0x00400 (  1024) load
I (139) esp_image: segment 3: paddr=0x0001834c vaddr=0x40080400 size=0x07cc4 ( 31940) load
I (161) esp_image: segment 4: paddr=0x00020018 vaddr=0x400d0018 size=0x18374 ( 99188) map
I (197) esp_image: segment 5: paddr=0x00038394 vaddr=0x400880c4 size=0x01a44 (  6724) load
I (206) boot: Loaded app from partition at offset 0x10000
I (206) boot: Disabling RNG early entropy source...
I (208) cpu_start: Pro cpu up.
I (212) cpu_start: Application information:
I (217) cpu_start: Project name:     spiffs
I (222) cpu_start: App version:      1
I (226) cpu_start: Compile time:     Dec 15 2019 13:33:56
I (232) cpu_start: ELF file SHA256:  5ad0af613e32e456...
I (238) cpu_start: ESP-IDF:          v4.0-beta2-174-g99fb9a3f7
I (245) cpu_start: Starting app cpu, entry point is 0x40080fc0
I (0) cpu_start: App cpu up.
I (255) heap_init: Initializing. RAM available for dynamic allocation:
I (262) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (268) heap_init: At 3FFB3068 len 0002CF98 (179 KiB): DRAM
I (274) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (281) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (287) heap_init: At 40089B08 len 000164F8 (89 KiB): IRAM
I (293) cpu_start: Pro cpu start user code
I (311) spi_flash: detected chip: generic
I (312) spi_flash: flash io: dio
W (312) spi_flash: Detected size(16384k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (323) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (333) example: Initializing SPIFFS
I (423) example: Partition size: total: 896321, used: 502
I (423) example: Opening file
E (503) example: Failed to open file for writing

igrr, 2019-12-18:
Can you try connecting a scope to the MTDI pin, and observe the signal around the time when reset happens?

bgintz-rsc, 2019-12-20:
Here are two scope captures and boot logs Ivan.  

# 16MB module with a 5K pulldown:  Program fails, flash read error persists, boot strapping 0x33
[image]

st:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6896
ho 0 tail 12 room 4
load:0x40078000,len:14076
load:0x40080400,len:4304
entry 0x400806e8
I (75) boot: Chip Revision: 1
I (75) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (41) boot: ESP-IDF v4.0-beta2-174-g99fb9a3f7 2nd stage bootloader
I (41) boot: compile time 13:34:03
I (41) boot: Enabling RNG early entropy source...
I (47) boot: SPI Speed      : 40MHz
I (52) boot: SPI Mode       : DIO
I (56) boot: SPI Flash Size : 2MB
I (60) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (70) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (78) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (85) boot:  2 factory          factory app      00 00 00010000 00100000
I (93) boot:  3 storage          Unknown data     01 82 00110000 000f0000
I (100) boot: End of partition table
I (105) boot_comm: chip revision: 1, min. application chip revision: 0
I (112) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x05ea4 ( 24228) map
I (130) esp_image: segment 1: paddr=0x00015ecc vaddr=0x3ffb0000 size=0x02070 (  8304) load
I (133) esp_image: segment 2: paddr=0x00017f44 vaddr=0x40080000 size=0x00400 (  1024) load
I (139) esp_image: segment 3: paddr=0x0001834c vaddr=0x40080400 size=0x07cc4 ( 31940) load
I (161) esp_image: segment 4: paddr=0x00020018 vaddr=0x400d0018 size=0x18374 ( 99188) map
I (197) esp_image: segment 5: paddr=0x00038394 vaddr=0x400880c4 size=0x01a44 (  6724) load
I (206) boot: Loaded app from partition at offset 0x10000
I (206) boot: Disabling RNG early entropy source...
I (208) cpu_start: Pro cpu up.
I (212) cpu_start: Application information:
I (217) cpu_start: Project name:     spiffs
I (222) cpu_start: App version:      1
I (226) cpu_start: Compile time:     Dec 15 2019 13:33:56
I (232) cpu_start: ELF file SHA256:  5ad0af613e32e456...
I (238) cpu_start: ESP-IDF:          v4.0-beta2-174-g99fb9a3f7
I (245) cpu_start: Starting app cpu, entry point is 0x40080fc0
I (0) cpu_start: App cpu up.
I (255) heap_init: Initializing. RAM available for dynamic allocation:
I (262) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (268) heap_init: At 3FFB3068 len 0002CF98 (179 KiB): DRAM
I (274) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (281) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (287) heap_init: At 40089B08 len 000164F8 (89 KiB): IRAM
I (293) cpu_start: Pro cpu start user code
I (311) spi_flash: detected chip: generic
I (312) spi_flash: flash io: dio
W (312) spi_flash: Detected size(16384k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (323) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (333) example: Initializing SPIFFS
I (423) example: Partition size: total: 896321, used: 502
I (423) example: Opening file
E (503) example: Failed to open file for writing

[... 43 lines cut ...]
I (226) cpu_start: Compile time:     Dec 15 2019 13:33:56
I (232) cpu_start: ELF file SHA256:  5ad0af613e32e456...
I (238) cpu_start: ESP-IDF:          v4.0-beta2-174-g99fb9a3f7
I (244) cpu_start: Starting app cpu, entry point is 0x40080fc0
I (0) cpu_start: App cpu up.
I (255) heap_init: Initializing. RAM available for dynamic allocation:
I (261) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (267) heap_init: At 3FFB3068 len 0002CF98 (179 KiB): DRAM
I (274) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (280) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (287) heap_init: At 40089B08 len 000164F8 (89 KiB): IRAM
I (293) cpu_start: Pro cpu start user code
I (311) spi_flash: detected chip: generic
I (311) spi_flash: flash io: dio
W (312) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (322) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (333) example: Initializing SPIFFS
I (423) example: Partition size: total: 896321, used: 502
I (423) example: Opening file
I (523) example: File written
I (533) example: Renaming file
I (583) example: Reading file
I (583) example: Read from file: 'Hello World!'
I (583) example: SPIFFS unmounted

igrr, 2019-12-20:
Is the "channel 1" showing the MTDI pin? In this case, it seems that something is pulling the signal high after power-on. I suppose this the FTDI chip on the JTAG adapter board that is driving the pin high.

In the case of the 8MB module, this is not causing an issue, because that module doesn't exhibit the "flash read error" at startup. So the chip captures the state of the pin while it is still "low".

In the case of the 16MB module, initially the strapping is also correct. However the chip resets shortly afterwards, and at that time MTDI is already high. This results in an incorrect bootstrapping.

The flash read error is related to the section 3.8 in ESP32 errata, and the cause is slightly longer time it takes the flash chip to become ready. See this recent issue for the explanation: https://github.com/espressif/esp-idf/issues/4441.

I can reproduce the issue using the ESP-PROG board connected to a DevKit-C. The default state of MTDI on ESP-PROG (without OpenOCD running) is high, which results in an incorrect strapping. This is confirmed by the [FTDI documentation](https://www.ftdichip.com/Support/Documents/AppNotes/AN_184%20FTDI%20Device%20Input%20Output%20Pin%20States.pdf) which says that ADBUS1, the pin used as MTDI, defaults to the RXD function, making it idle high.

Note that OpenOCD for the ESP32 can be configured to keep MTDI pin in the correct state to prevent such conflicts. This is controlled by the `ESP32_FLASH_VOLTAGE` parameter in the configuration files, and should be correctly set to 3.3V in `esp-wroom-32.cfg`, resulting in MTDI pin low when idle. This only applies to the case when OpenOCD is either running, or was running at least once after ESP-PROG was powered up. Only if the ESP-PROG was plugged in, and OpenOCD hasn't run yet, then MTDI will be high.

bgintz-rsc, 2019-12-20:
Nope.  Ch1 is reset.  Ch 2 is MTDI

bgintz-rsc, 2020-01-03:
I would respectfully request that you attempt to replicate this issue. The scope traces show that pulling down the MTDI signal does not solve the issue nore does it cause the initial state of the line to be read as low.  

We have a significant production order and SMT build scheduled and need to resolve this discrepancy (or at the least rule out any potential  ill affects from it).

Thank you

igrr, 2020-01-03:
@bgintz-rsc As mentioned above, I can reproduce the issue, but only until OpenOCD is started. Once OpenOCD is started, the issue no longer occurs on my end. Is this the same in your case, or you can also reproduce this when OpenOCD is running?

bgintz-rsc, 2020-01-07:
Apologies Ivan.  I had not read your response completely.  Your explanation makes sense and I will try to replicate the behavior with openocd.  

However regarding:  

_"I can reproduce the issue using the ESP-PROG board connected to a DevKit-C. The default state of MTDI on ESP-PROG (without OpenOCD running) is high, which results in an incorrect strapping. This is confirmed by the FTDI documentation which says that ADBUS1, the pin used as MTDI, defaults to the RXD function, making it idle high."_

Did you attempt to pull MTDI low while testing this as you had suggested previously to me?  Did it still fail?

bgintz-rsc, 2020-01-08:
Another question Ivan:  We have a number of boards already built.  Do we need to add the pulldown to the MTDI line on the 16MB module for production units that will not use JTAG?  Can we affect this change via an entry in sdkconfig.defaults instead?

igrr, 2020-01-08:
Regarding the 2nd question: the chip does have an internal weak pull-down on MTDI, so in case when you aren't connecting anything to this pin, an additional resistor is not required. However, please refer to the ESP32 hardware design guidelines document, and post hardware questions on ESP32.com or mail them to sales@espressif.com. This issue tracker is limited to bug reports and feature requests related to the SDK (ESP-IDF).

Regarding the first question: will try and report back.

bgintz-rsc, 2020-01-24:
Hello Ivan.  

FYI - I thought of and successfully tested a fix for this issue.  I am posting it here even though it is not per se an SDK issue.  

The solution is to hard code the flash voltage to 3.3V using the espefuse command.  The MTDI pin is  then ignored on boot and the module initializes correctly.

IMO this is clearly a design flaw in the 16MB modules and should be added to the errata for the part.  Please pass this on to the appropriate contact.

igrr, 2020-01-26:
Thanks for sharing, we'll add the tip about burning the flash voltage eFuse to the JTAG debugging guide!

tom-borcin, 2021-03-26:
Hello @bgintz-rsc, we are closing this issue, because we believe it is resolved now.

The problem is now documented in JTAG debugging guide: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/jtag-debugging/tips-and-quirks.html?highlight=flash_voltage#why-to-set-spi-flash-voltage-in-openocd-configuration

Thank you for reporting this!

closed 2021-03-26 by tom-borcin