espressif/esp-idf #4489: power off: restart after brownout

syu-source, 2019-12-16:
I am using esp32 to control LED, having a ISSUE during power off:

After cut off the power,  because of the capacitor, It takes about 2 seconds for voltage to drop from 3.3v to 1v.   Esp32 browns out and restarts 2 or 3 times during voltage dropping, which causes my LED to flash.
Is there a way to stop esp32 to restart during power off?

syu-source, 2019-12-17:
After disable brownout detection, it works.
Is there a guarantee that during VCC voltage droping, esp32 turns off and does not restart. Saying, turn off @2.2v, and will not try to restart here?  
I'm thinking that  maybe WIFI requires more power and likely brings esp32 down, but restart procedure requires less power hence ESP32 goes up.

AloyseTech, 2019-12-18:
You could try to handle the brown out reset at startup and wait ~2 sec. You can use `rtc_get_reset_reason()` to get the reset reason. It will return `ESP_RST_BROWNOUT` in case of a brown out detection.

costaud, 2019-12-19:
Is it possible to add a diode in the circuit to make the voltage drop quickly ?

syu-source, 2019-12-21:
> You could try to handle the brown out reset at startup and wait ~2 sec. You can use `rtc_get_reset_reason()` to get the reset reason. It will return `ESP_RST_BROWNOUT` in case of a brown out detection.

Thanks for replay.

It will cause slow response of led.   In fact, in order to bring pwm signal quicker, I put the ledc function into bootloader, to save 500ms boot time of RTOS.  I cannot wait the 2 seconds.

New update, after disable brownout detection, it is kind of working.  During my test, my led flashes one time of around 30 times of power  off.
It appears that there is no guarantee?

syu-source, 2019-12-21:
> Is it possible to add a diode in the circuit to make the voltage drop quickly ?

Trying to refine the circuit, but how quick is quick enough..?