espressif/esp-idf #4490: Build failure in current ESP-IDF master tree

AxelLin, 2019-12-16:
Module or chip used:
ESP32-WROOM-32
IDF version (run git describe --tags to find it):
ESP-IDF v4.1-dev-1533-g9dbb7526c5d1
Build System: [Make|CMake]
CMake
Compiler version (run xtensa-esp32-elf-gcc --version to find it):
xtensa-esp32-elf-gcc (crosstool-NG esp32-2019r1-rc2) 8.2.0
Operating System: [Windows|Linux|macOS]
Linux
Power Supply: [USB|external 5V|external 3.3V|Battery]
USB

After update today's master tree, I got below build failure:

/home/axel/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/axel/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/lib/libstdc++.a(eh_personality.o):(.literal._ZL21base_of_encoded_valuehP15_Unwind_Context+0x0): undefined reference to `_Unwind_GetTextRelBase'
/home/axel/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/axel/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/lib/libstdc++.a(eh_personality.o):(.literal._ZL21base_of_encoded_valuehP15_Unwind_Context+0x4): undefined reference to `_Unwind_GetDataRelBase'
/home/axel/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/axel/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/lib/libstdc++.a(eh_personality.o):(.literal._ZL21base_of_encoded_valuehP15_Unwind_Context+0x8): undefined reference to `_Unwind_GetRegionStart'
/home/axel/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/axel/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/lib/libstdc++.a(eh_personality.o):(.literal.__gxx_personality_v0+0x8): undefined reference to `_Unwind_GetLanguageSpecificData'
/home/axel/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/axel/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/lib/libstdc++.a(eh_personality.o):(.literal.__gxx_personality_v0+0xc): undefined reference to `_Unwind_GetIPInfo'
/home/axel/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/axel/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/lib/libstdc++.a(eh_personality.o):(.literal.__gxx_personality_v0+0x10): undefined reference to `_Unwind_SetGR'
/home/axel/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/axel/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/lib/libstdc++.a(eh_personality.o):(.literal.__gxx_personality_v0+0x14): undefined reference to `_Unwind_SetIP'
/home/axel/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/axel/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/lib/libstdc++.a(eh_personality.o): in function `base_of_encoded_value(unsigned char, _Unwind_Context*)':
/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/gcc/libstdc++-v3/../libgcc/unwind-pe.h:114: undefined reference to `_Unwind_GetTextRelBase'
/home/axel/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/gcc/libstdc++-v3/../libgcc/unwind-pe.h:116: undefined reference to `_Unwind_GetDataRelBase'
/home/axel/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/gcc/libstdc++-v3/../libgcc/unwind-pe.h:118: undefined reference to `_Unwind_GetRegionStart'
/home/axel/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/axel/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/lib/libstdc++.a(eh_personality.o): in function `parse_lsda_header(_Unwind_Context*, unsigned char const*, lsda_header_info*)':
/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/eh_personality.cc:55: undefined reference to `_Unwind_GetRegionStart'
/home/axel/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/axel/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/lib/libstdc++.a(eh_personality.o): in function `__gxx_personality_v0':
/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/eh_personality.cc:438: undefined reference to `_Unwind_GetLanguageSpecificData'
/home/axel/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/eh_personality.cc:448: undefined reference to `_Unwind_GetIPInfo'
/home/axel/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/eh_personality.cc:712: undefined reference to `_Unwind_SetGR'
/home/axel/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/eh_personality.cc:714: undefined reference to `_Unwind_SetGR'
/home/axel/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/eh_personality.cc:716: undefined reference to `_Unwind_SetIP'
/home/axel/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/axel/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/lib/libstdc++.a(eh_personality.o): in function `~end_catch_protect':
/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/eh_personality.cc:742: undefined reference to `_Unwind_Resume'
/home/axel/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/axel/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/lib/libstdc++.a(eh_terminate.o):(.literal._ZN10__cxxabiv111__terminateEPFvvE+0x0): undefined reference to `_Unwind_Resume'
/home/axel/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/axel/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/lib/libstdc++.a(eh_terminate.o): in function `__cxxabiv1::__terminate(void (*)())':
/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/eh_terminate.cc:43: undefined reference to `_Unwind_Resume'
/home/axel/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/axel/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/lib/libstdc++.a(eh_globals.o):(.literal._ZL15eh_globals_dtorPv+0x0): undefined reference to `_Unwind_DeleteException'
/home/axel/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/axel/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/lib/libstdc++.a(eh_globals.o): in function `eh_globals_dtor(void*)':
/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/eh_globals.cc:83: undefined reference to `_Unwind_DeleteException'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

AxelLin, 2019-12-16:
Bisect to below commit:

commit 5b83b198c60394a9e544aaf239d2432c81c54a20
Author: Alexey Gerenkov <alexey@espressif.com>
Date:   Tue Oct 1 20:42:32 2019 +0300

    gcov: Fixes libgcov and libapp_trace cross dependency linking issue

    Closes IDF-834

You can reproduce the build failure by below steps:
cd examples/system/app_trace_to_host
idf.py menuconfig
  Application Level Tracing -> Data Destination (Trace memory)  -> Select "None"
idf.py build

igrr, 2019-12-16:
You seem to be running an unsupported version of the toolchain (for the given IDF commit): `crosstool-NG esp32-2019r1-rc2) 8.2.0`. Please follow the getting started guide and install the correct toolchain version (may involve running install.sh if you are on Linux):

https://docs.espressif.com/projects/esp-idf/en/latest/get-started/index.html#step-3-set-up-the-tools

AxelLin, 2019-12-16:
It works after upgrading to xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0 and run idf.py fullclean.
Thanks.

closed 2019-12-16 by AxelLin