espressif/esp-idf #4491: Esp32-Pico-D4 panic @gpio_config()

syu-source, 2019-12-16:
esp32-pico-d4 panic but ESP32-DevKitC works just fine with exactly the same binary image.

ESP-IDF version:  v4.0-dev-1287-gd7e659df2

Code causes panic on pico: 
```
void ir_gpio_init()
{
	INFO("INIT 1");
    gpio_config_t io_conf = {0};
    io_conf.pin_bit_mask = 1ULL << GPIO_IR_RECV;
    io_conf.intr_type = GPIO_INTR_ANYEDGE;  //GPIO_PIN_INTR_NEGEDGE
    io_conf.mode = GPIO_MODE_INPUT;
    //io_conf.pull_down_en = 1;
    //io_conf.pull_up_en = 1;
	INFO("INIT 1.2");

    gpio_config(&io_conf);
	INFO("INIT 1.3");

    gpio_set_pull_mode(GPIO_IR_RECV, GPIO_FLOATING);
	INFO("INIT 2");

    io_conf.pin_bit_mask = 1 << 2;
    io_conf.mode = GPIO_MODE_OUTPUT;
    io_conf.intr_type = GPIO_INTR_DISABLE;
    gpio_config(&io_conf);
	/************************/
	INFO("INIT 3");

   //change gpio intrrupt type for one pin
    gpio_set_intr_type(GPIO_IR_RECV, GPIO_INTR_ANYEDGE);//GPIO_INTR_ANYEDGE
}
```

Output from idf-monitor:
`
```
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6412
load:0x40078000,len:11872
load:0x40080400,len:6636
entry 0x40080768
I (29) boot: ESP-IDF v4.0-dev-1287-gd7e659df2-dirty 2nd stage bootloader
I (30) boot: compile time 14:33:02
I (31) boot: Enabling RNG early entropy source...
I (36) boot: SPI Speed      : 40MHz
I (41) boot: SPI Mode       : DIO
I (45) boot: SPI Flash Size : 4MB
I (49) boot: Partition Table:
I (52) boot: ## Label            Usage          Type ST Offset   Length
I (59) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (67) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (74) boot:  2 factory          factory app      00 00 00010000 00100000
I (82) boot: End of partition table
I (86) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x0718c ( 29068) map
I (105) esp_image: segment 1: paddr=0x000171b4 vaddr=0x3ffb0000 size=0x02000 (  8192) load
I (109) esp_image: segment 2: paddr=0x000191bc vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at E:/AIOT/esp-idf/components/freertos/xtensa_vectors.S:1778

I (113) esp_image: segment 3: paddr=0x000195c4 vaddr=0x40080400 size=0x06a4c ( 27212) load
I (133) esp_image: segment 4: paddr=0x00020018 vaddr=0x400d0018 size=0x14d9c ( 85404) map
0x400d0018: _stext at ??:?

I (164) esp_image: segment 5: paddr=0x00034dbc vaddr=0x40086e4c size=0x026f4 (  9972) load
0x40086e4c: prvAddCurrentTaskToDelayedList at E:/AIOT/esp-idf/components/freertos/tasks.c:3678

I (174) boot: Loaded app from partition at offset 0x10000
I (174) boot: Disabling RNG early entropy source...
I (175) cpu_start: Pro cpu up.
I (179) cpu_start: Application information:
I (184) cpu_start: Project name:     controlboard
I (189) cpu_start: App version:      99155ec-dirty
I (195) cpu_start: Compile time:     Dec 16 2019 14:32:09
I (201) cpu_start: ELF file SHA256:  e8be6140ce2af965...
I (207) cpu_start: ESP-IDF:          v4.0-dev-1287-gd7e659df2-dirty
I (214) cpu_start: Starting app cpu, entry point is 0x40080f6c
0x40080f6c: call_start_cpu1 at E:/AIOT/esp-idf/components/esp32/cpu_start.c:283

I (206) cpu_start: App cpu up.
I (224) heap_init: Initializing. RAM available for dynamic allocation:
I (231) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (237) heap_init: At 3FFB7068 len 00028F98 (163 KiB): DRAM
I (243) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (250) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (256) heap_init: At 40089540 len 00016AC0 (90 KiB): IRAM
I (262) cpu_start: Pro cpu start user code
I (280) spi_flash: detected chip: generic
I (281) spi_flash: flash io: dio
I (281) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (289) gpio: GPIO[0]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 1| Intr:0
I (299) gpio: GPIO[33]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 1| Pulldown: 1| Intr:0
I [309] app_ir.c(358): IR recieving task started.
I [309] app_ir.c(59): INIT 1
I [319] app_ir.c(66): INIT 1.2
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x400e28eb  PS      : 0x00060a30  A0      : 0x800e3062  A1      : 0x3ffb8cf0
0x400e28eb: rtc_gpio_is_valid_gpio at E:/AIOT/esp-idf/components/driver/include/driver/rtc_io.h:42
 (inlined by) gpio_pullup_dis at E:/AIOT/esp-idf/components/driver/gpio.c:54

A2      : 0x00000010  A3      : 0x3ffae8f0  A4      : 0x00000000  A5      : 0x3ffb8c30
A6      : 0x00000008  A7      : 0x00000004  A8      : 0x3f40682c  A9      : 0x3ff4904c
A10     : 0x00000000  A11     : 0x3ffb8d3c  A12     : 0x3ffb8d30  A13     : 0x3ffae8f0
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x00000010  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000010  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff

ELF file SHA256: e8be6140ce2af965c4dbfa21815c878308ec69e5b09fb1d6b721d7b722610122

Backtrace: 0x400e28e8:0x3ffb8cf0 0x400e305f:0x3ffb8d20 0x400d2748:0x3ffb8d80 0x400d2bb9:0x3ffb8df0 0x400d23d4:0x3ffb8e20 0x400d0c56:0x3ffb8e40 0x40088219:0x3ffb8e60
0x400e28e8: rtc_gpio_is_valid_gpio at E:/AIOT/esp-idf/components/driver/include/driver/rtc_io.h:42
 (inlined by) gpio_pullup_dis at E:/AIOT/esp-idf/components/driver/gpio.c:54

0x400e305f: gpio_config at E:/AIOT/esp-idf/components/driver/gpio.c:291

0x400d2748: ir_gpio_init at E:/AIOT/workspace/ledec4/main/app_ir.c:68

0x400d2bb9: ir_init at E:/AIOT/workspace/ledec4/main/app_ir.c:359

0x400d23d4: app_main at E:/AIOT/workspace/ledec4/main/ledc_example_main.c:116

0x400d0c56: main_task at E:/AIOT/esp-idf/components/esp32/cpu_start.c:549

0x40088219: vPortTaskWrapper at E:/AIOT/esp-idf/components/freertos/port.c:143

Rebooting...
```

`

bitmakk, 2019-12-16:
The backtrace clearly tells you, that this IOconfig is not valid.  The 2 kit has different gpio pin allocation, that may be your problem. 
rtc_gpio_is_valid_gpio is crashing, so not a valid gpio number in the function gpio_pullup_dis.
Saying shortly: you set up a pullup on a gpio, that can't be pulled up. You may check the docs for the correct GPIO pin numbers for pico.

syu-source, 2019-12-16:
Sorry I didn't add info about GPIO number.

In this case the GPIO number I am using is 16. Checked with datasheet, this number should do the work.

From PICO-D4 datasheet:
IO16 25 I/O GPIO16, HS1_DATA4, U2RXD, EMAC_CLK_OUT
From Esp32 datasheet:
GPIO16 25 I/O GPIO16, HS1_DATA4, U2RXD, EMAC_CLK_OUT

syu-source, 2019-12-16:
I tried all combination about poll up/down * en/dis,  all end up crashing on pico and working fine on esp32.

igrr, 2019-12-16:
GPIO16 is used to connect SPI flash in the PICO-D4 SIP. Please check note 2 in section 2.2 of the PICO-D4 datasheet:

> 2. Pins IO16, IO17, CMD, CLK, SD0 and SD1 are used for connecting the embedded flash, and are not recommended for other uses. For details, please see Section 6 Schematics.

Once you have reconfigured the pin as a GPIO, instructions/data fetched from SPI flash by the cache become incorrect, which leads to this crash.

syu-source, 2019-12-17:
Oh, crap.
That should be the problem... I missed this note. 
That explains the IligalInstruction issue. Thanks a lot.
I'm closing this issue.

syu-source, 2019-12-17:
> GPIO16 is used to connect SPI flash in the PICO-D4 SIP. Please check note 2 in section 2.2 of the PICO-D4 datasheet:
> 
> > 1. Pins IO16, IO17, CMD, CLK, SD0 and SD1 are used for connecting the embedded flash, and are not recommended for other uses. For details, please see Section 6 Schematics.
> 
> Once you have reconfigured the pin as a GPIO, instructions/data fetched from SPI flash by the cache become incorrect, which leads to this crash.

Thanks a lot. this "wired" thing suffered me days.

Alvin1Zhang, 2019-12-17:
@syu-source Thanks for the updates. Has your issue been resolved? If it gets resolved, would you please help close the issue? Thanks.

syu-source, 2019-12-17:
Problem solved.

closed 2019-12-17 by syu-source