espressif/esp-idf #4493: OTA via direct WIFI connection, no AP needed?

Bwanna, 2019-12-16:
## Environment
- Development Kit:     ESP32-WROOM-32D
- Operating System:     Windows
## Problem Description:
I'd like to upload my project sketch while away from the home network.  Is there a method to do OTA without needing an AP?  A method for connecting directly to the ESP32 and do a webupload?

AloyseTech, 2019-12-16:
I see 2 solutions : 
1. 
You could create an Soft AP on the ESP32, use your phone to connect to this AP, then a have an app on your smartphone with a webserver serving the binary. It would work out of the box using the ESP32 OTA component. But you have to figure the phone-side server.
This does not require internet connection.
2. 
Use your phone as an AP for the ESP32, which would give internet access to the ESP to fetch an update on a remote server.
This obviously require an internet connection and a remote server.

Bwanna, 2019-12-16:
Ok. I can do that. Don't know why I didn't think if it. Really appreciate the help!

chegewara, 2019-12-16:
> I see 2 solutions :
> 1.
> You could create an Soft AP on the ESP32, use your phone to connect to this AP, then a have an app on your smartphone with a webserver serving the binary. It would work out of [...]

Why do you need web server on smartphone? 
You can have web server on esp32 with simple html code to upload binary file protected with user/password.

Bwanna, 2019-12-16:
> Why do you need web server on smartphone?
> You can have web server on esp32 with simple html code to upload binary file protected with user/password.

Would that be an efficient option to the WebUpload approach in the core OTA?

chegewara, 2019-12-16:
Yes, it is common use case on arduino and here you have some nice project (its old, but good example):
https://github.com/warmcat/lws-esp32-factory

There is of course 4th option, OTA over bluetooth (BLE).

Bwanna, 2019-12-16:
> Yes, it is common use case on arduino and here you have some nice project (its old, but good example):
> https://github.com/warmcat/lws-esp32-factory
> 
> There is of course 4th option, OTA over bluetooth (BLE).

In researching what I needed, OTA over Bluetooth seemed to not work (well?) and I don't have the skill level to figure that out.  Has that changed?  Has it been developed and available to try?  Also, Please let me know if this is the incorrect place to be discussing this topic... I don't want to hijack or be inappropriate.
Thanks!

chegewara, 2019-12-16:
Yes, better place for such questions is forum, esp32.com.

syu-source, 2019-12-17:
Yes, it works for me.
1. esp32 config as AP and web server.
2. connect to it and http post OTA image.
3. write ota image to right address.

key APIs I used:

```
esp_ota_get_next_update_partition()
esp_partition_erase_range()
esp_ota_begin()
while(data recieved)
    esp_ota_write()
_my_validate_image()_
esp_ota_set_boot_partition()
```

Alvin1Zhang, 2019-12-17:
@chegewara Thanks for the notes. @Bwanna Thanks for reporting, for how questions and third party questions, please help submit a post on our forum at https://esp32.com. Will close this ticket now. Thanks.

closed 2019-12-17 by Alvin1Zhang