espressif/esp-idf #4495: SD Card File Can't open

caleb221, 2019-12-16:
Hello!
I am unable to write a jpeg image to the SD Card after doing some Image processing ( deep learning using the ESP-FACE library)
--> The code follows the SD Card example provided by ESP-idf
--> Testing the code BEFORE the image processing works without issue (image saves as expected)
--> The issue only happens AFTER the processing.
I am unsure if I am doing something wrong, but testing it BEFORE processing works just fine
I will put The SD card saving Code and after the function that calls it

 
void saveImage(camera_fb_t *in,int id)
{
   size_t _jpg_buf_len = 0;
   uint8_t * _jpg_buf = NULL;
//camera_fb_t * out = NULL;

int size = in->len;

 sdmmc_host_t host = SDMMC_HOST_DEFAULT();
 sdmmc_slot_config_t slot_config = SDMMC_SLOT_CONFIG_DEFAULT();
 gpio_set_pull_mode(15, GPIO_PULLUP_ONLY);   // CMD, needed in 4- and 1- line modes
 gpio_set_pull_mode(2, GPIO_PULLUP_ONLY);    // D0, needed in 4- and 1-line modes
 gpio_set_pull_mode(4, GPIO_PULLUP_ONLY);    // D1, needed in 4-line mode only
 gpio_set_pull_mode(12, GPIO_PULLUP_ONLY);   // D2, needed in 4-line mode only
 gpio_set_pull_mode(13, GPIO_PULLUP_ONLY);   // D3, needed in 4- and 1-line modes

//gpio_pulldown_dis(13);
//gpio_pullup_en(13);

esp_vfs_fat_sdmmc_mount_config_t mount_config = {
        .format_if_mount_failed = true,//false,
        .max_files = 5,
        .allocation_unit_size = size//img->w*img->c  // 16 * 1024
    };
    sdmmc_card_t* card;
    esp_err_t ret = esp_vfs_fat_sdmmc_mount("/sdcard", &host, &slot_config, &mount_config, &card);

    if (ret != ESP_OK) {
        if (ret == ESP_FAIL) {
            printf( "[SD KA] Failed to mount filesystem. "
                "If you want the card to be formatted, set format_if_mount_failed = true.");
        } else {
            printf( "[SD KA] Failed to initialize the card (%s). "
                "Make sure SD card lines have pull-up resistors in place.", esp_err_to_name(ret));
        }
        return;
    }
    //sdmmc_card_print_info(stdout, card);
    printf("\n[SD KA] OPENING FILE\n");
    	FILE* f = fopen("/sdcard/fack.jpg", "w");
    	if (f == NULL)
       	{
        	printf( "[SD KA] Failed to open file for writing");
        	return;
	}
	printf("\n[SD KA] FILE OPENED ...size: %i \n",size);
       //=====================
      //write the image    
  size_t fb_len = 0;
        if (in->format == PIXFORMAT_JPEG)
        {   printf("\nJPEG FORMAT! LETS TRY WRITING\n");
            fb_len = in->len;
	    printf("SIZE: %i inside: %i  \n",size,in->len);
            fwrite((const char *)in->buf,sizeof(char *),in->len,f);
	    //use size instead of in->len if not good
        }
        else
        {
          printf("\n WRONG FORMAT!\n");
	  fclose(f);
	   esp_vfs_fat_sdmmc_unmount();
	  return;
        };
    //fprintf(f, "Hello %s!\n", card->cid.name);
    fclose(f);
    printf("\nFile written\n");
 esp_vfs_fat_sdmmc_unmount();
}

*************CALLED BY**************
void analyzeImage()
{
  size_t _jpg_buf_len = 0;
  uint8_t * _jpg_buf = NULL;
  camera_fb_t * pic = NULL;  
  esp_err_t err = ESP_OK;
  // Take Picture with Camera

vTaskDelay(1000/portMAX_DELAY);
  pic = esp_camera_fb_get();  
    if(!pic)
    {
      printf("ERR TAKING PICTURE");
       return;//NULL;
    }

    if(pic->format == PIXFORMAT_JPEG)
    {
        jpg_chunking_t jchunk = {0};
        err = frame2jpg_cb(pic, 80/*quality*/, jpg_encode_stream, &jchunk)?ESP_OK:ESP_FAIL;
        _jpg_buf_len = pic->len;
        _jpg_buf = pic->buf;
    	
	//saveImage(pic,0);
	
	if(err == ESP_FAIL)
        {
          printf("ERROR IN JPEG: ");
	  return;//NULL;
	}
    }
   //allocate array

   dl_matrix3du_t *image_matrix = dl_matrix3du_alloc(1, pic->width, pic->height, 3);
   if(!image_matrix)
   {
     dl_matrix3du_free(image_matrix);
     esp_camera_fb_return(pic);
     printf("image matrix allocation failed");
     return;//NULL;
   }

  if(!fmt2rgb888(pic->buf,pic->len,pic->format,image_matrix->item))
  {
    dl_matrix3du_free(image_matrix);
    esp_camera_fb_return(pic);
    printf("FORMAT 2 RGB 888 FAILED");
    return;// NULL;
  }
initConf(1);
initOut(1,1,1,1);
float minLeaf=80; //120;//80;
float pyramid=.707;//.5;//.707;

//pnet forward -->
box_array_t *pBoxes;
pBoxes=pnet_forward( image_matrix,minLeaf,pyramid,currentConfig );

if(pBoxes != NULL )//|| pBoxes->box[0].box_p[0]< 0  )
{
	if( pBoxes->box[0].box_p[0]> 0 && pBoxes->box[0].box_p[3] > 0 )
	{
	        setOutput(NULL,3);
	        initConf(2);//init Rnet config
		box_array_t *rBoxes;
		rBoxes=rnet_forward(image_matrix, pBoxes,currentConfig);
	       if(rBoxes!=NULL)
	       {

      if(rBoxes->box[0].box_p[0]< 50000000 && rBoxes->box[0].box_p[1]< 50000000 )
	          {
			  printf("\n[GET IMG] VALID RNET BOX!!!!!\n ");
	  		  draw_face_boxes(image_matrix,rBoxes,0);      
if(!fmt2jpg(image_matrix->item, pic->width*pic->height*3, pic->width, pic->height, PIXFORMAT_RGB888, 80, &_jpg_buf, &_jpg_buf_len)){

		printf("BACK 2 JPG FAIL.");
         //=============================	
		}
   	else
    	{ 
   	  saveImage(pic,0);
	}
}
}

   }
 }
 	   //free the camera frame for next picture
  	  esp_camera_fb_return(pic);
	  pic=NULL;
	  dl_matrix3du_free(image_matrix);
//return image_matrix; 
}

igrr, 2019-12-17:
Can you please explain this part of code?
```c++
.allocation_unit_size = size//img->wimg->c // 16 * 1024
```
what is your intention when setting the filesystem allocation unit size to the framebuffer size?

As for the troubleshooting, could you please enable verbose log output in `menuconfig`, under Component config, Log output, and attach the complete log you get? Also please check and print out the amount of heap memory remaining before you start writing the file to the SD card.

caleb221, 2019-12-19:
Hey, my intention is to write a jpeg image to the file, i figured i only needed to allocate as much memory as the image would take, I changed it back to the original 16*1024 in the example for this output 

The free heap after opening the drive and before opening the file is: 3865700 

[SD] size: 3988
D (63739) sdmmc_periph: peripheral version 5342270a, hardware config 03c44c83
V (63739) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): checking args
V (63749) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): Args okay. Resulting flags 0xE
D (63759) intr_alloc: Connected src 37 to int 3 (cpu 0)
I (63759) gpio: GPIO[13]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
D (63769) sdmmc_periph: slot=1 host_div=10 card_div=20 freq=400kHz
D (63779) sdmmc_periph: slot=1 width=1
V (63779) sdmmc_cmd: sending cmd slot=1 op=52 arg=80000c08 flags=1c00 data=0x0 blklen=0 datalen=0 timeout=1000
V (63789) sdmmc_req: make_hw_cmd: opcode=52, rexp=1, crc=1, auto_stop=0
V (63799) sdmmc_req: sdmmc_handle_event: event 00000100 00000000, unhandled 00000000 00000000
V (63809) sdmmc_req: process_events: state=SENDING_CMD evt=100 dma=0
D (63819) sdmmc_req: process_command_response: error 0x107  (status=00000100)
V (63819) sdmmc_req: process_events state=SENDING_CMD next_state=SENDING_CMD
V (63829) sdmmc_req: sdmmc_handle_event: events unhandled: 00000000 00000000
V (63839) sdmmc_req: sdmmc_handle_event: event 00000004 00000000, unhandled 00000000 00000000
V (63849) sdmmc_req: process_events: state=SENDING_CMD evt=4 dma=0
V (63849) sdmmc_req: process_events state=SENDING_CMD next_state=IDLE
V (63859) sdmmc_req: process_events state=IDLE next_state=IDLE
V (63859) sdmmc_req: sdmmc_handle_event: events unhandled: 00000000 00000000
V (63869) sdmmc_cmd: cmd response 00000000 00000000 00000000 00000000 err=0x107 state=0
V (63879) sdmmc_io: sdmmc_io_rw_direct: sdmmc_send_cmd returned 0x107
V (63889) sdmmc_cmd: sending cmd slot=1 op=0 arg=0 flags=20 data=0x0 blklen=0 datalen=0 timeout=1000
V (63889) sdmmc_req: make_hw_cmd: opcode=0, rexp=0, crc=0, auto_stop=0
V (63899) sdmmc_req: sdmmc_handle_event: event 00000004 00000000, unhandled 00000000 00000000
V (63909) sdmmc_req: process_events: state=SENDING_CMD evt=4 dma=0
V (63919) sdmmc_req: process_events state=SENDING_CMD next_state=IDLE
V (63919) sdmmc_req: process_events state=IDLE next_state=IDLE
V (63929) sdmmc_req: sdmmc_handle_event: events unhandled: 00000000 00000000
V (63939) sdmmc_cmd: cmd response 00000000 00000000 00000000 00000000 err=0x0 state=0
V (63959) sdmmc_cmd: sending cmd slot=1 op=8 arg=1aa flags=1c30 data=0x0 blklen=0 datalen=0 timeout=1000
V (63959) sdmmc_req: make_hw_cmd: opcode=8, rexp=1, crc=1, auto_stop=0
V (63959) sdmmc_req: sdmmc_handle_event: event 00000004 00000000, unhandled 00000000 00000000
V (63969) sdmmc_req: process_events: state=SENDING_CMD evt=4 dma=0
V (63979) sdmmc_req: process_events state=SENDING_CMD next_state=IDLE
V (63979) sdmmc_req: process_events state=IDLE next_state=IDLE
V (63989) sdmmc_req: sdmmc_handle_event: events unhandled: 00000000 00000000
V (63999) sdmmc_cmd: cmd response 000001aa 00000000 00000000 00000000 err=0x0 state=0
D (63999) sdmmc_sd: SDHC/SDXC card
V (64009) sdmmc_cmd: sending cmd slot=1 op=5 arg=0 flags=1030 data=0x0 blklen=0 datalen=0 timeout=1000
V (64019) sdmmc_req: make_hw_cmd: opcode=5, rexp=1, crc=0, auto_stop=0
V (64019) sdmmc_req: sdmmc_handle_event: event 00000100 00000000, unhandled 00000000 00000000
V (64029) sdmmc_req: process_events: state=SENDING_CMD evt=100 dma=0
D (64039) sdmmc_req: process_command_response: error 0x107  (status=00000100)
V (64049) sdmmc_req: process_events state=SENDING_CMD next_state=SENDING_CMD
V (64049) sdmmc_req: sdmmc_handle_event: events unhandled: 00000000 00000000
V (64059) sdmmc_req: sdmmc_handle_event: event 00000004 00000000, unhandled 00000000 00000000
V (64069) sdmmc_req: process_events: state=SENDING_CMD evt=4 dma=0
V (64079) sdmmc_req: process_events state=SENDING_CMD next_state=IDLE
V (64079) sdmmc_req: process_events state=IDLE next_state=IDLE
V (64089) sdmmc_req: sdmmc_handle_event: events unhandled: 00000000 00000000
V (64089) sdmmc_cmd: cmd response 000001aa 00000000 00000000 00000000 err=0x107 state=0
D (64099) sdmmc_io: sdmmc_init_io: io_send_op_cond (1) returned 0x107; not IO card
V (64109) sdmmc_cmd: sending cmd slot=1 op=55 arg=0 flags=1c00 data=0x0 blklen=0 datalen=0 timeout=1000
V (64119) sdmmc_req: make_hw_cmd: opcode=55, rexp=1, crc=1, auto_stop=0
V (64129) sdmmc_req: sdmmc_handle_event: event 00000004 00000000, unhandled 00000000 00000000
V (64139) sdmmc_req: process_events: state=SENDING_CMD evt=4 dma=0
V (64139) sdmmc_req: process_events state=SENDING_CMD next_state=IDLE
V (64149) sdmmc_req: process_events state=IDLE next_state=IDLE
V (64149) sdmmc_req: sdmmc_handle_event: events unhandled: 00000000 00000000
V (64159) sdmmc_cmd: cmd response 00000120 00000000 00000000 00000000 err=0x0 state=0
V (64169) sdmmc_cmd: sending cmd slot=1 op=41 arg=40ff8000 flags=1030 data=0x0 blklen=0 datalen=0 timeout=1000
V (64179) sdmmc_req: make_hw_cmd: opcode=41, rexp=1, crc=0, auto_stop=0
V (64189) sdmmc_req: sdmmc_handle_event: event 00000004 00000000, unhandled 00000000 00000000
V (64189) sdmmc_req: process_events: state=SENDING_CMD evt=4 dma=0
V (64199) sdmmc_req: process_events state=SENDING_CMD next_state=IDLE
[... 177 lines cut ...]

[SD] OPENING FILE
V (66439) vfs_fat: vfs_fat_open: path="/meh.jpg", flags=601, mode=1b6
[... cut ...]
V (66549) sdmmc_req: make_hw_cmd: opcode=13, rexp=1, crc=1, auto_stop=0
V (66559) sdmmc_req: sdmmc_handle_event: event 00000004 00000000, unhandled 00000000 00000000
V (66569) sdmmc_req: process_events: state=SENDING_CMD evt=4 dma=0
V (66569) sdmmc_req: process_events state=SENDING_CMD next_state=IDLE
V (66579) sdmmc_req: process_events state=IDLE next_state=IDLE
V (66589) sdmmc_req: sdmmc_handle_event: events unhandled: 00000000 00000000
V (66589) sdmmc_cmd: cmd response 00000900 00000000 00000000 00000000 err=0x0 state=4

igrr, 2019-12-19:
Thank you, one more request: please enable the following two options in `sdkconfig`:
```
CONFIG_ESP_TASK_WDT=y
CONFIG_ESP_TASK_WDT_PANIC=y
```
These should result in a panic when the task hangs up, along with a backtrace which will be decoded by IDF monitor.
If necessary, you may adjust the task watchdog timeout using `CONFIG_ESP_TASK_WDT_TIMEOUT_S` option (it is 5 seconds by default).

Also I haven't realized at first that you are using PSRAM. Instead of checking the total available memory size, could you print out just the size of the internal free memory? This would be
```
heap_caps_get_free_size(MALLOC_CAP_8BIT | MALLOC_CAP_INTERNAL)
```

caleb221, 2019-12-19:
Ok cool, Heres the dump and the free memory:
free heap memory:
[SD] FREE HEAP BEFORE OPENING FILE: 3865700 
free 8 bit
[SD] FREE CAP 8 BIT: 3898416
Free internal: 
[SD] FREE INTERNAL: 131900 
um, also there was no panic after a few minutes of waiting

[SD] size: 4090
D (63739) sdmmc_periph: peripheral version 5342270a, hardware config 03c44c83
V (63739) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): checking args
V (63749) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): Args okay. Resulting flags 0xE
D (63759) intr_alloc: Connected src 37 to int 3 (cpu 0)
I (63759) gpio: GPIO[13]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
D (63769) sdmmc_periph: slot=1 host_div=10 card_div=20 freq=400kHz
D (63779) sdmmc_periph: slot=1 width=1
V (63779) sdmmc_cmd: sending cmd slot=1 op=52 arg=80000c08 flags=1c00 data=0x0 blklen=0 datalen=0 timeout=1000
V (63789) sdmmc_req: make_hw_cmd: opcode=52, rexp=1, crc=1, auto_stop=0
V (63799) sdmmc_req: sdmmc_handle_event: event 00000100 00000000, unhandled 00000000 00000000
V (63809) sdmmc_req: process_events: state=SENDING_CMD evt=100 dma=0
D (63819) sdmmc_req: process_command_response: error 0x107  (status=00000100)
V (63819) sdmmc_req: process_events state=SENDING_CMD next_state=SENDING_CMD
V (63829) sdmmc_req: sdmmc_handle_event: events unhandled: 00000000 00000000
V (63839) sdmmc_req: sdmmc_handle_event: event 00000004 00000000, unhandled 00000000 00000000
V (63849) sdmmc_req: process_events: state=SENDING_CMD evt=4 dma=0
V (63849) sdmmc_req: process_events state=SENDING_CMD next_state=IDLE
V (63859) sdmmc_req: process_events state=IDLE next_state=IDLE
V (63859) sdmmc_req: sdmmc_handle_event: events unhandled: 00000000 00000000
V (63869) sdmmc_cmd: cmd response 00000000 00000000 00000000 00000000 err=0x107 state=0
V (63879) sdmmc_io: sdmmc_io_rw_direct: sdmmc_send_cmd returned 0x107
V (63889) sdmmc_cmd: sending cmd slot=1 op=0 arg=0 flags=20 data=0x0 blklen=0 datalen=0 timeout=1000
V (63889) sdmmc_req: make_hw_cmd: opcode=0, rexp=0, crc=0, auto_stop=0
V (63899) sdmmc_req: sdmmc_handle_event: event 00000004 00000000, unhandled 00000000 00000000
V (63909) sdmmc_req: process_events: state=SENDING_CMD evt=4 dma=0
V (63919) sdmmc_req: process_events state=SENDING_CMD next_state=IDLE
V (63919) sdmmc_req: process_events state=IDLE next_state=IDLE
V (63929) sdmmc_req: sdmmc_handle_event: events unhandled: 00000000 00000000
V (63939) sdmmc_cmd: cmd response 00000000 00000000 00000000 00000000 err=0x0 state=0
V (63959) sdmmc_cmd: sending cmd slot=1 op=8 arg=1aa flags=1c30 data=0x0 blklen=0 datalen=0 timeout=1000
V (63959) sdmmc_req: make_hw_cmd: opcode=8, rexp=1, crc=1, auto_stop=0
V (63959) sdmmc_req: sdmmc_handle_event: event 00000004 00000000, unhandled 00000000 00000000
V (63969) sdmmc_req: process_events: state=SENDING_CMD evt=4 dma=0
V (63979) sdmmc_req: process_events state=SENDING_CMD next_state=IDLE
V (63979) sdmmc_req: process_events state=IDLE next_state=IDLE
V (63989) sdmmc_req: sdmmc_handle_event: events unhandled: 00000000 00000000
V (63999) sdmmc_cmd: cmd response 000001aa 00000000 00000000 00000000 err=0x0 state=0
D (63999) sdmmc_sd: SDHC/SDXC card
V (64009) sdmmc_cmd: sending cmd slot=1 op=5 arg=0 flags=1030 data=0x0 blklen=0 datalen=0 timeout=1000
V (64019) sdmmc_req: make_hw_cmd: opcode=5, rexp=1, crc=0, auto_stop=0
V (64019) sdmmc_req: sdmmc_handle_event: event 00000100 00000000, unhandled 00000000 00000000
V (64029) sdmmc_req: process_events: state=SENDING_CMD evt=100 dma=0
D (64039) sdmmc_req: process_command_response: error 0x107  (status=00000100)
V (64049) sdmmc_req: process_events state=SENDING_CMD next_state=SENDING_CMD
V (64049) sdmmc_req: sdmmc_handle_event: events unhandled: 00000000 00000000
V (64059) sdmmc_req: sdmmc_handle_event: event 00000004 00000000, unhandled 00000000 00000000
V (64069) sdmmc_req: process_events: state=SENDING_CMD evt=4 dma=0
V (64079) sdmmc_req: process_events state=SENDING_CMD next_state=IDLE
V (64079) sdmmc_req: process_events state=IDLE next_state=IDLE
V (64089) sdmmc_req: sdmmc_handle_event: events unhandled: 00000000 00000000
V (64089) sdmmc_cmd: cmd response 000001aa 00000000 00000000 00000000 err=0x107 state=0
D (64099) sdmmc_io: sdmmc_init_io: io_send_op_cond (1) returned 0x107; not IO card
V (64109) sdmmc_cmd: sending cmd slot=1 op=55 arg=0 flags=1c00 data=0x0 blklen=0 datalen=0 timeout=1000
V (64119) sdmmc_req: make_hw_cmd: opcode=55, rexp=1, crc=1, auto_stop=0
V (64129) sdmmc_req: sdmmc_handle_event: event 00000004 00000000, unhandled 00000000 00000000
V (64139) sdmmc_req: process_events: state=SENDING_CMD evt=4 dma=0
V (64139) sdmmc_req: process_events state=SENDING_CMD next_state=IDLE
V (64149) sdmmc_req: process_events state=IDLE next_state=IDLE
V (64149) sdmmc_req: sdmmc_handle_event: events unhandled: 00000000 00000000
V (64159) sdmmc_cmd: cmd response 00000120 00000000 00000000 00000000 err=0x0 state=0
[... 184 lines cut ...]

[SD] OPENING FILE
V (66439) vfs_fat: vfs_fat_open: path="/meh.jpg", flags=601, mode=1b6
V (66449) sdmmc_cmd: sending cmd slot=1 op=17 arg=20b4 flags=1c50 data=0x3ffba51c blklen=512 datalen=512 timeout=1000
V (66459) sdmmc_req: make_hw_cmd: opcode=17, rexp=1, crc=1, auto_stop=0
V (66459) sdmmc_req: fill 4 desc=0 rem=0 next=1 last=1 sz=512
V (66469) sdmmc_req: sdmmc_handle_event: event 00000004 00000000, unhandled 00000000 00000000
V (66479) sdmmc_req: process_events: state=SENDING_CMD evt=4 dma=0
[... cut ...]
V (66559) sdmmc_req: make_hw_cmd: opcode=13, rexp=1, crc=1, auto_stop=0
V (66569) sdmmc_req: sdmmc_handle_event: event 00000004 00000000, unhandled 00000000 00000000
V (66569) sdmmc_req: process_events: state=SENDING_CMD evt=4 dma=0
V (66579) sdmmc_req: process_events state=SENDING_CMD next_state=IDLE
V (66589) sdmmc_req: process_events state=IDLE next_state=IDLE
V (66589) sdmmc_req: sdmmc_handle_event: events unhandled: 00000000 00000000
V (66599) sdmmc_cmd: cmd response 00000900 00000000 00000000 00000000 err=0x0 state=4

2dark4u, 2020-03-22:
D (63819) sdmmc_req: process_command_response: error 0x107 (status=00000100)
As I see from: GPIO_PULLUP_ONLY You using internal pullup resistors. Try to turn them off, normally there is external pullup resistors ~10K on SPI data pins, when You using internal resistors with external, resistance will drops to 5K, since there is two pullup resistors on each SPI pin.
 
Also You can try to change SPI bus pullup resistors to 15K (it helps sometimes)

Mliron, 2021-12-15:
Hi Caleb.
I have tried to reproduce your issue without any luck. The code you have provided does not work on its own, so I filled the blanks with code from [ArduinoCore CameraWebServer](https://github.com/espressif/arduino-esp32/tree/master/libraries/ESP32/examples/Camera/CameraWebServer) example, since it looks like you drew inspiration from that.
If that is the case then there is no reason why changing the content of an already allocated buffer should affect the opening of a file.
So could you provide more information about the issue? Like what enviroment you are using, what libraries do you include, what hardware do you use, etc?
Is there anything connected on the pins that you use for the SD card?
Do you use integrated SD module or external?
Also could you please provide the definitions for all custom functions needed to run your code?
You have had a copule of timeouts, is there a possibility of a loose connection, high resistance or long wires on the data lines?
If you are using an older IDF version, try downloading latest updates, maybe the issue is already solved.

closed 2022-02-11 by Mliron