espressif/esp-idf #4496: idf.py menuconfig not working after last update

pilnikov, 2019-12-16:
Z:\examples\Ka_mod>idf.py menuconfig
Executing action: menuconfig
Traceback (most recent call last):
  File "Z:\esp-idf\tools\idf.py", line 764, in <module>
    main()
  File "Z:\esp-idf\tools\idf.py", line 682, in main
    cli(sys.argv[1:], prog_name=PROG)
  File "z:\espressif\python_env\idf4.1_py3.7_env\lib\site-packages\click\core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "z:\espressif\python_env\idf4.1_py3.7_env\lib\site-packages\click\core.py", line 717, in main
    rv = self.invoke(ctx)
  File "z:\espressif\python_env\idf4.1_py3.7_env\lib\site-packages\click\core.py", line 1164, in invoke
    return _process_result(rv)
  File "z:\espressif\python_env\idf4.1_py3.7_env\lib\site-packages\click\core.py", line 1102, in _process_result
    **ctx.params)
  File "z:\espressif\python_env\idf4.1_py3.7_env\lib\site-packages\click\core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "Z:\esp-idf\tools\idf.py", line 614, in execute_tasks
    task.run(ctx, global_args, task.action_args)
  File "Z:\esp-idf\tools\idf.py", line 207, in run
    self.callback(self.name, context, global_args, **action_args)
  File "Z:\esp-idf\tools\idf_py_actions\core_ext.py", line 40, in menuconfig
    os.environ['MENUCONFIG_STYLE'] = style.encode(sys.getfilesystemencoding() or 'utf-8')
  File "z:\espressif\python_env\idf4.1_py3.7_env\lib\os.py", line 683, in __setitem__
    value = self.encodevalue(value)
  File "z:\espressif\python_env\idf4.1_py3.7_env\lib\os.py", line 739, in check_str
    raise TypeError("str expected, not %s" % type(value).__name__)
TypeError: str expected, not bytes

Z:\examples\Ka_mod>

works like this only 

Z:\examples\Ka_mod>cd build

Z:\examples\Ka_mod\build>ninja menuconfig
[0/1] cmd.exe /C "cd /D Z:\examples\Ka_mod\build && z:\esp...TARGET=esp32 --output config Z:/examples/Ka_mod/sdkconfig"
Loading defaults file Z:/examples/Ka_mod/sdkconfig.defaults...
Loaded configuration 'Z:/examples/Ka_mod/sdkconfig'
No changes to save (for 'Z:/examples/Ka_mod/sdkconfig')
Loading defaults file Z:/examples/Ka_mod/sdkconfig.defaults...

Z:\examples\Ka_mod\build>

$ git describe --tags
v4.1-dev-1533-g9dbb7526c

dobairoland, 2019-12-16:
We are sorry for the inconvenience. This is already fixed but it hasn't made it to Github yet. A temporary workaround is if you switch to Python 2 or edit `Z:\esp-idf\tools\idf_py_actions\core_ext.py`, line 40, to `os.environ['MENUCONFIG_STYLE'] = style`. The final fix is a better one and will ensure compatibility between Python 2 and 3.

For future reference, commit 7ab43bffde268a192bca1aea2bbb4e3fcc7fe595 fixes the issue.

projectgus, 2019-12-17:
Hi @pilnikov ,

The fix by @dobairoland has just pushed to GitHub - 7ab43bffde268a192bca1aea2bbb4e3fcc7fe595 -  so if you update to the latest master branch then menuconfig should work with Python 3 again.

Sorry again for the inconvenience.

pilnikov, 2019-12-17:
All working fine. Tnx.

closed 2019-12-17 by pilnikov