espressif/esp-idf #4497: Missing extern "C" in "components/esp_wifi/include/esp_wifi_default.h"

svenbieg, 2019-12-16:
I am calling esp_netif_create_default_wifi_ap and esp_netif_create_default_wifi_sta from a cpp-file. Usually there is an #ifdef __cplusplus extern "C" statement at the beginning of the header.

svenbieg, 2019-12-18:
Same as IDFGH-2241

anderstogern, 2019-12-20:
It seems this issue is still around!

With latest master the `extern C` deifnition is still missing. Adding it manually fixes the problem and compilation completes successfully.

#4389 focuses on the order of the initializers and refers to this as a separate issue, so I think that this issue should be reopened.

closed 2020-01-07 by espressif-bot

Alvin1Zhang, 2020-01-08:
@svenbieg @anderstogern Thanks for reporting, please help check whether the issue still happens at your side, thanks.

svenbieg, 2020-01-09:
It's working now, thank You!

Alvin1Zhang, 2020-01-09:
@svenbieg Thanks for the updates.