espressif/esp-idf #4501: "Espressif IoT Development Framework Configuration" not loaded, exits with type error

peterstadie, 2019-12-17:
## Environment

- Development Kit: ESP32-DevKitC
- Kit version DevKitC: v4
- Module or chip used:  ESP32-WROOM-32D
- IDF version: v4.1-dev-1533-g9dbb7526c
- Build System:  CMake
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
- Operating System: Windows
- Power Supply:  USB

## Problem Description
I am in the process of setting up the development environment on a Windows 10 machine. I followed https://docs.espressif.com/projects/esp-idf/en/latest/get-started/
However, I did not use https://dl.espressif.com/dl/esp-idf-tools-setup-2.2.exe from here https://docs.espressif.com/projects/esp-idf/en/latest/get-started/windows-setup.html.
Instead I followed this https://docs.espressif.com/projects/esp-idf/en/latest/get-started/windows-setup-scratch.html to setup from scratch.

The following two commands worked
```
cd %userprofile%\esp\esp-idf
install.bat
```
and
`%userprofile%\esp\esp-idf\export.bat`

I do NOT have IDF_TOOLS_PATH in my environment variables. This seems only to be installed when using the esp-idf-tools-setup-2.2.exe.

I followed the setup guide and copied the hello_world directory to %userprofile%\esp\

I executed 
`cd %userprofile%\esp\hello_world
idf.py menuconfig`

### Expected Behavior
According to the getting started guide, the "Espressif IoT Development Framework Configuration" should appear now. Instead I got this:

### Actual Behavior
```
  File "C:\Users\pstadie\.espressif\python_env\idf4.1_py3.8_env\lib\os.py", line 734, in check_str
    raise TypeError("str expected, not %s" % type(value).__name__)
TypeError: str expected, not bytes`

```
Config Tool is not loaded

### Steps to repropduce
don't know

### Code to reproduce this issue
don't know

## Debug Logs

```
`C:\Users\pstadie\esp\hello_world>idf.py menuconfig
Executing action: menuconfig
Traceback (most recent call last):
  File "C:\Users\pstadie\esp\esp-idf\tools\idf.py", line 764, in <module>
    main()
  File "C:\Users\pstadie\esp\esp-idf\tools\idf.py", line 682, in main
    cli(sys.argv[1:], prog_name=PROG)
  File "C:\Users\pstadie\.espressif\python_env\idf4.1_py3.8_env\lib\site-packages\click\core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "C:\Users\pstadie\.espressif\python_env\idf4.1_py3.8_env\lib\site-packages\click\core.py", line 717, in main
    rv = self.invoke(ctx)
  File "C:\Users\pstadie\.espressif\python_env\idf4.1_py3.8_env\lib\site-packages\click\core.py", line 1164, in invoke
    return _process_result(rv)
  File "C:\Users\pstadie\.espressif\python_env\idf4.1_py3.8_env\lib\site-packages\click\core.py", line 1101, in _process_result
    value = ctx.invoke(self.result_callback, value,
  File "C:\Users\pstadie\.espressif\python_env\idf4.1_py3.8_env\lib\site-packages\click\core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "C:\Users\pstadie\esp\esp-idf\tools\idf.py", line 614, in execute_tasks
    task.run(ctx, global_args, task.action_args)
  File "C:\Users\pstadie\esp\esp-idf\tools\idf.py", line 207, in run
    self.callback(self.name, context, global_args, **action_args)
  File "C:\Users\pstadie\esp\esp-idf\tools\idf_py_actions\core_ext.py", line 40, in menuconfig
    os.environ['MENUCONFIG_STYLE'] = style.encode(sys.getfilesystemencoding() or 'utf-8')
  File "C:\Users\pstadie\.espressif\python_env\idf4.1_py3.8_env\lib\os.py", line 678, in __setitem__
    value = self.encodevalue(value)
  File "C:\Users\pstadie\.espressif\python_env\idf4.1_py3.8_env\lib\os.py", line 734, in check_str
    raise TypeError("str expected, not %s" % type(value).__name__)
TypeError: str expected, not bytes`
```

## this is my PATH environment var
`PATH=C:\Users\pstadie\.espressif\tools\xtensa-esp32-elf\esp-2019r2-8.2.0\xtensa-esp32-elf\bin;C:\Users\pstadie\.espressif\tools\xtensa-esp32s2-elf\esp-2019r2-8.2.0\xtensa-esp32s2-elf\bin;C:\Users\pstadie\.espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin;C:\Users\pstadie\.espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin;C:\Users\pstadie\.espressif\tools\cmake\3.13.4\bin;C:\Users\pstadie\.espressif\tools\openocd-esp32\v0.10.0-esp32-20191114\openocd-esp32\bin;C:\Users\pstadie\.espressif\tools\ninja\1.9.0\;C:\Users\pstadie\.espressif\tools\idf-exe\1.0.1\;C:\Users\pstadie\.espressif\tools\ccache\3.7\;C:\Users\pstadie\.espressif\python_env\idf4.1_py3.8_env\Scripts;C:\Users\pstadie\esp\esp-idf\tools;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;C:\Program Files (x86)\Common Files\Oracle\Java\javapath;C:\Program Files\PuTTY\;C:\Program Files (x86)\PuTTY\;C:\Users\pstadie\AppData\Local\Programs\Python\Python38-32\Scripts\;C:\Users\pstadie\AppData\Local\Programs\Python\Python38-32\;C:\Users\pstadie\AppData\Local\Programs\Python\Launcher\;C:\Users\pstadie\AppData\Local\Microsoft\WindowsApps;C:\Users\pstadie\AppData\Roaming\Python\Python38\Scripts;C:\Users\pstadie\AppData\Local\Programs\Git\cmd;C:\Users\pstadie\AppData\Local\Programs\Microsoft VS Code\bin;C:\mingw-w64;`

dobairoland, 2019-12-17:
Please rebase to the latest master branch. Your issue seems to be a duplicate of https://github.com/espressif/esp-idf/issues/4496 fixed in https://github.com/espressif/esp-idf/commit/7ab43bffde268a192bca1aea2bbb4e3fcc7fe595.

peterstadie, 2019-12-18:
I went back to the stable ESP-IDF version. Problem solved.

closed 2019-12-18 by peterstadie