espressif/esp-idf #4503: undefined reference during 'build' after updating esp-idf

arnkmar, 2019-12-17:
## Environment

- Development Kit:     none
- Module or chip used:  ESP32-WROOM-32
- IDF version : v4.1-dev-1543-g431066f16
- Build System: CMake
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
- Operating System:  Windows
- Power Supply: external 5V

## Problem Description

After updating esp-idf to the latest version, I am no longer able to compile my project due to ```undefined reference to '<function name>'``` errors.

### Expected Behavior
Project should build without errors as before.

### Actual Behavior
Build fails due to ```undefined reference to '<function name>' ``` errors.

## Build output

```
C:\Users\arnkmar\projects\myProject>idf.py build
Executing action: all (aliases: build)
Running ninja in directory c:\users\arnkmar\projects\myProject\build
Executing "ninja all"...
[1/5] Performing build step for 'bootloader'
ninja: no work to do.
[2/3] Linking CXX executable empty.elf
FAILED: empty.elf
cmd.exe /C "cd . && C:\Installations\.espressif-master-1811\tools\xtensa-esp32-elf\esp-2019r2-8.2.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-g++.exe  -mlongcalls -Wno-frame-address  -nostdlib @CMakeFiles\empty.elf.rsp  -o empty.elf  && cd ."
c:/installations/.espressif-master-1811/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: c:/installations/.espressif-master-1811/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/lib/no-rtti\libstdc++.a(eh_personality.o):(.literal._ZL21base_of_encoded_valuehP15_Unwind_Context+0x0): undefined reference to `_Unwind_GetTextRelBase'
c:/installations/.espressif-master-1811/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: c:/installations/.espressif-master-1811/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/lib/no-rtti\libstdc++.a(eh_personality.o):(.literal._ZL21base_of_encoded_valuehP15_Unwind_Context+0x4): undefined reference to `_Unwind_GetDataRelBase'
c:/installations/.espressif-master-1811/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: c:/installations/.espressif-master-1811/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/lib/no-rtti\libstdc++.a(eh_personality.o):(.literal._ZL21base_of_encoded_valuehP15_Unwind_Context+0x8): undefined reference to `_Unwind_GetRegionStart'
c:/installations/.espressif-master-1811/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: c:/installations/.espressif-master-1811/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/lib/no-rtti\libstdc++.a(eh_personality.o):(.literal.__gxx_personality_v0+0x8): undefined reference to `_Unwind_GetLanguageSpecificData'
c:/installations/.espressif-master-1811/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: c:/installations/.espressif-master-1811/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/lib/no-rtti\libstdc++.a(eh_personality.o):(.literal.__gxx_personality_v0+0xc): undefined reference to `_Unwind_GetIPInfo'
c:/installations/.espressif-master-1811/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: c:/installations/.espressif-master-1811/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/lib/no-rtti\libstdc++.a(eh_personality.o):(.literal.__gxx_personality_v0+0x10): undefined reference to `_Unwind_SetGR'
c:/installations/.espressif-master-1811/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: c:/installations/.espressif-master-1811/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/lib/no-rtti\libstdc++.a(eh_personality.o):(.literal.__gxx_personality_v0+0x14): undefined reference to `_Unwind_SetIP'
c:/installations/.espressif-master-1811/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: c:/installations/.espressif-master-1811/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/lib/no-rtti\libstdc++.a(eh_personality.o): in function `base_of_encoded_value(unsigned char, _Unwind_Context*)':
/builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/xtensa-esp32-elf/src/gcc/libstdc++-v3/../libgcc/unwind-pe.h:114: undefined reference to `_Unwind_GetTextRelBase'
c:/installations/.espressif-master-1811/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: /builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/xtensa-esp32-elf/src/gcc/libstdc++-v3/../libgcc/unwind-pe.h:116: undefined reference to `_Unwind_GetDataRelBase'
c:/installations/.espressif-master-1811/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: /builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/xtensa-esp32-elf/src/gcc/libstdc++-v3/../libgcc/unwind-pe.h:118: undefined reference to `_Unwind_GetRegionStart'
c:/installations/.espressif-master-1811/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: c:/installations/.espressif-master-1811/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/lib/no-rtti\libstdc++.a(eh_personality.o): in function `parse_lsda_header(_Unwind_Context*, unsigned char const*, lsda_header_info*)':
/builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/eh_personality.cc:55: undefined reference to `_Unwind_GetRegionStart'
c:/installations/.espressif-master-1811/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: c:/installations/.espressif-master-1811/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/lib/no-rtti\libstdc++.a(eh_personality.o): in function `__gxx_personality_v0':
/builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/eh_personality.cc:438: undefined reference to `_Unwind_GetLanguageSpecificData'
c:/installations/.espressif-master-1811/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: /builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/eh_personality.cc:448: undefined reference to `_Unwind_GetIPInfo'
c:/installations/.espressif-master-1811/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: /builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/eh_personality.cc:712: undefined reference to `_Unwind_SetGR'
c:/installations/.espressif-master-1811/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: /builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/eh_personality.cc:714: undefined reference to `_Unwind_SetGR'
c:/installations/.espressif-master-1811/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: /builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/eh_personality.cc:716: undefined reference to `_Unwind_SetIP'
[... cut ...] ter-1811/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/lib/no-rtti\libstdc++.a(eh_globals.o): in function `eh_globals_dtor(void*)':
/builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/eh_globals.cc:83: undefined reference to `_Unwind_DeleteException'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```

igrr, 2019-12-17:
Can you please try `idf.py fullclean` and then `idf.py build`? If this still doesn't work, can you please check if you can build any of the IDF example projects?

arnkmar, 2019-12-20:
Hello,

The issue is resolved with proposed solution. I was able to build an example project. 
Additionally, I had to update submodules ```git submodule update --recursive``` to build my project successfully. 

Thanks!

closed 2019-12-20 by arnkmar