espressif/esp-idf #4504: BLE Cannot Stop Advertising

jonhunterking66, 2019-12-17:
## Environment

- Development Kit:    ESP32-DevKitC
- Kit version (for WroverKit/PicoKit/DevKitC): v4
- Module or chip used: ESP32-WROOM-32D
- IDF version (run ``git describe --tags`` to find it): 
    v4.1-dev-1086-g93a8603c5
- Build System:         [Make|CMake]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
- Operating System:     Linux
- Power Supply:         USB

## Problem Description

I am calling esp_ble_stop_advertising after setting up and advertising ble. It does not stop advertising unless I place the esp_ble_stop_advertising within the ESP_GAP_BLE_ADV_STOP_COMPLETE_EVT in my gapevent handler

### Expected Behavior

When esp_ble_stop_advertising is called the device should stop advertising. 

### Actual Behavior

Device continues to advertising 

### Code to reproduce this issue
Sorry to post all the code here, my repo is private. I will post only relevant code here
gap event handler 
```cpp
static void gap_event_handler(esp_gap_ble_cb_event_t event, esp_ble_gap_cb_param_t *param)
{
    switch (event) {
#ifdef CONFIG_SET_RAW_ADV_DATA
    case ESP_GAP_BLE_ADV_DATA_RAW_SET_COMPLETE_EVT:
        adv_config_done &= (~adv_config_flag);
        if (adv_config_done==0){
            esp_ble_gap_start_advertising(&adv_params);
        }
        break;
    case ESP_GAP_BLE_SCAN_RSP_DATA_RAW_SET_COMPLETE_EVT:
        adv_config_done &= (~scan_rsp_config_flag);
        if (adv_config_done==0){
            esp_ble_gap_start_advertising(&adv_params);
        }
        break;
#else
    case ESP_GAP_BLE_ADV_DATA_SET_COMPLETE_EVT:
        adv_config_done &= (~adv_config_flag);
        if (adv_config_done == 0){
            esp_ble_gap_start_advertising(&adv_params);
        }
        break;
    case ESP_GAP_BLE_SCAN_RSP_DATA_SET_COMPLETE_EVT:
        adv_config_done &= (~scan_rsp_config_flag);
        if (adv_config_done == 0){
            esp_ble_gap_start_advertising(&adv_params);
        }
        break;
#endif
    case ESP_GAP_BLE_ADV_START_COMPLETE_EVT:
        //advertising start complete event to indicate advertising start successfully or failed
        if (param->adv_start_cmpl.status != ESP_BT_STATUS_SUCCESS) {
            ESP_LOGE(GATTS_TAG, "Advertising start failed\n");
        }
        break;
    case ESP_GAP_BLE_ADV_STOP_COMPLETE_EVT:
        if (param->adv_stop_cmpl.status != ESP_BT_STATUS_SUCCESS) {
            ESP_LOGE(GATTS_TAG, "Advertising stop failed\n");
        }
        else {
            ESP_LOGI(GATTS_TAG, "Stop adv successfully\n");
        }
        break;
    case ESP_GAP_BLE_UPDATE_CONN_PARAMS_EVT:
         ESP_LOGI(GATTS_TAG, "update connetion params status = %d, min_int = %d, max_int = %d,conn_int = %d,latency = %d, timeout = %d",
                  param->update_conn_params.status,
                  param->update_conn_params.min_int,
                  param->update_conn_params.max_int,
                  param->update_conn_params.conn_int,
                  param->update_conn_params.latency,
                  param->update_conn_params.timeout);
        break;
    default:
        break;
    }
}
```
Initializing ble
```cpp
void bleInit(){
    esp_err_t ret;

    // Initialize NVS.
    ret = nvs_flash_init();
    if (ret == ESP_ERR_NVS_NO_FREE_PAGES) {
        ESP_ERROR_CHECK(nvs_flash_erase());
        ret = nvs_flash_init();
    }
    ESP_ERROR_CHECK( ret );

    ESP_ERROR_CHECK(esp_bt_controller_mem_release(ESP_BT_MODE_CLASSIC_BT));

    esp_bt_controller_config_t bt_cfg = BT_CONTROLLER_INIT_CONFIG_DEFAULT();
    ret = esp_bt_controller_init(&bt_cfg);
    if (ret) {
        ESP_LOGE(GATTS_TAG, "%s initialize controller failed\n", __func__);
        return;
    }

    ret = esp_bt_controller_enable(ESP_BT_MODE_BLE);
    if (ret) {
        ESP_LOGE(GATTS_TAG, "%s enable controller failed\n", __func__);
        return;
    }
    ret = esp_bluedroid_init();
    if (ret) {
        ESP_LOGE(GATTS_TAG, "%s init bluetooth failed\n", __func__);
        return;
    }
    ret = esp_bluedroid_enable();
    if (ret) {
        ESP_LOGE(GATTS_TAG, "%s enable bluetooth failed\n", __func__);
        return;
    }

    ret = esp_ble_gatts_register_callback(gatts_event_handler);
    if (ret){
        ESP_LOGE(GATTS_TAG, "gatts register error, error code = %x", ret);
        return;
    }
    ret = esp_ble_gap_register_callback(gap_event_handler);
    if (ret){
        ESP_LOGE(GATTS_TAG, "gap register error, error code = %x", ret);
        return;
    }
    ret = esp_ble_gatts_app_register(PROFILE_A_APP_ID);
    if (ret){
        ESP_LOGE(GATTS_TAG, "gatts app register error, error code = %x", ret);
        return;
    }
//    ret = esp_ble_gatts_app_register(PROFILE_B_APP_ID);
//   if (ret){
//        ESP_LOGE(GATTS_TAG, "gatts app register error, error code = %x", ret);
//        return;
//    }
    esp_err_t local_mtu_ret = esp_ble_gatt_set_local_mtu(500);
    if (local_mtu_ret){
        ESP_LOGE(GATTS_TAG, "set local  MTU failed, error code = %x", local_mtu_ret);
    }
}
```

In my task I call the bleInit function, then proceed to call the esp_ble_stop_advertising function after.

GYC-Espressif, 2019-12-18:
Hello, I want to know the position you call the ` esp_ble_stop_advertising ` , do you call it directly after calling the  `bleInit()` ? or call that in other tasks? our BLE stack is working in the FreeRTOS system, we need to wait for the BLE stack to fully start,  you can call the ` esp_ble_stop_advertising ` after the ESP_GAP_BLE_ADV_START_COMPLETE_EVT is already coming. If you still have problems, please provide me with the log of the program. Thank you.

jonhunterking66, 2019-12-18:
@GYC-Espressif thanks for your response. I do call esp_ble_stop_advertising() immediately after bleInit() within the same task. Are you suggesting that I wait for the ESP_GAP_BLE_ADV_START_COMPLETE_EVT to occur before I call esp_ble_stop_advertising()?

jonhunterking66, 2019-12-18:
@GYC-Espressif that worked. I added a delay between bleInit() and esp_ble_stop_advertising() and that worked! Cheers!

closed 2019-12-18 by jonhunterking66