espressif/esp-idf #4509: Timer crash

PaulFreund, 2019-12-18:
## Environment

- Module or chip used:  [ESP32-WROOM-32]
- IDF version [v4.1-dev-1543-g431066f16]
- Build System:         [Make]
- Compiler version [xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0]
- Operating System:     [Linux Debian WSL]
- Power Supply:         [USB]

## Problem Description

Crash when timer_isr_register is called

### Expected Behavior

No crash as with previous versions

### Actual Behavior

Crash

### Steps to repropduce

1. Execute

### Code to reproduce this issue

```cpp
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include <driver/timer.h>
#include "esp_log.h"

static void TimerInterruptHandler(void* context) {
    // @NOTE: This crashes regardles of spinlock ownership
    // timer_spinlock_take(TIMER_GROUP_0); // @TODO: Not sure in which case is this required?
    timer_set_alarm(TIMER_GROUP_0, TIMER_0, TIMER_ALARM_EN);
    // timer_spinlock_give(TIMER_GROUP_0); // @TODO: Not sure in which case is this required?
    return;
}

void app_main(void) {

    // Setup timer interrupt
    timer_config_t timerConfig = {
        .alarm_en = TIMER_ALARM_EN,
        .counter_en = TIMER_START,
        .intr_type = TIMER_INTR_LEVEL,
        .counter_dir = TIMER_COUNT_UP,
        .auto_reload = true,
        .divider = 80, // Timer Clock = 80Mhz -> Divider of 80 -> 1Mhz 
    };

    if(timer_init(TIMER_GROUP_0, TIMER_0, &timerConfig) != ESP_OK) { return; }
    if(timer_set_counter_value(TIMER_GROUP_0, TIMER_0, 0) != ESP_OK) { return; }
    if(timer_set_alarm_value(TIMER_GROUP_0, TIMER_0, 1000) != ESP_OK) { return; }
    if(timer_enable_intr(TIMER_GROUP_0, TIMER_0) != ESP_OK) { return; }

    ESP_LOGE("main", "Crash comes with the next line");
    if(timer_isr_register(TIMER_GROUP_0, TIMER_0, TimerInterruptHandler, (void*)NULL, 0, NULL) != ESP_OK) { return; }
    if(timer_start(TIMER_GROUP_0, TIMER_0) != ESP_OK) { return; }

    ESP_LOGE("main", "Startup complete");

    while(true) {
        vTaskDelay(1);
    }
}
```

## Debug Logs

```
--- idf_monitor on /dev/ttyS23 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:940
load:0x40078000,len:11116
ho 0 tail 12 room 4
load:0x40080400,len:3112
entry 0x400805d4
E (123) main: Crash comes with the next line
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)
Core 0 register dump:
PC      : 0x4000bfdf  PS      : 0x00060034  A0      : 0x80086edd  A1      : 0x3ffb0a20  
A2      : 0x00060023  A3      : 0x00060021  A4      : 0x00000000  A5      : 0x00000001  
A6      : 0x0000cdcd  A7      : 0x00000134  A8      : 0x3ffb296c  A9      : 0x00000001  
A10     : 0x0000cdcd  A11     : 0x3ffb235c  A12     : 0x00000002  A13     : 0x0000cdcd  
A14     : 0x00000002  A15     : 0x00000134  SAR     : 0x00000013  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0xffffffff  
Core 0 was running in ISR context:
EPC1    : 0x40088719  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x4000bfdf
0x40088719: _not_l1 at /home/CENSORED/esp-idf/components/freertos/xtensa_context.S:185

ELF file SHA256: 8aa36bb2fc4745d40398ef6151b97e5d20a9c134245f78e2354b2b6c4df5d756

Backtrace: 0x4000bfdc:0x3ffb0a20 |<-CORRUPTED

Core 1 register dump:
PC      : 0x400e1e6a  PS      : 0x00060034  A0      : 0x800d17a5  A1      : 0x3ffb53f0  
0x400e1e6a: esp_pm_impl_waiti at /home/CENSORED/esp-idf/components/esp32/pm_esp32.c:486

A2      : 0x00000000  A3      : 0x00060023  A4      : 0x00060020  A5      : 0x00000001  
A6      : 0x0000abab  A7      : 0x00000000  A8      : 0x800863bc  A9      : 0x3ffb53d0  
A10     : 0x3ffb6408  A11     : 0x3ffb0498  A12     : 0x3ffb0498  A13     : 0x00000000  
A14     : 0x00060020  A15     : 0x00000000  SAR     : 0x00000000  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

ELF file SHA256: 8aa36bb2fc4745d40398ef6151b97e5d20a9c134245f78e2354b2b6c4df5d756

Backtrace: 0x400e1e67:0x3ffb53f0 0x400d17a2:0x3ffb5410 0x40087bb5:0x3ffb5430
0x400e1e67: esp_pm_impl_waiti at /home/CENSORED/esp-idf/components/esp32/pm_esp32.c:484

0x400d17a2: esp_vApplicationIdleHook at /home/CENSORED/esp-idf/components/esp_common/src/freertos_hooks.c:63

0x40087bb5: prvIdleTask at /home/CENSORED/esp-idf/components/freertos/tasks.c:3381

Rebooting...```

igrr, 2019-12-18:
Can you please try setting `.counter_en = false`, or moving the call to `timer_isr_register` before `timer_init`?

PaulFreund, 2019-12-18:
# Variant 1 - .counter_en = false

```
--- idf_monitor on /dev/ttyS23 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:940
load:0x40078000,len:11116
ho 0 tail 12 room 4
load:0x40080400,len:3112
entry 0x400805d4
E (123) main: Crash comes with the next line
E (123) main: Startup complete
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)
Core 0 register dump:
PC      : 0x400e172c  PS      : 0x00060034  A0      : 0x800d1d6c  A1      : 0x3ffb0a50  
0x400e172c: timer_set_alarm at /home/CENSORED/esp-idf/components/driver/timer.c:194 (discriminator 2)

A2      : 0x00000000  A3      : 0x3ffb150c  A4      : 0x00000001  A5      : 0x3ffb2810  
A6      : 0x3ffb4f38  A7      : 0x3ffb4e3f  A8      : 0x800e172c  A9      : 0x3ffb0a30  
A10     : 0x3ffb4f38  A11     : 0x00000000  A12     : 0x80087616  A13     : 0x3ffb4e00  
A14     : 0x00000003  A15     : 0x00060123  SAR     : 0x00000000  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  
Core 0 was running in ISR context:
EPC1    : 0x40088719  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x400e172c
0x40088719: _not_l1 at /home/CENSORED/esp-idf/components/freertos/xtensa_context.S:185

0x400e172c: timer_set_alarm at /home/CENSORED/esp-idf/components/driver/timer.c:194 (discriminator 2)

ELF file SHA256: ade534786d9b52535695422b7ef945baa748fce52b8ba1edcb6a8712b004bea4

Backtrace: 0x400e1729:0x3ffb0a50 0x400d1d69:0x3ffb0a80 0x40081d1d:0x3ffb0aa0 0x40088716:0x3ffb4e60 0x400d17a2:0x3ffb4e80 0x40087bb5:0x3ffb4ea0
0x400e1729: timer_set_alarm at /home/CENSORED/esp-idf/components/driver/timer.c:193 (discriminator 2)

0x400d1d69: TimerInterruptHandler at /home/CENSORED/main/main.c:9

0x40081d1d: _xt_lowint1 at /home/CENSORED/esp-idf/components/freertos/xtensa_vectors.S:1153

0x40088716: _not_l1 at /home/CENSORED/esp-idf/components/freertos/xtensa_context.S:185

0x400d17a2: esp_vApplicationIdleHook at /home/CENSORED/esp-idf/components/esp_common/src/freertos_hooks.c:63

0x40087bb5: prvIdleTask at /home/CENSORED/esp-idf/components/freertos/tasks.c:3381

Core 1 register dump:
PC      : 0x400e1e6a  PS      : 0x00060034  A0      : 0x800d17a5  A1      : 0x3ffb53f0  
0x400e1e6a: esp_pm_impl_waiti at /home/CENSORED/esp-idf/components/esp32/pm_esp32.c:486

A2      : 0x00000000  A3      : 0x00060023  A4      : 0x00060020  A5      : 0x00000001  
A6      : 0x0000abab  A7      : 0x00000000  A8      : 0x800863bc  A9      : 0x3ffb53d0  
A10     : 0x3ffb6400  A11     : 0x3ffb0498  A12     : 0x3ffb0498  A13     : 0x00000000  
A14     : 0x00060020  A15     : 0x00000000  SAR     : 0x00000000  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

ELF file SHA256: ade534786d9b52535695422b7ef945baa748fce52b8ba1edcb6a8712b004bea4

Backtrace: 0x400e1e67:0x3ffb53f0 0x400d17a2:0x3ffb5410 0x40087bb5:0x3ffb5430
0x400e1e67: esp_pm_impl_waiti at /home/CENSORED/esp-idf/components/esp32/pm_esp32.c:484

0x400d17a2: esp_vApplicationIdleHook at /home/CENSORED/esp-idf/components/esp_common/src/freertos_hooks.c:63

0x40087bb5: prvIdleTask at /home/CENSORED/esp-idf/components/freertos/tasks.c:3381

Rebooting...
```

# Variant 2 - timer_isr_register before timer_init:

```
E (123) timer_group: timer_isr_register(261): HW TIMER NEVER INIT ERROR
```

PaulFreund, 2019-12-18:
Just in case the sdkconfig:

[sdkconfig.txt](https://github.com/espressif/esp-idf/files/3978686/sdkconfig.txt)

KonstantinKondrashov, 2019-12-18:
Hi! Could you try this [6992.zip](https://github.com/espressif/esp-idf/files/3978700/6992.zip)?

PaulFreund, 2019-12-18:
Hi @KonstantinKondrashov unfortunately I tested original and variant 1 (not variant 2 as it results in a logical error) both still exhibit the crashing behavior after patching.

PaulFreund, 2019-12-19:
After looking at the timer_group_example I found that I did not clear the interrupt with ```timer_group_clr_intr_status_in_isr```. I am not sure why this worked previously but I see why it is needed. I think this can be closed if you don't see any other possible side effects

PaulFreund, 2019-12-19:
Just a small update, the example code provided earlier beaves correctly if I include ```timer_group_clr_intr_status_in_isr``` or ```TIMERG0.int_clr_timers.t0 = 1``` but my full application still has crashes coming from the timer, I will work on extracting a more complete example that exhibits this problem.

PaulFreund, 2019-12-20:
The problem is really hard to isolate, it requires a combination of different peripheral setups as it seems, currently Timer, UART, SD Card via SPI and WiFi. One interresting thing I found is that sometimes on reboot I get:

```rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)```

But I am using Timer Group 0

PaulFreund, 2019-12-20:
This is my smalles example exhibiting the behaviour so far:

```C
#include <string.h>
#include <nvs_flash.h>
#include <driver/gpio.h>
#include <driver/sdspi_host.h>
#include <esp_vfs_fat.h>
#include <freertos/FreeRTOS.h>
#include <freertos/task.h>
#include <driver/uart.h>
#include <driver/timer.h>
#include <esp_wifi.h>
#include <esp_log.h>

static void TimerInterruptHandler(void* context) {
    timer_spinlock_take(TIMER_GROUP_0);
    timer_group_clr_intr_status_in_isr(TIMER_GROUP_0, TIMER_0);    
    timer_group_enable_alarm_in_isr(TIMER_GROUP_0, TIMER_0);
    timer_spinlock_give(TIMER_GROUP_0);
}

void app_main(void) {
    
    // NVS
    if(ESP_OK != nvs_flash_init()) {
        ESP_LOGE("main", "nvs_flash_init failed");
        return;
    }

    // GPIO
    if(ESP_OK != gpio_install_isr_service(0)) {
        ESP_LOGE("main", "gpio_install_isr_service failed");
        return;
    }

    // MicroSD
    {
        sdmmc_host_t hostConfig;
        memset(&hostConfig, 0, sizeof(hostConfig));
        hostConfig.flags = SDMMC_HOST_FLAG_SPI;
        hostConfig.slot = VSPI_HOST;
        hostConfig.max_freq_khz = SDMMC_FREQ_PROBING;
        hostConfig.io_voltage = 3.3f;
        hostConfig.init = &sdspi_host_init;
        hostConfig.set_bus_width = NULL;
        hostConfig.set_card_clk = &sdspi_host_set_card_clk;
        hostConfig.do_transaction = &sdspi_host_do_transaction;
        hostConfig.deinit = &sdspi_host_deinit;
        hostConfig.command_timeout_ms = 0;

        sdspi_slot_config_t slotConfig;
        memset(&slotConfig, 0, sizeof(slotConfig));
        slotConfig.gpio_miso = GPIO_NUM_26;
        slotConfig.gpio_mosi = GPIO_NUM_25;
        slotConfig.gpio_sck  = GPIO_NUM_33;
        slotConfig.gpio_cs   = GPIO_NUM_32;
        slotConfig.gpio_cd   = GPIO_NUM_NC;
        slotConfig.gpio_wp   = GPIO_NUM_NC;
        slotConfig.dma_channel = 1;

        esp_vfs_fat_sdmmc_mount_config_t mountConfig;
        memset(&mountConfig, 0, sizeof(mountConfig));
        mountConfig.format_if_mount_failed = true;
        mountConfig.max_files = 10;
        mountConfig.allocation_unit_size = 0; // sector size

        sdmmc_card_t* microSDCard = NULL;
        if (ESP_OK != esp_vfs_fat_sdmmc_mount("/sd", &hostConfig, &slotConfig, &mountConfig, &microSDCard)) { 
            ESP_LOGE("main", "esp_vfs_fat_sdmmc_mount failed"); 
[... 161 lines cut ...]
ho 0 tail 12 room 4
load:0x40080400,len:3112
entry 0x400805d4
E (1155) main: Startup complete
Guru Meditation Error: Core  0 panic'ed (Unhandled debug exception)
Debug exception reason: BREAK instr 
Core 0 register dump:
PC      : 0x400803c0  PS      : 0x00060336  A0      : 0x3ffd1628  A1      : 0x3ffb4808  
0x400803c0: _DoubleExceptionVector at /home/freunpau/Robug_Eluxnet/esp-idf/components/freertos/xtensa_vectors.S:542

A2      : 0x3ffcb2d4  A3      : 0x3ffcc750  A4      : 0x00000001  A5      : 0x0000cdcd  
A6      : 0x3ffb34e4  A7      : 0x00000000  A8      : 0x800ecba9  A9      : 0x3ffcc6e0  
A10     : 0x00000001  A11     : 0x00000000  A12     : 0x3ffcb324  A13     : 0x00000000  
A14     : 0x3ffcb364  A15     : 0x00000001  SAR     : 0x0000001e  EXCCAUSE: 0x00000001  
EXCVADDR: 0xffffffe0  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

ELF file SHA256: 2a2e80b14d395f5abdc2952cdeeebdcadaec0002d4397bbbbb4a05b48de177a7

Backtrace: 0x400803bd:0x3ffb4808 |<-CORRUPTED
0x400803bd: _UserExceptionVector at ??:?

Rebooting...
```

The only thing connected is a MicroSD card via SPI (which was working fine so far and is only initialized not used in this example

KonstantinKondrashov, 2019-12-24:
Hi @PaulFreund!

Thanks for your report. The first issue was due to the wrong handle interrupt. Your last example uses it now correctly.

```c
static void TimerInterruptHandler(void* context) {
    timer_spinlock_take(TIMER_GROUP_0);
    timer_group_clr_intr_status_in_isr(TIMER_GROUP_0, TIMER_0);    
    timer_group_enable_alarm_in_isr(TIMER_GROUP_0, TIMER_0);
    timer_spinlock_give(TIMER_GROUP_0);
}
```

Regarding the last piece of code, I am not sure that the timer leads to this issue as in the first case. 
Could you look at https://github.com/espressif/esp-idf/blob/master/components/freertos/Kconfig#L86-L103. I think it might be a reason. 

I can not run last your code completely because I do not have MicroSD, I ignored an error in `esp_vfs_fat_sdmmc_mount()` and continued. The issue does not occur.

PaulFreund, 2020-01-13:
Hi @KonstantinKondrashov, coming back form vacation I did an update of ESP-IDF to master and it seems the issue is gone, no restart in 5 minutes so far. I will try to track down which change fixed it with a binary search.

PaulFreund, 2020-01-13:
Working with my last example above it seems that commit https://github.com/espressif/esp-idf/commit/f2f008e167ba292661d6a08c998d7928b928ee77 fixes the issue (I can isolate that single commit). I will next try to integrate with my actual code.

PaulFreund, 2020-01-13:
Actual project also works fine now after a few hours on master, I think we can close this.

closed 2020-01-13 by PaulFreund