espressif/esp-idf #4510: 关于gethostbyname()成功率低问题

match101, 2019-12-18:
你好：
        我们在线上发现了部分网络环境设备connect的失败率很高，然后我们发现了在esp SDK里使用了gethostbyname此类被淘汰的接口，在修改为getaddrinfo后问题得到了解决。
        官方是否可以在SDK里逐步替换gethostbyname此类老旧的接口？

AxelLin, 2019-12-24:
Hi,
I don't get why gethostbyname() is an issue and why getaddrinfo() is better.
Are there any links about the discussion of gethostbyname() issue?

match101, 2019-12-24:
hi,
you can just type #man gethostbyname , and there some notes like that:

"CONFORMING TO
       POSIX.1-2001  specifies gethostbyname(), gethostbyaddr(), sethostent(),
       endhostent(),  gethostent(),  and  h_errno;  gethostbyname(),  gethost‐
       byaddr(),   and  h_errno  are  marked  obsolescent  in  that  standard.
       POSIX.1-2008 removes the specifications  of  gethostbyname(),  gethost‐
       byaddr(),  and h_errno, recommending the use of getaddrinfo(3) and get‐
       nameinfo(3) instead."
       "The gethostbyname() and gethostbyaddr() functions are obsolete.
         Applications should use getaddrinfo(3) and getnameinfo(3) instead"

/*************************************************************************/
they removed the specifications  of  gethostbyname()  20 years ago.
also you can search "gethostbyname  obsolete" with baidu/google, 
there are too many links about the discussion of gethostbyname() issue

thanks a lot

zhangyanjiaoesp, 2020-01-19:
@match101 
Thanks for your reporting, we can replace the older interfaces `gethostbyname()` step by step.

neorevx, 2020-03-21:
https://github.com/espressif/esp-idf/blob/6330b3345e87eb4401e7be7c8b6fea2870c35d9f/components/tcp_transport/transport_tcp.c#L39

I think it should be replaced asap. tcp_adapter is the base for http client. Concurrent http connections can generate wrong destinations IPs.

AxelLin, 2020-04-01:
> @match101
> Thanks for your reporting, we can replace the older interfaces `gethostbyname()` step by step.

@zhangyanjiaoesp 

Then replace it in the basic components like tcp_transport/transport_tcp.c
is a good start. Currently I open-coded my tcp connection code instead of
using tcp_transport because it uses the gethostbyname(). However, it's
difficult to avoid the issue since there are other components using tcp_transport().

Agree to @neorevx , this should be fixed ASAP.

Alvin1Zhang, 2020-05-27:
@match101 Thanks for report and @AxelLin for the updates. The fix has been available at https://github.com/espressif/esp-idf/commit/80e5eb896cb6acd285e98e4a06480a1dbf233994. Appreciate your patience waiting for the fix. Feel free to reopen if the issue still happens. Thanks.

closed 2020-05-27 by Alvin1Zhang