espressif/esp-idf #4513: idf.py fails if SECURE_BOOTLOADER_REFLASHABLE is selected

fgiancane8, 2019-12-18:
INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

## Problem Description

I am trying to have a __secure boot__ build for testing purposes.

If ```SECURE_BOOTLOADER_REFLASHABLE``` is enabled, then this is my output from CMake:
```
[ 50%] Built target gen_secure_boot_signing_key
[ 50%] Built target partition_table
[ 50%] Built target gen_secure_boot_keys
[100%] Performing build step for 'bootloader'
-- Project version: v4.1-dev-1543-g431066f16
-- Building ESP-IDF components for target esp32
-- Adding linker script /home/fgiancane8/esp/esp-idf/components/esp32/ld/esp32.peripherals.ld
-- Adding linker script /home/fgiancane8/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script /home/fgiancane8/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
-- Adding linker script /home/fgiancane8/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script /home/fgiancane8/esp/esp-idf/components/bootloader/subproject/main/ld/esp32/bootloader.ld
-- Adding linker script /home/fgiancane8/esp/esp-idf/components/bootloader/subproject/main/ld/esp32/bootloader.rom.ld
-- Components: bootloader bootloader_support efuse esp32 esp_common esp_rom esptool_py log main micro-ecc partition_table soc spi_flash xtensa
-- Component paths: /home/fgiancane8/esp/esp-idf/components/bootloader /home/fgiancane8/esp/esp-idf/components/bootloader_support /home/fgiancane8/esp/esp-idf/components/efuse /home/fgiancane8/esp/esp-idf/components/esp32 /home/fgiancane8/esp/esp-idf/components/esp_common /home/fgiancane8/esp/esp-idf/components/esp_rom /home/fgiancane8/esp/esp-idf/components/esptool_py /home/fgiancane8/esp/esp-idf/components/log /home/fgiancane8/esp/esp-idf/components/bootloader/subproject/main /home/fgiancane8/esp/esp-idf/components/bootloader/subproject/components/micro-ecc /home/fgiancane8/esp/esp-idf/components/partition_table /home/fgiancane8/esp/esp-idf/components/soc /home/fgiancane8/esp/esp-idf/components/spi_flash /home/fgiancane8/esp/esp-idf/components/xtensa
-- Configuring done
CMake Error in CMakeLists.txt:
  Cannot find source file:

    gen_project_binary

  Tried extensions .c .C .c++ .cc .cpp .cxx .cu .m .M .mm .h .hh .h++ .hm
  .hpp .hxx .in .txx

-- Generating done
-- Build files have been written to: /home/fgiancane8/esp/ocs-firmware/build/bootloader
```
If on the other hand, I choose ```SECURE_BOOTLOADER_ONE_TIME_FLASH``` option, everything goes as expected.
### Expected Behavior
```idf.py bootloader``` produces a bootloader.
### Actual Behavior
```idf.py bootloader``` fails to produce a bootloader.
### Steps to reproduce
1. In any project, issue
```idf.py menuconfig```
2. Select ```SECURE_BOOTLOADER_REFLASHABLE```
3. run ```idf.py bootloader```

Alvin1Zhang, 2019-12-19:
@fgiancane8 Thanks for reporting. Thanks.

fgiancane8, 2019-12-19:
@Alvin1Zhang you're welcome. It seems that the CMake file is recognizing the gen_project_binary Makefile target as a source file.

However, the bootloader makefile is correctly generated, as if I issue the
`make gen_project_binary`
in the build/bootloader folder, the target is correctly executed by `make` command.

projectgus, 2019-12-20:
Hi @fgiancane8 ,

Thanks for the detailed bug report. You're right that the build system is incorrectly passing a target name to a clause which needs a source file.

A fix will be merged soon and this issue will be updated, in the meantime if you wish you can apply this short patch to your IDF directory:

```patch
diff --git a/components/bootloader/subproject/CMakeLists.txt b/components/bootloader/subproject/CMakeLists.txt
index 9a49fee85c..e6bdbb7dba 100644
--- a/components/bootloader/subproject/CMakeLists.txt
+++ b/components/bootloader/subproject/CMakeLists.txt
@@ -76,7 +76,7 @@ if(CONFIG_SECURE_BOOTLOADER_REFLASHABLE)
         COMMAND ${CMAKE_COMMAND} -E echo "DIGEST ${bootloader_digest_bin}"
         COMMAND ${ESPSECUREPY} digest_secure_bootloader --keyfile "${secure_bootloader_key}"
         -o "${bootloader_digest_bin}" "${CMAKE_BINARY_DIR}/bootloader.bin"
-        MAIN_DEPENDENCY gen_project_binary
+        MAIN_DEPENDENCY "${CMAKE_BINARY_DIR}/.bin_timestamp"
         DEPENDS gen_secure_bootloader_key gen_project_binary
         VERBATIM)
```

fgiancane8, 2020-01-05:
Hi @projectgus,

Sorry I was in holiday leave, thus I am responding right now. I will be back in office next week so I will test this patch and report back to you if everything goes well.

Thanks for your support

fgiancane8, 2020-01-07:
Hi @projectgus ,
I can confirm that the patch fixed the issue.

Many thanks!

closed 2020-01-07 by espressif-bot