espressif/esp-idf #4515: High SENSOR_VP leakage in deep sleep

pylorak, 2019-12-19:
## Environment

- Development Kit:   none
- Module or chip used:  ESP32-WROOM-32D
- IDF version: v3.3.1
- Build System:  Make
- Compiler version: 1.22.0-80-g6c4433a5
- Operating System:   Windows
- Power Supply:     External 3.3V

## Problem Description

Using the attached schematic (VBAT is 4.0V, BAT_ADC is connected to SENSOR_VP, BAT_ADC_EN driven low), leakage current over SENSOR_VP pin is over 60uA in deep sleep mode.

### Expected Behavior

Leakage negligible.

### Actual Behavior

~60nA leakage with VBAT = 4.0V. Furthermore, leakage increases rapidly if VBAT is even just slightly increased (20-30uA for every 0.1V in VBAT)

### Steps to repropduce

[image]

If R11 is desoldered, leakage stops (whole circuit consumption goes down to 13uA). If R12 is removed, total circuit consumption stays high (>70uA). Therefor reason for high consumption must be leakage over ESP32 pin (sensor_vp).

### Code excerpt

void app_main(void)
{
#if 1 // enabling or disabling makes no difference to leakage
    adc1_config_width(ADC_WIDTH_BIT_12);
    adc1_config_channel_atten(ADC1_CHANNEL_0, ADC_ATTEN_DB_0);
    esp_adc_cal_characteristics_t *adc_chars = calloc(1, sizeof(esp_adc_cal_characteristics_t));
    esp_adc_cal_value_t val_type = esp_adc_cal_characterize(ADC_UNIT_1, ADC_ATTEN_DB_0, ADC_WIDTH_BIT_12, 1100, adc_chars);
#endif

#if 1 // enabling or disabling makes no difference to leakage
    rtc_gpio_init(GPIO_NUM_36);
#endif

    esp_deep_sleep_start();
}

### Additional notes

Calling rtc_gpio_init() on SENSOR_VP makes no difference. It can be a bug in ESP-IDF (pin not isolated), or maybe this is a hardware issue? SENSOR_VP has no pullup or pulldown in hardware, and it also has no output circuitry, so I expect the hardware leakage to be near zero (ADC input should be high-impedance). ESP32 datasheet has no information about expected ADC/analog pin leakage, only for digital inputs which is 50nA maximum.

negativekelvin, 2019-12-19:
Why not use a high side switch?  Vbat > Vcc could leak through protection diode.

pylorak, 2019-12-19:
> Why not use a high side switch?
All things indicated this should work and it requires less components (and also less kinds of components as there is otherwise no P-channel MOSFET in the design). I will have to do that if there is no solution in software.

>Vbat > Vcc could leak through protection diode.
My first thought as well. But if this were the case, then digital pins would also leak in deep sleep and the rtc_gpio_init() function would have no use. The rtc_gpio_init() IDF function is capable of isolating pins such that they do not leak (they are then either kept powered on during deep sleep or there are no diodes -  no info about such diodes in either the datasheet or the TRM). This is why I thought this might be a software issue. Or, if rtc_gpio_init() in incapable of isolating SENSOR_VP due to hardware, then it should return ESP_ERR_INVALID_ARG like it does for other non-supported pins.

closed 2019-12-19 by pylorak