espressif/esp-idf #4516: Unable to create socket: errno 23

YSP0316, 2019-12-19:
Unable to create socket: errno 23

The socket cannot be recreated after TCP server restarts, as follows

[image]
[image]

```c++
static void tcp_client_task(void *pvParameters)
{
    char rx_buffer[20]; //"WeChat"
    char addr_str[128];
    int addr_family;
    int ip_protocol;
    while (1) {
        printf("\n-----------------------#1-------------------------\n");
        app_wifi_wait_connected();
        struct sockaddr_in destAddr;
        destAddr.sin_addr.s_addr = inet_addr(HOST_IP_ADDR);//
        destAddr.sin_family = AF_INET;
        destAddr.sin_port = htons(PORT);
        addr_family = AF_INET;
        ip_protocol = IPPROTO_IP;
        inet_ntoa_r(destAddr.sin_addr, addr_str, sizeof(addr_str) - 1);
        int sock =  socket(addr_family, SOCK_STREAM, ip_protocol);
        printf("\n-----------------------#2-------------------------\n");
        if (sock < 0) {
            ESP_LOGE(TAG, "Unable to create socket: errno %d", errno);
            vTaskDelay(1000 / portTICK_RATE_MS);
            continue;//break;
        }
        int err = connect(sock, (struct sockaddr *)&destAddr, sizeof(destAddr));
        printf("\n-----------------------#3-------------------------\n");
        if (err != 0) {
            #ifdef ManBodebug
            ESP_LOGE(TAG, "Socket unable to connect: errno %d", errno);
            #endif
            vTaskDelay(1000 / portTICK_RATE_MS);
            continue;
        }
        //ESP_LOGI(TAG, "Socket Successfully connected\n");
        err = send(sock, WifiMac, 13, 0);
        if (err < 0) {
                ESP_LOGE(TAG, "Error occured during sending: errno %d", errno);
                continue;//break;
            }
        ManBo.soketState=1; //Socket 连接成功标志位
        while (1) {//数据传输
            int len = recv(sock, rx_buffer, sizeof(rx_buffer) - 1, 0);
            // Error occured during receiving
            if (len < 0) 
            {
                ESP_LOGE(TAG, "recv failed: errno %d", errno);
                break;
            }
            else {
                rx_buffer[len] = 0; // Null-terminate whatever we received and treat like a string
                printf( "%s", rx_buffer);
            }
            int err = send(sock, WifiMac, 13, 0);
            if (err < 0) {
                ESP_LOGE(TAG, "Error occured during sending: errno %d", errno);
                break;
            }
            vTaskDelay(200 / portTICK_PERIOD_MS);
        }
        if (sock != -1) {
            ESP_LOGE(TAG, "Shutting down socket and restarting...");
            shutdown(sock, 0);
            close(sock);
            ManBo.soketState=0; //
        }
    }
    vTaskDelete(NULL);
}
```

Alvin1Zhang, 2019-12-19:
@YSP0316 Thanks for reporting the issue. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/blob/master/docs/issue_template.md)? Information like elf, sdk configuration, backtrace, log outputs, commit ID, hardware and etc. would help us debug further. Thanks.

YSP0316, 2019-12-19:
thank youThe problem has been solved

The problem is that I did not close the last socket before creating a new one, so the creation was unsuccessful.

Alvin1Zhang, 2019-12-19:
@YSP0316 Glad you resolved the issue, will close the ticket. Feel free to reopen if necessary. Thanks.

closed 2019-12-19 by Alvin1Zhang

wawan-ikhwan, 2022-08-03:
@YSP0316 how you close the socket?