espressif/esp-idf #4520: Found no platform_util.h under  mbedtls

ElliottSeer, 2019-12-19:
尝试编译Hello World
提示 
G:/ESP_IDF/esp-idf/components/mbedtls/port/esp32/aes.c:30:10: fatal error: mbedtls/platform_util.h: No such file or directory
 #include "mbedtls/platform_util.h"
          ^~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
在 espressif/mbedtls 仓下没有找到 platform_util,h 疑惑

Running ninja in directory g:\esp_idf\esp-idf\examples\get-started\hello_world\build
Executing "ninja all"...
[94/851] Building C object esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.dir/__/__/port/esp32/aes.c.obj
FAILED: esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.dir/__/__/port/esp32/aes.c.obj
ccache D:\APP\espressif\tools\xtensa-esp32-elf\esp-2019r2-8.2.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-gcc.exe -DMBEDTLS_CONFIG_FILE=\"
mbedtls/esp_config.h\" -Iconfig -I../../../../components/mbedtls/port/include -I../../../../components/mbedtls/mbedtls/include -I../..
/../../components/newlib/platform_include -I../../../../components/freertos/include -I../../../../components/heap/include -I../../../.
./components/log/include -I../../../../components/soc/esp32/include -I../../../../components/soc/include -I../../../../components/esp_
rom/include -I../../../../components/esp_common/include -I../../../../components/xtensa/include -I../../../../components/xtensa/esp32/
include -I../../../../components/esp32/include -I../../../../components/driver/include -I../../../../components/esp_ringbuf/include -I
../../../../components/esp_event/include -I../../../../components/esp_netif/include -I../../../../components/lwip/include/apps -I../..
/../../components/lwip/include/apps/sntp -I../../../../components/lwip/lwip/src/include -I../../../../components/lwip/port/esp32/inclu
de -I../../../../components/lwip/port/esp32/include/arch -I../../../../components/vfs/include -I../../../../components/esp_wifi/includ
e -I../../../../components/esp_wifi/esp32/include -I../../../../components/esp_eth/include -I../../../../components/tcpip_adapter/incl
ude -I../../../../components/efuse/include -I../../../../components/efuse/esp32/include -I../../../../components/app_trace/include -ml
ongcalls -Wno-frame-address -Wall -Wextra -W -Wdeclaration-after-statement -Wwrite-strings -Wlogical-op -Wshadow -Wmissing-declaration
s -Wmissing-prototypes   -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -nostdlib -Wall -Werror=all -Wno-error=unused
-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-paramet
er -Wno-sign-compare -ggdb -Og -std=gnu99 -Wno-old-style-declaration -D_GNU_SOURCE -DIDF_VER=\"v4.1-dev-1086-g93a8603c5-dirty\" -DGCC_
NOT_5_2_0 -DESP_PLATFORM -MD -MT esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.dir/__/__/port/esp32/aes.c.obj -MF esp-idf\mbed
tls\mbedtls\library\CMakeFiles\mbedcrypto.dir\__\__\port\esp32\aes.c.obj.d -o esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.di
r/__/__/port/esp32/aes.c.obj   -c G:/ESP_IDF/esp-idf/components/mbedtls/port/esp32/aes.c
G:/ESP_IDF/esp-idf/components/mbedtls/port/esp32/aes.c:30:10: fatal error: mbedtls/platform_util.h: No such file or directory
 #include "mbedtls/platform_util.h"
          ^~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
[107/851] Building C object esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedtls.dir/ssl_srv.c.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

Alvin1Zhang, 2019-12-19:
@ElliottSeer Thanks for reporting the issue. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/blob/master/docs/issue_template.md)? Information like elf, sdk configuration, backtrace, log outputs, commit ID, hardware and etc. would help us debug further. Thanks.

mahavirj, 2019-12-20:
@ElliottSeer Did you update `mbedtls` submodule pointer? Here is file path for quick reference, https://github.com/espressif/mbedtls/blob/mbedtls-2.16.2-idf/include/mbedtls/platform_util.h

ElliottSeer, 2019-12-20:
> @ElliottSeer Did you update `mbedtls` submodule pointer? Here is file path for quick reference, https://github.com/espressif/mbedtls/blob/mbedtls-2.16.2-idf/include/mbedtls/platform_util.h

Yes, I did update mbedtls in wrong branch. I have to git clone every dependence myself due to the bad network environment in China. Could you build another repository on gitee? So developers in China can access esp idf more convenient.

closed 2019-12-20 by ElliottSeer