espressif/esp-idf #4521: Writing data to flash permanently

Dipendra77, 2019-12-19:
Hi,
I am configuring the wi-fi credentials from an android app, allowing user to connect to the wifi network they desire to connect. I am sending the wifi credentials through a udp. But the wifi credentials are lost on next boot and expects us to configure again. Can anyone tell me if we can store these credentials to some flash sector permanently so it remembers the previously configured network and connects to it even after reboot, unless user wants to change the credentials to other network.
Thank you in advance.

stickbreaker, 2019-12-19:
[NVS(Non Volatile Storage)](https://github.com/espressif/esp-idf/tree/master/components/nvs_flash)

mahavirj, 2019-12-23:
@Dipendra77 You might also want to look at https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/provisioning/provisioning.html#sample-code where we provide end-to-end examples (along with Android/iOS SDKs) for wifi provisioning use-cases (credentials do get stored in flash using NVS). 

For how-to questions please consider using esp32.com forum.

closed 2019-12-23 by mahavirj