espressif/esp-idf #4523: BLE Start Advertising Causes -> Core  1 panic'ed (LoadProhibited)

jonhunterking66, 2019-12-19:
## Environment

- Development Kit:      ESP32-DevKitC
- Kit version (for WroverKit/PicoKit/DevKitC): v4
- Module or chip used:  ESP32-WROOM-32D
- IDF version (run ``git describe --tags`` to find it): 
   v4.1-dev-1086-g93a8603c5
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
- Operating System:     Linux
- Power Supply:         USB

## Problem Description
I have two tasks running initialized in my main. One is a "main task" that contains a switch statement. The various cases will eventually contain binary semaphores to trigger various task functions.

The other task is BLE task that initializes BLE then stops advertising after. In its while loop it waits for a semaphore then starts advertising when it receives one. 

### Expected Behavior

BLE will initialize properly then proceed to stop advertising. When the main task steps into the case that gives a binary semaphore, the BLE task waiting for the semaphore will trigger and start advertising. 

### Actual Behavior

BLE does initialize properly and successfully stops scanning. However, when the main task enters the case that gives the binary semaphore to the ble task and start advertising is called, I get a guru mediation error. 

### Code to reproduce this issue
** I am not putting in all the code to keep it less spamy. Let me know if you want something else!

**MAIN CODE**
```cpp
xSemaphoreHandle bleSemaphore;

void app_main()
{
  oto oto1; //struct to hold config and pass to functions
  esp_log_level_set(TAG, ESP_LOG_DEBUG);
  //wifiInit();
  //bleInit();
  //vTaskDelay(100/portTICK_PERIOD_MS);
  bleSemaphore = xSemaphoreCreateBinary();
  xTaskCreate(&main_task, "MAIN", 1024 *3, &oto1, 2, NULL);
  xTaskCreate(&ble_task, "BLE", 2048, &oto1, 1, NULL);
  //xTaskCreate(&wifi_task, "WIFI", 1024 * 3, NULL, 5, NULL);
  //xTaskCreate(&ble_task, "BLE", 2048, &oto1, 3, NULL);
}

void main_task(void * params){
    oto *oto1 = (oto *) params;
    oto1 -> state = 0;
    while(1){
        switch ((oto1->state))
        {
        case Sleep: 
            printf("Case 0\n");
            oto1->state = 1;
            break;
        case BleAdvertisement: 
            printf("Case 1\n");
            xSemaphoreGive(bleSemaphore);
            oto1->state = 2;            
            break;
        case BleConnect: 
            printf("Case 2\n");
            oto1->state = 3;
            break; 
        case Wifi: 
            printf("Case 3\n");
            oto1->state = 4;
            break;   
        case Run: 
[... 191 lines cut ...]

I (1138) BLE: ADD_CHAR_EVT, status 133,  attr_handle 0, service_handle 40

I (1148) BLE: ADD_DESCR_EVT, status 0, attr_handle 45, service_handle 40

I (1148) BLE: ADD_DESCR_EVT, status 133, attr_handle 0, service_handle 40

I (1168) BLE: Stop adv successfully

Case 0
Case 1
In BLE Task
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x4000c344  PS      : 0x00060730  A0      : 0x800dc545  A1      : 0x3ffc9af0  
A2      : 0x3ffc9b08  A3      : 0x00000000  A4      : 0x00000020  A5      : 0x3ffc9b08  
A6      : 0x00000100  A7      : 0x00000002  A8      : 0x40000000  A9      : 0x3ffc9ae0  
A10     : 0x00000000  A11     : 0x00000001  A12     : 0x3ffc9b10  A13     : 0x3ffb65fc  
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x00000008  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffc  

ELF file SHA256: 53cce7ebaaa24d9e94682a8ab5e16962240a6a7411d16acd8574edb4929fdb7f

Backtrace: 0x4000c341:0x3ffc9af0 |<-CORRUPTED
```

gengyuchao, 2020-01-13:
You set the wrong param in the function `esp_ble_gap_start_advertising(&adv_params);`  of ble_task.

jonhunterking66, 2020-01-13:
@gengyuchao ESP_BT_MODE_BLE is incorrect?

gengyuchao, 2020-01-14:
```
/**
 * @brief           This function is called to start advertising.
 *
 * @param[in]       adv_params: pointer to User defined adv_params data structure.

 * @return
 *                  - ESP_OK : success
 *                  - other  : failed
 *
 */
esp_err_t esp_ble_gap_start_advertising (esp_ble_adv_params_t *adv_params);
```
You need to set the adv_params like that
```
static esp_ble_adv_params_t adv_params = {
    .adv_int_min        = 0x100,
    .adv_int_max        = 0x100,
    .adv_type           = ADV_TYPE_IND,
    .own_addr_type      = BLE_ADDR_TYPE_RANDOM,
    .channel_map        = ADV_CHNL_ALL,
    .adv_filter_policy = ADV_FILTER_ALLOW_SCAN_ANY_CON_ANY,
};
```

jonhunterking66, 2020-01-14:
@gengyuchao sorry, I did not include the full code. I do have adv_params initialized as such:
```
static esp_ble_adv_params_t adv_params = {
    .adv_int_min        = 0x20,
    .adv_int_max        = 0x40,
    .adv_type           = ADV_TYPE_IND,
    .own_addr_type      = BLE_ADDR_TYPE_PUBLIC,
    //.peer_addr            =
    //.peer_addr_type       =
    .channel_map        = ADV_CHNL_ALL,
    .adv_filter_policy = ADV_FILTER_ALLOW_SCAN_ANY_CON_ANY,
};
```

GYC-Espressif, 2020-01-14:
So, Please change `esp_ble_gap_start_advertising(ESP_BT_MODE_BLE);` to 'esp_ble_gap_start_advertising(&adv_params );'

gengyuchao, 2020-01-14:
So, Please change 'esp_ble_gap_start_advertising(ESP_BT_MODE_BLE);' to 'esp_ble_gap_start_advertising(&adv_params);'.That will resolve you problem.

jonhunterking66, 2020-01-14:
Ahhh, okay, I will give it a try. Thank you!

closed 2020-01-16 by jonhunterking66

Alvin1Zhang, 2020-01-17:
@jonhunterking66 Thanks for reporting and updates. Feel free to reopen if the issue still happens. Thanks.