espressif/esp-idf #4524: BLE stop advertising causes single core RTOS to stop running

jonhunterking66, 2019-12-19:
## Environment

-Development Kit: ESP32-DevKitC
-Kit version (for WroverKit/PicoKit/DevKitC): v4
-Module or chip used: ESP32-WROOM-32D
-IDF version (run git describe --tags to find it):
v4.1-dev-1086-g93a8603c5
-Compiler version (run xtensa-esp32-elf-gcc --version to find it):
xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
-Operating System: Linux
-Power Supply: USB

## Problem Description

When running on a single core (configured in menuconfig or by using a pinned task) calling esp_ble_gap_stop_advertising causes code to stop running
### Expected Behavior

esp_ble_gap_stop_advertising works successfully and my main task continues to run 

### Actual Behavior

once stop advertising is called my code stops

*** MAIN CODE ***
### Code to reproduce this issue

```cpp
xSemaphoreHandle bleSemaphore;

void app_main()
{
  oto oto1; //struct to hold config and pass to functions
  esp_log_level_set(TAG, ESP_LOG_DEBUG);
  //wifiInit();
  //bleInit();
  //vTaskDelay(100/portTICK_PERIOD_MS);
  bleSemaphore = xSemaphoreCreateBinary();
  xTaskCreate(&main_task, "MAIN", 1024 *3, &oto1, 2, NULL);
  xTaskCreate(&ble_task, "BLE", 2048, &oto1, 1, NULL);
  //xTaskCreate(&wifi_task, "WIFI", 1024 * 3, NULL, 5, NULL);
  //xTaskCreate(&ble_task, "BLE", 2048, &oto1, 3, NULL);
}

void main_task(void * params){
    oto *oto1 = (oto *) params;
    oto1 -> state = 0;
    while(1){
        switch ((oto1->state))
        {
        case Sleep: 
            printf("Case 0\n");
            oto1->state = 1;
            break;
        case BleAdvertisement: 
            printf("Case 1\n");
            xSemaphoreGive(bleSemaphore);
            oto1->state = 2;            
            break;
        case BleConnect: 
            printf("Case 2\n");
            oto1->state = 3;
            break; 
        case Wifi: 
            printf("Case 3\n");
            oto1->state = 4;
            break;   
        case Run: 
            printf("Case 4\n");
            oto1->state = 3;
            break;                        
[... 178 lines cut ...]
I (985) BLE: prf_char[0] =11

I (985) BLE: prf_char[1] =22

I (995) BLE: prf_char[2] =33

I (995) BLE: ADD_CHAR_EVT, status 0,  attr_handle 44, service_handle 40

E (1005) BLE: IN TWO
I (1005) BLE: the gatts demo char length = 3

I (1005) BLE: prf_char[0] =11

I (1015) BLE: prf_char[1] =22

I (1015) BLE: prf_char[2] =33

I (1025) BLE: ADD_CHAR_EVT, status 133,  attr_handle 0, service_handle 40

I (1025) BLE: ADD_DESCR_EVT, status 0, attr_handle 45, service_handle 40

I (1035) BLE: ADD_DESCR_EVT, status 133, attr_handle 0, service_handle 40

I (1055) BLE: Stop adv successfully
```

GYC-Espressif, 2020-03-03:
Hello. Sorry for the late reply. The `oto oto1; //struct to hold config and pass to functions` in your `app_main()` has been free after the final task was created. So the memory may be used by other function and the content of oto1 will be changed randomly. To solve this question, you can set `oto oto1;` as a global variable, or don't stop `app_main()` function.  Hope it can help you.

Alvin1Zhang, 2020-05-27:
@jonhunterking66 Thanks for reporting. Would you please help share if any updates for the issue? Thanks.

jonhunterking66, 2020-05-27:
Hey @Alvin1Zhang I found a work around to solve this issue. I did not try and stick with this method. I will close the ticket.

closed 2020-05-27 by jonhunterking66

Alvin1Zhang, 2020-05-28:
@jonhunterking66 Thanks for sharing the updates.