espressif/esp-idf #4526: The timestamp of the captured packet in sniffer mode is not updated for a long time

wgcod, 2019-12-20:
When I use sniffer mode to capture some WiFi packets, I find that sometimes the timestamps of dozens of consecutive packets are the same, lasting even 300ms. Is this normal? Time is not accurate enough? Or ampdu? But I use Wireshark to capture packets on Linux. The time of each packet is different. I want to make sure. Thank you.

zhangyanjiaoesp, 2019-12-20:
@wgcod Please describe how you call the sniffer API, and provide the logs of your test.

wgcod, 2019-12-20:
The test is very simple. I used the sniffer demo to output the time stamp of each package in the callback function, as you can see.

zhangyanjiaoesp, 2019-12-20:
@wgcod Have you enabled the modem sleep or light sleep? There is description in IDF:
 https://github.com/espressif/esp-idf/blob/30372f5a4ff2c0dfdaad95f544dc36bcdda30b75/components/esp_wifi/include/esp_wifi_types.h#L335

wgcod, 2019-12-24:
No, I test with demo without any modification. Save the data to SD card and open it with Wireshark. Sometimes, a dozen or more packet timestamps are the same. When I capture packets with Wireshark at the same time, I can see that the time of these packets is different, but only 2-3us is related between packets. Is it because the clock is not accurate. If so, is it configurable? You can try it. Just use demo.

zhangyanjiaoesp, 2020-07-15:
hi @wgcod ,
I have tested with `simple_sniffer` example in IDF, and open the capture packets with Wireshark, for some packets the timestamps are the same, but only about a dozen milliseconds. And the time inaccuracy is  acceptable for ESP32. 
If you want the timing to be particularly accurate, you can try our new chip ESP32S2.
https://www.espressif.com/en/products/socs/esp32-s2

[image]

Alvin1Zhang, 2020-07-17:
Thanks for reporting and sorry for slow turnaround, feel free to reopen if the issue still happens. Thanks.

closed 2020-07-17 by Alvin1Zhang