espressif/esp-idf #4527: Need timestamp while receiving CAN messages

keelung-yang, 2019-12-20:
Since CAN messages are in RX queue and can_receive() is called in some task, the timestamp of returning from can_receive() is not the real time of receiving from bus.

On vehicle, there are many messages sent to bus in cycles. If we want show them by calling can_receive(), the cycles are not accurate as they should be. So, it's better to add timestamp to a received CAN message.

keelung-yang, 2019-12-20:
Here is what I got by using timestamp on can_receive() returned successfullly.
[image]

keelung-yang, 2019-12-26:
Here is the performance of a third-part tool:
[image]

And what I got by using
```c
can_receive(&msg, pdMS_TO_TICKS(10);
```
[image]

Dazza0, 2019-12-26:
@keelung-yang I was thinking of adding a `can_get_diagnostics()` which would read some sort of event queue. Each event queue would contain an event ID and a time stamp. Would this meet your requirements?

keelung-yang, 2019-12-27:
@Dazza0 As what I understand, can_get_diagnostics() and events are for general using, events cann't match with each received message. So, it's cann't help with accurate received timestamp.

Dazza0, 2020-03-13:
@keelung-yang Unfortunately the hardware does not store any time information when it receives a message. All it does is place the message into RX FIFO. If you require the exact time when the message is considered successfully received (i.e., the moment the message is considered valid as per the CAN2.0B protocol), this can only achieve by a hardware change.

However, if you can accept a less precise time stamp, the current system time can be collected by the ISR when in reads messages from the RX FIFO. However, this precision of an ISR time stamp will be dependent on the latency of the ISR. This can be affected by:

- Other higher priority ISRs that need to run before the CAN ISR
- Critical sections that disable interrupts

Would an ISR time stamp be precise enough for your application?

keelung-yang, 2020-03-30:
@Dazza0 
Since ISR is a bad solution, just forget it please. Drivers should be keep generial.
And, where are the documents about CAN hardware you referred? I checked some datasheet and reference manual shortly, no enough infomation found just now.

Dazza0, 2020-04-07:
> And, where are the documents about CAN hardware you referred? I checked some datasheet and reference manual shortly, no enough infomation found just now.

We don't have a chapter on regarding CAN in the TRM, however the features and interface of the CAN peripheral is nearly identical to the SJA1000.

Alvin1Zhang, 2021-05-25:
Thanks for reporting, feel free to reopen.

Dazza0, 2021-09-29:
Reopening this in-case HW time-stamp added to future chips

keelung-yang, 2025-11-16:
> Reopening this in-case HW time-stamp added to future chips

@Dazza0 Thanks for reopening! The HW time-stamp is critical in automotive industry.
I'm happy if there are chips which support this feature.
I know, there are many many works needed to be done to bring ESP32 into automotive industry.
That's the market policy of Espressif, I think.