espressif/esp-idf #4529: MQTT event posting delay in v4.1-dev-1572-g30372f5a4

PranavD10, 2019-12-20:
## Environment

- Development Kit: ESP32-DevKit
- Kit version : v1
- Module or chip used:  ESP32-WROOM-32
- IDF version :  v4.1-dev-1572-g30372f5a4
- Build System: Make
- Compiler version : xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
- Operating System: Linux
- Power Supply: USB

## Problem Description
I am working on a project which uses mqtt, to send and receive the data. The project was working fine, but after i updated the esp-idf and installed the tools by running install.sh, the project is facing issues. The issue is the mqtt messages are received after a significant amount of inconsistent delay.

### Expected Behavior
The mqtt messages should be received without delay.
### Actual Behavior

### Steps to repropduce

run the mqtt/tcp example

ESP-Marius, 2019-12-20:
Hi,

I've tested this, but I am unable to reproduce it. When I tested this commit I get consistent and low delays.

Could you provide more info about your setup. Are you testing against the eclipse broker used in the example?

PranavD10, 2019-12-20:
Hi,
I have tested the same thing with 2 different mqtt servers both were local mqtt servers.
The mqtt server is sending the messages to the other machines in real time. the problem i am facing is only after i updated the idf.
Also if i send multiple messages one after another, it is receiving all the message data at once (like multiple mqtt data events), after the fixed delay.
I think the mqtt task is not posting the data to the handler in real time.
Also can you tell me where i can find the change log of this update.
Thank you

ESP-Marius, 2019-12-23:
Hmm, I can't see how the MQTT client would buffer these events.

Just to make it clear, you are having the same problem when running the mqtt tcp example right? From which idf version did you update?

As for the changelogs: for our releases you can find the changelogs [here](https://github.com/espressif/esp-idf/releases), but there are no changelogs available for the master branch updates.

Alvin1Zhang, 2020-05-27:
@PranavD10 Thanks for reporting and sharing the details. Would you please help share if any updates for this issue? Thanks.

Alvin1Zhang, 2020-07-01:
@PranavD10 Thanks for reporting, will close now due to short of feedback, feel free to reopen with more updates. Thanks.

closed 2020-07-01 by Alvin1Zhang