espressif/esp-idf #4533: led_strip  example only works with Level Shifter

ISOMAX, 2019-12-21:
Hello, 

the Example 
https://github.com/espressif/esp-idf/tree/master/examples/peripherals/rmt/led_strip
only works when i add a Level Shifter. 

i use a simple 74AXP1T34GWH from Nexperia along with 2 pieces of 220Ohm Resistors on the Input and output in series. 

this gives a perfect Rainbow.

suda-morris, 2019-12-23:
Sorry in my hardware WS2812 is 3.3V capable.
From WS2812 datasheet, VIH should not smaller than 0.7VDD, and VIL should not greater than 0.3VDD. 
Maybe your hardware can't meet the limitation?

Alvin1Zhang, 2020-01-03:
@ISOMAX Thanks for reporting, would you please help share if any updates for this issue? Thanks.

ISOMAX, 2020-01-11:
I will check this again, OK ?

Alvin1Zhang, 2020-01-13:
@ISOMAX Many thanks for the updates.

closed 2020-01-21 by Alvin1Zhang