espressif/esp-idf #4535: Why are my firmware performance got worse after upgrade to esp-idf stable version?

elad30300, 2019-12-22:
I use esp32-wroom-32 chip. My product uses ble. Until now - I used esp-idf version 3.0, but because of lack of features that exist in the stable version I updated the esp-idf version to the stable version (currently 3.3.1 according to docs). I've been noticing that the first ble operations sometimes take more time and thus delay the system start, I get the following error message: "BT_HCI: command_timed_out hci layer timeout waiting for response to a command. opcode:" when the opcode varies and there is no stable value.

I would really like to know if someone knows how to solve this problem.

drony, 2019-12-24:
I think nobody read changes. please try to set this esp_wifi_set_ps(WIFI_PS_NONE)

elad30300, 2019-12-25:
when I do that I get linking error region `dram0_0_seg' overflowed by 3576 bytes which indicates for too much memory statically allocated, when I don't call the function (but still icnlude "esp_wifi.h") I don't get it.
Just to clarify - I don't use WIFI in my code at all besided esp_wifi_set_ps(WIFI_PS_NONE).

GYC-Espressif, 2019-12-26:
Can you provide the commit ID you used? Thank you.

UtkarshBhatthere, 2020-01-01:
Changing IDF versions in between development cycles is a taboo. I think you should refactor your application code according to newer IDF version (If the changelog shows any major changes in the Libraries being used.)

elad30300, 2020-01-01:
Hi, thank you guys for the comments, I managed to (probably, still in testing) by look at the examples project configuration and according to them i changed the cpu in which the bluetooth task runs on.

closed 2020-01-09 by Alvin1Zhang