espressif/esp-idf #4539: BT with light sleep enabled works when connected with some devices but not others

Kemsa, 2019-12-23:
## Environment

- Module or chip used: ESP32-WROOM-32D
- IDF version Release 4.0 compiled with visualGDB
- Build System:         Make
- Operating System:     Windows
- Power Supply:      external lab power supply

## Problem Description
Hello,

I have developped a firmware that uses the light sleep function with bluetooth. I have been quite successful so far, with a power consumption between 5 and 10mA when connected to my smartphone or computer. This is the expected behaviour. But if I connect the esp32 to a motorola MOTOTRBO radio, the light sleep is never activated, the power consumption remains stable at 100mA (although the current is OK during the advertising stage. It goes up after connexion).
I have checked that no frames are exchanged at application level (I can't say anything about the protocol layer, my knowledge of the protocol is not good enough).

Do you have any insight on why the esp32 would not go to sleep? What are the locks that prevent the light sleep?
I think the radio I try to connect to is using bluetooth 4.0, I don't know if it can be the source of the problem.

Wth-Esp, 2020-06-17:
Hi @Kemsa 

Could you please describe the functions you use? Classic Bluetooth or BLE or BT/BLE coex?

Thanks
Weitianhua

Wth-Esp, 2020-06-17:
Hi @Kemsa 

And here is some doc for reference.

[power management](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/power_management.html)

[sleep mode](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/sleep_modes.html)

Thanks
Weitianhua

Wth-Esp, 2020-06-17:
Hi @Kemsa

And the sleep mode related configuration suggestion is here, please try this.

[ESP-IDF sleep mode menuconfig options.pdf](https://github.com/espressif/esp-idf/files/4790205/ESP-IDF.sleep.mode.menuconfig.options.pdf)

Thanks
Weitianhua

Alvin1Zhang, 2020-08-19:
@Kemsa Thanks for reporting and sorry for slow turnaround, would you please help share if any updates for this issue? Thanks.

Kemsa, 2020-08-19:
Hi, the problem came from the connecting device and not from the esp32, it's all fixed now, but thanks for enquiring.

closed 2020-08-19 by Kemsa