espressif/esp-idf #4540: CMake: Source Code Generation fails with "add_custom_command command is not scriptable"

mringwal, 2019-12-23:
Hi

I need to generate source code as part of the build process.

In the current documentation, the use of the CMake add_custom_command" is recommended:

https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/build-system.html#source-code-generation

However, if I add this to esp-idf/example/get-started/blink/main/CMakeLists.txt

```
add_custom_command(
     OUTPUT logo.h
     COMMAND echo "test" > logo.h
     )
```
and run `idf.py build`, I get this error

```
Make Error at /Projects/esp32/esp-idf/tools/cmake/component.cmake:222 (message):
  CMake Error at
  /Projects/esp32/esp-idf/examples/get-started/blink/main/CMakeLists.txt:1
  (add_custom_command):

    add_custom_command command is not scriptable
```

Any ideas on how to run a script to create a source file? Thanks!

This is on the current master branch: 30372f5a4ff2c0dfdaad95f544dc36bcdda30b75

igrr, 2019-12-23:
Can you try wrapping this in `if(NOT CMAKE_BUILD_EARLY_EXPANSION)`, as in the following snippet?

https://github.com/espressif/esp-idf/blob/9dbb7526c5d140b22529a1cfb0b4b46a0ab288aa/components/soc/esp32s2beta/sources.cmake#L23-L26

mringwal, 2019-12-23:
Thanks for the hint!

With ``if(NOT CMAKE_BUILD_EARLY_EXPANSION)``, the command is executed. That should be enough to get it working.

${COMPONENT_LIB} does not seem to be defined, hence the ``add_dependencies(${COMPONENT_LIB} logo)`` fails.

It would be great, it the ``if(NOT CMAKE_BUILD_EARLY_EXPANSION)`` could be added to the documentation.

mringwal, 2020-02-19:
Sorry to keep it open so long, works for me. Thanks.

closed 2020-02-19 by mringwal