espressif/esp-idf #4541: setjmp doesn't work (Double Exception)

szmodz, 2019-12-23:
toolchain: 2019r2
esp-idf: latest master (30372f5a4ff2c0dfdaad95f544dc36bcdda30b75)
hardware: ESP32-DevKitC WROVER-B

test case:
https://github.com/szmodz/esp-idf/tree/master/examples/get-started/blink

https://github.com/szmodz/esp-idf/commit/e2befd3236291e7877d02099c1d1e12708e6988c

I'm not sure if it depends on config options, but I've included the config just in case.

szmodz, 2019-12-23:
output:

> start
> Guru Meditation Error: Core  0 panic'ed (Unhandled debug exception)
> Debug exception reason: BREAK instr 
> Core 0 register dump:
> PC      : 0x400803c0  PS      : 0x00060a36  A0      : 0x800d4b24  A1      : 0x3ffb42d0  
> 0x400803c0: _DoubleExceptionVector at /Users/szmodz/xxx/clu_hw/esp-idf/components/freertos/xtensa_vectors.S:542
> 
> A2      : 0x00000000  A3      : 0x3ffb007c  A4      : 0x3ffb42e0  A5      : 0x000000fe  
> A6      : 0x00000001  A7      : 0x00000000  A8      : 0x8008bcde  A9      : 0x3ffb42a0  
> A10     : 0x00000000  A11     : 0x00000032  A12     : 0x00060620  A13     : 0x000000fe  
> A14     : 0x00000001  A15     : 0x00000000  SAR     : 0x00000014  EXCCAUSE: 0x00000001  
> EXCVADDR: 0x000000ee  LBEG    : 0x40085fbd  LEND    : 0x40085fcd  LCOUNT  : 0xfffffffe  
> 0x40085fbd: srand at /builds/idf/crosstool-NG/.build/HOST-x86_64-apple-darwin12/xtensa-esp32-elf/src/newlib/newlib/libc/stdlib/rand.c:67
> 
> 0x40085fcd: rand at /builds/idf/crosstool-NG/.build/HOST-x86_64-apple-darwin12/xtensa-esp32-elf/src/newlib/newlib/libc/stdlib/rand.c:72
> 
> 
> ELF file SHA256: 06a4d9f244c0a65c7dca5bf5f7ec2dd089e7ffbf41595da03462069c3d41709d
> 
> Backtrace: 0x400803bd:0x3ffb42d0 0x400d4b21:0x3ffb42e0 0x400d0b62:0x3ffb4350 0x4008c741:0x3ffb4370
> 0x400803bd: _UserExceptionVector at ??:?
> 
> 0x400d4b21: tableToXml at /Users/szmodz/xxx/clu_hw/clu_core/modules/class/http/lxml.c:125 (discriminator 2)
> 
> 0x400d0b62: _stext at ??:?
> 
> 0x4008c741: rtc_clk_cal_internal at /Users/szmodz/xxx/clu_hw/esp-idf/components/soc/esp32/rtc_time.c:61
> 
> 
> Rebooting...
> ets Jun  8 2016 00:22:57
>

szmodz, 2019-12-23:
this is very interesting:
> 
0x40085fcd: rand at /builds/idf/crosstool-NG/.build/HOST-x86_64-apple-darwin12/xtensa-esp32-elf/src/newlib/newlib/libc/stdlib/rand.c:72

szmodz, 2019-12-23:
my application uses Lua, which relies on setjmp/longjmp, so this is a show stopper.

szmodz, 2019-12-23:
it worked fine with the previous master (93a8603c545fb8e54741d6685146e2f3b874378d)

igrr, 2019-12-23:
Can you please try reverting https://github.com/espressif/esp-idf/commit/a7aea569773c9eb281483b3840b84033d934ba3f and see if the issue still happens?

szmodz, 2019-12-23:
> error: commit a7aea569773c9eb281483b3840b84033d934ba3f is a merge but no -m option was given.

just to be sure: could you specify the exact command?

szmodz, 2019-12-27:
Seems to work with a7aea569773c9eb281483b3840b84033d934ba3f reverted.

igrr, 2019-12-27:
Thanks for confirming! We have the fix for this issue in review. It may take a few days due to the holidays.

closed 2020-01-06 by espressif-bot