espressif/esp-idf #4543: IDF Monitor refers to host source

szmodz, 2019-12-23:
see https://github.com/espressif/esp-idf/issues/4541

> 0x40085fcd: rand at /builds/idf/crosstool-NG/.build/HOST-x86_64-apple-darwin12/xtensa-esp32-elf/src/newlib/newlib/libc/stdlib/rand.c:72

szmodz, 2019-12-23:
It's the ".build/HOST-x86_64-apple-darwin12" part that worries me. I'm not an expert on crosstool-NG, but I believe we're dealing with target libraries here.

szmodz, 2019-12-23:
In case this is normal, still doesn't invalidate #4541

igrr, 2019-12-27:
crosstool-NG puts the build directory into `${CT_WORK_DIR}/${CT_HOST:+HOST-${CT_HOST}/}${CT_TARGET}`, which makes the build directory specific for the given host-target combination (https://github.com/espressif/crosstool-NG/blob/72ba393991f2eac2df3dbd3207d4eec79b8c42a3/config/global/paths.in#L45). The source code of GCC and newlib is extracted into this directory and patched. When newlib is compiled, the directory name gets embedded into the debug information.

I suppose it would be more clear if the default was `${CT_WORK_DIR}/${CT_HOST:+HOST-${CT_HOST}/}TARGET-${CT_TARGET}`, resulting in a directory like `.build/HOST-x86_64-apple-darwin12/TARGET-xtensa-esp32-elf/`. But this option is not configurable, so we can't override it at build time.

closed 2020-07-24 by igrr