espressif/esp-idf #4545: Crash in I2S when use ESP_INTR_FLAG_IRAM flag in the driver config

phonec, 2019-12-26:
## Environment

- Development Kit: WEMOS LOLIN32 Pro
- Module: ESP32-WROVER chip rev1
- IDF version: v3.3.1 
- Operating System: Linux Ubuntu
- Build System:  Make
- Power Supply:  USB

## Problem Description
When adding a flag ESP_INTR_FLAG_IRAM to the *.intr_alloc_flags* configuration of i2s driver leads to a crash, because at runtime not all of the data from the *i2s_intr_handler_default()* interrupt handler function can be found in IRAM, although the function itself has such an attribute: IRAM_ATTR.
In particular, the array of pointers I2S[] does not have an IRAM_ATTR attribute and may not be available when the cache is disabled during such an interrupt.

## Solving
Adding this attribute to the I2S pointer on line 102: esp-idf/components/driver/i2s.c
```
102    static IRAM_ATTR i2s_dev_t * I2S [I2S_NUM_MAX] = {&I2S0, &I2S1};
```
perhaps it should completely solve this problem, at least it works for me.

## Steps to repropduce
Just add this flag to the configuration, for example in the *esp-idf/examples/ peripherals/i2s/main/i2s_example_main.c*

## Debug Logs
```
Guru Meditation Error: Core  0 panic'ed (Cache disabled but cached memory region accessed)
Core 0 register dump:
PC      : 0x4008753c  PS      : 0x00050034  A0      : 0x40083e10  A1      : 0x3ffb0670
...
```
```
(gdb) list *0x4008753c
0x4008753c is in i2s_intr_handler_default (/home/ESP32/esp-idf/components/driver/i2s.c:520).
515
516         portBASE_TYPE high_priority_task_awoken = 0;
517
518         lldesc_t *finish_desc;
519
520         if (i2s_reg->int_st.out_dscr_err || i2s_reg->int_st.in_dscr_err) {
521             ESP_EARLY_LOGE(I2S_TAG, "dma error, interrupt status: 0x%08x", i2s_reg->int_st.val);
522             if (p_i2s->i2s_queue) {
523                 i2s_event.type = I2S_EVENT_DMA_ERROR;
524                 if (xQueueIsQueueFullFromISR(p_i2s->i2s_queue)) {
```

xiongyumail, 2019-12-31:
@phonec 

Thanks for the problem you found, it may be better to use `DRAM_ATTR`. We will merge the changes into the corresponding branches later.

phonec, 2020-01-06:
Great. The placement of the I2S interrupt handler in IRAM is necessary to avoid sound distortion during other long-term SPI FLASH operations.

phonec, 2020-01-11:
it would be nice to see it also in the 3.3.x

ctag-fh-kiel, 2020-01-16:
I have similar issues. The fix is great, but I haven't been able to my external codec (WM8731) run glitch free when reading a file from flash and serving the file's content with the http server... 
Played with the i2s buffers -> no success, played with the i2s INTR flags --> no success
Any advice is much appreciated. WM8731 running at 44.1kHz...

ctag-fh-kiel, 2020-01-18:
My issue was in fact related to the http server core id. I configured the httpd to use core 0 and i2s core 1. It was unrelated to flash access.

Alvin1Zhang, 2020-04-10:
@phonec @ctag-fh-kiel The issue has been fixed and fix available on GitHub. Please help try if the issue still happens, feel free to reopen if the issue persists. Thanks.

closed 2020-04-10 by Alvin1Zhang