espressif/esp-idf #4546: BLE例程在连接时总是不成功

linrx1511, 2019-12-27:
有问题的都是bluedroid的例程，ble_spp_server，ble_throughput等，手机是华为P20，
Esp32控制台显示已经连接上，但上位机好像获取不到服务，试了几种BLE调试器都似一样效果
```
--- idf_monitor on COM6 921600 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                              I (28) boot: ESP-IDF v4.1-dev-256-g9f145ff16-dirty 2nd stage bootloader
I (29) boot: compile time 09:40:00
I (29) boot: Enabling RNG early entropy source...
I (29) boot: SPI Speed      : 40MHz
I (30) boot: SPI Mode       : DIO
I (30) boot: SPI Flash Size : 4MB
I (31) boot: Partition Table:
I (31) boot: ## Label            Usage          Type ST Offset   Length
I (32) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (33) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (34) boot:  2 factory          factory app      00 00 00010000 00100000
I (35) boot: End of partition table
I (35) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x1bc30 (113712) map
I (78) esp_image: segment 1: paddr=0x0002bc58 vaddr=0x3ffbdb60 size=0x02f50 ( 12112) load
I (83) esp_image: segment 2: paddr=0x0002ebb0 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at D:/projects/esp/esp-idf/components/freertos/xtensa_vectors.S:1778

I (84) esp_image: segment 3: paddr=0x0002efb8 vaddr=0x40080400 size=0x01058 (  4184) load
I (86) esp_image: segment 4: paddr=0x00030018 vaddr=0x400d0018 size=0x729d0 (469456) map
0x400d0018: _stext at ??:?

I (255) esp_image: segment 5: paddr=0x000a29f0 vaddr=0x40081458 size=0x10fb8 ( 69560) load
0x40081458: call_start_cpu0 at D:/projects/esp/esp-idf/components/esp32/cpu_start.c:222

I (296) boot: Loaded app from partition at offset 0x10000
I (296) boot: Disabling RNG early entropy source...
I (297) cpu_start: Pro cpu up.
I (297) cpu_start: Application information:
I (297) cpu_start: Project name:     throughput_server_demo
I (298) cpu_start: App version:      1
I (299) cpu_start: Compile time:     Dec 27 2019 10:19:21
I (300) cpu_start: ELF file SHA256:  b712062dae6c17e2...
I (300) cpu_start: ESP-IDF:          v4.1-dev-256-g9f145ff16-dirty
I (301) cpu_start: Starting app cpu, entry point is 0x40081288
0x40081288: call_start_cpu1 at D:/projects/esp/esp-idf/components/esp32/cpu_start.c:271

I (0) cpu_start: App cpu up.
I (303) heap_init: Initializing. RAM available for dynamic allocation:
I (303) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (304) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (305) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (306) heap_init: At 3FFBDB5C len 00000004 (0 KiB): DRAM
I (306) heap_init: At 3FFC8D30 len 000172D0 (92 KiB): DRAM
I (307) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (308) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (309) heap_init: At 40092410 len 0000DBF0 (54 KiB): IRAM
I (309) cpu_start: Pro cpu start user code
I (320) spi_flash: detected chip: generic
I (320) spi_flash: flash io: dio
I (320) cpu_start: Chip Revision: 1
I (320) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (365) BTDM_INIT: BT controller compile version [eec6ee1]
I (365) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (455) phy: phy_version: 4102, 2fa7a43, Jul 15 2019, 13:06:06, 0, 0
I (645) GATTS_DEMO: REGISTER_APP_EVT, status 0, app_id 0

W (645) BT_BTM: BTM_BleWriteAdvData, Partial data write into ADV
W (645) BT_BTM: BTM_BleWriteScanRsp, Partial data write into ADV
I (645) GATTS_DEMO: CREATE_SERVICE_EVT, status 0,  service_handle 40

I (645) GATTS_DEMO: SERVICE_START_EVT, status 0, service_handle 40

I (655) GATTS_DEMO: ADD_CHAR_EVT, status 0,  attr_handle 42, service_handle 40

I (655) GATTS_DEMO: the gatts demo char length = 3

I (655) GATTS_DEMO: prf_char[0] =11

I (655) GATTS_DEMO: prf_char[1] =22

I (655) GATTS_DEMO: prf_char[2] =33

I (655) GATTS_DEMO: ADD_DESCR_EVT, status 0, attr_handle 43, service_handle 40

I (28055) GATTS_DEMO: ESP_GATTS_CONNECT_EVT, conn_id 0, remote 66:32:72:d3:48:6b:
I (43455) GATTS_DEMO: ESP_GATTS_DISCONNECT_EVT
I (66565) GATTS_DEMO: ESP_GATTS_CONNECT_EVT, conn_id 0, remote 66:32:72:d3:48:6b:
I (89775) GATTS_DEMO: ESP_GATTS_DISCONNECT_EVT
```

GYC-Espressif, 2019-12-30:
Hello,这是一个已经被修复的问题，请您使用 `git pull` 拉取最新的 IDF 解决，谢谢。

linrx1511, 2019-12-31:
> Hello,这是一个已经被修复的问题，请您使用 `git pull` 拉取最新的 IDF 解决，谢谢。

拉取了最新的，编译到最后一步出错,帮忙看看是什么原因，谢谢。
```
Generated D:/projects/esp/ble_throughput/throughput_server/build/bootloader/bootloader.bin
[1150/1151] Linking CXX executable throughput_server_demo.elf
FAILED: throughput_server_demo.elf
cmd.exe /C "cd . && C:\Users\lenovo\.espressif\tools\xtensa-esp32-elf\esp32-2019r1-8.2.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-g++.exe  -mlongcalls -Wno-frame-address  -nostdlib @CMakeFiles\throughput_server_demo.elf.rsp  -o throughput_server_demo.elf  && cd ."
c:/users/lenovo/.espressif/tools/xtensa-esp32-elf/esp32-2019r1-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: c:/users/lenovo/.espressif/tools/xtensa-esp32-elf/esp32-2019r1-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/lib\libstdc++.a(eh_personality.o):(.literal._ZL21base_of_encoded_valuehP15_Unwind_Context+0x0): undefined reference to `_Unwind_GetTextRelBase'
c:/users/lenovo/.espressif/tools/xtensa-esp32-elf/esp32-2019r1-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: c:/users/lenovo/.espressif/tools/xtensa-esp32-elf/esp32-2019r1-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/lib\libstdc++.a(eh_personality.o):(.literal._ZL21base_of_encoded_valuehP15_Unwind_Context+0x4): undefined reference to `_Unwind_GetDataRelBase'
c:/users/lenovo/.espressif/tools/xtensa-esp32-elf/esp32-2019r1-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: c:/users/lenovo/.espressif/tools/xtensa-esp32-elf/esp32-2019r1-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/lib\libstdc++.a(eh_personality.o):(.literal._ZL21base_of_encoded_valuehP15_Unwind_Context+0x8): undefined reference to `_Unwind_GetRegionStart'
c:/users/lenovo/.espressif/tools/xtensa-esp32-elf/esp32-2019r1-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: c:/users/lenovo/.espressif/tools/xtensa-esp32-elf/esp32-2019r1-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/lib\libstdc++.a(eh_personality.o):(.literal.__gxx_personality_v0+0x8): undefined reference to `_Unwind_GetLanguageSpecificData'
c:/users/lenovo/.espressif/tools/xtensa-esp32-elf/esp32-2019r1-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: c:/users/lenovo/.espressif/tools/xtensa-esp32-elf/esp32-2019r1-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/lib\libstdc++.a(eh_personality.o):(.literal.__gxx_personality_v0+0xc): undefined reference to `_Unwind_GetIPInfo'
c:/users/lenovo/.espressif/tools/xtensa-esp32-elf/esp32-2019r1-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: c:/users/lenovo/.espressif/tools/xtensa-esp32-elf/esp32-2019r1-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/lib\libstdc++.a(eh_personality.o):(.literal.__gxx_personality_v0+0x10): undefined reference to `_Unwind_SetGR'
c:/users/lenovo/.espressif/tools/xtensa-esp32-elf/esp32-2019r1-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: c:/users/lenovo/.espressif/tools/xtensa-esp32-elf/esp32-2019r1-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/lib\libstdc++.a(eh_personality.o):(.literal.__gxx_personality_v0+0x14): undefined reference to `_Unwind_SetIP'
c:/users/lenovo/.espressif/tools/xtensa-esp32-elf/esp32-2019r1-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: c:/users/lenovo/.espressif/tools/xtensa-esp32-elf/esp32-2019r1-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/lib\libstdc++.a(eh_personality.o): in function `base_of_encoded_value(unsigned char, _Unwind_Context*)':
/builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/xtensa-esp32-elf/src/gcc/libstdc++-v3/../libgcc/unwind-pe.h:114: undefined reference to `_Unwind_GetTextRelBase'
c:/users/lenovo/.espressif/tools/xtensa-esp32-elf/esp32-2019r1-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: /builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/xtensa-esp32-elf/src/gcc/libstdc++-v3/../libgcc/unwind-pe.h:116: undefined reference to `_Unwind_GetDataRelBase'
c:/users/lenovo/.espressif/tools/xtensa-esp32-elf/esp32-2019r1-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: /builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/xtensa-esp32-elf/src/gcc/libstdc++-v3/../libgcc/unwind-pe.h:118: undefined reference to `_Unwind_GetRegionStart'
c:/users/lenovo/.espressif/tools/xtensa-esp32-elf/esp32-2019r1-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: c:/users/lenovo/.espressif/tools/xtensa-esp32-elf/esp32-2019r1-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/lib\libstdc++.a(eh_personality.o): in function `parse_lsda_header(_Unwind_Context*, unsigned char const*, lsda_header_info*)':
[... cut ...]
c:/users/lenovo/.espressif/tools/xtensa-esp32-elf/esp32-2019r1-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: c:/users/lenovo/.espressif/tools/xtensa-esp32-elf/esp32-2019r1-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/lib\libstdc++.a(eh_globals.o): in function `eh_globals_dtor(void*)':
/builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/eh_globals.cc:83: undefined reference to `_Unwind_DeleteException'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```

linrx1511, 2020-01-01:
问题已解决，工具更新后就可以了，谢谢

closed 2020-01-01 by linrx1511