espressif/esp-idf #4547: fopen execution time more than 100ms after calling frequently

SophistCedar, 2019-12-27:
Environment
•	Chip used: ESP32D0WDQ6
•	Peripheral used: Kingston SD card(16G)
•	IDF version: v3.3
•	Power Supply: USB
Problem Description
I'm suffering fopen execution time becoming longer with its called times increasing. After calling fopen for many times (around 2000 or more), the fopen execution time increase to 110ms or more from 1ms in the beginning. However, there is no exception and it completely works as I expected. I can't find any other similar case like this and it is really hard to figure it out because it can work well but the execution time becomes longer and longer. And if I already put some files in the directory (around 2000 files), the fopen execution time is 116ms in the beginning which is unacceptable. 
I have already tested the SD card in my computer and called fopen more than 10000 times to create files in the SD card and the execution is always 5ms. So it seems not the SD card reason. If there's a solution it would be great and I'd like to know what is the reason and how to debug like this.
Code to reproduce this issue

#include <stdio.h>
#include <stdlib.h>
#include <string.h>
#include <math.h>
#include <unistd.h>//usleep使用
#include <sys/unistd.h>
#include <sys/stat.h>
#include <time.h>
#include <sys/time.h>
#include "esp_vfs_fat.h"
#include "driver/sdmmc_host.h"
#include "driver/sdspi_host.h"
#include "sdmmc_cmd.h"
#include <dirent.h> 
#include <sys/types.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_system.h"
#include "driver/spi_master.h"
#include "driver/gpio.h"
#include <esp_log.h>
#include "esp_err.h"
#include <esp_heap_caps.h>

static char tag[] = "sdcard";   //声明静态字符串
void sdcard_init(void){
    sdmmc_host_t host = SDSPI_HOST_DEFAULT();
    sdspi_slot_config_t slot_config = SDSPI_SLOT_CONFIG_DEFAULT();

    esp_vfs_fat_sdmmc_mount_config_t mount_config = {
        .format_if_mount_failed = true,
        .max_files = 5,
        .allocation_unit_size = 16 * 1024
    };
    sdmmc_card_t* card;
    esp_err_t ret = esp_vfs_fat_sdmmc_mount("/sdcard", &host, &slot_config, &mount_config, &card);

    if (ret != ESP_OK) {
        if (ret == ESP_FAIL) {
            ESP_LOGE(tag, "Failed to mount filesystem. "
                "If you want the card to be formatted, set format_if_mount_failed = true.");
        } else {
            ESP_LOGE(tag, "Failed to initialize the card (%s). "
                "Make sure SD card lines have pull-up resistors in place.", esp_err_to_name(ret));
        }
        return;
    }
}

void app_main(void)
{
    sdcard_init();
    DIR *dr = opendir("/sdcard/data");  
    if (dr == NULL)  
    { 
        ESP_LOGD(tag, "no data dir, trying to create");
        mkdir("/sdcard/data", 0777);
    } 
    char filename[80];
    int filenum = 100000;
    char str[] = "This is runoob.com";
    while(1){
                sprintf(filename, "/sdcard/data/file-%d.bin", filenum);
                filenum++;
                FILE *fp = NULL;
                            TickType_t sticks = xTaskGetTickCount();
                fp = fopen(filename, "wb");
                            TickType_t eticks = xTaskGetTickCount();
                            int tdiff = eticks-sticks;
                            ESP_LOGI(tag,"fopen %d\n", tdiff);

                ESP_LOGI(tag,"open new file %s\n", filename);
                fwrite(str, sizeof(str) , 1, fp );
                fclose(fp);
                usleep(1000);
    }
    esp_vfs_fat_sdmmc_unmount(); 
}

SophistCedar, 2019-12-27:
[image]
**as the picture shows, the fopen execution time is 241ms after calling around 2300 times.**

igrr, 2019-12-27:
Do I understand correctly that in your test the execution time seems to depend on the number of files stored on the SD card? If the number of the files is the same, will the duration of `fopen` call be the same, after repeated calls?

SophistCedar, 2019-12-27:
In my test, the execution time in the beginning depend on the number of files stored on the SD card. And the duration of fopen call always increases after repeated calls.

igrr, 2019-12-27:
> And the duration of fopen call always increases after repeated calls.

Does this happen if you don't create new files? I.e. always fopen the same file.

SophistCedar, 2019-12-27:
No. the execution time is stable if fopen donot create new files.

SophistCedar, 2019-12-27:
> Does this happen if you don't create new files? I.e. always fopen the same file.
if fopen donot create new files, just shorten existed files to length zero, the execution time depends on the files stored on the SD card and wonot increase.

stickbreaker, 2019-12-28:
@SophistCedar You are not encountering a bug.  When you add files to a directory, the FAT file system has to verify that your new file does not already exist. The only way to verify it is not a duplicate is to check every existing file.  So it has to read every entry in the directory.  To reduce this latency you are going to have to reduce the number of compare operations the file system has to do.  One way is to break your single directory with 2k+ files into multiple directories with 128..256 files (one directory page, dependent on implementation of fs). 

Chuck.

SophistCedar, 2019-12-28:
@stickbreaker Thank you so much.
I still have a question. If fopen existed files, how does the FAT file system work? Does it also need to read every entry in the directory to verify it is a duplicate? Or the FAT file system could find the existed file quickly without reading every entry based on certain setting. Because in my test, the fopen execution time won't increase when open the existed files after repeated calls.

stickbreaker, 2019-12-28:
@SophistCedar the file system keeps a small cache of the last accessed directory in memory.  If you are reopening the same file it does not have to load a directory page from the SDcard.  When a new file is added the file system has to:
* verify the "new" file is not a duplicate name of an existing file.
Scan through every directory page associated with the current directory (/data/).
* find an unused entry in a directory page to store the "new" file name.
If there is no space in the current directory page it has to add a new directory page, allocate the directory cluster from the FAT table (reusing delete clusters first), link the new directory page to the last directory page.
* assign a unused cluster as the starting point for the "new" file.
to find the "unused" cluster, either the first entry in the "delete" chain of the FAT or if there are no "deleted" cluster, then it has to scan the FAT for an "unused" entry.
* update the "new" file directory record with size, date.

Reusing an existing file only needs:
* find the "reused" file in the directory
* update the "reused" file directory record with new size and modification date.

here is a good starting point for FAT file systems on [Wiki](https://en.wikipedia.org/wiki/Design_of_the_FAT_file_system)

Chuck.

SophistCedar, 2019-12-28:
@stickbreaker. Thanks. now i understand it.

closed 2019-12-28 by SophistCedar