espressif/esp-idf #4549: Use LED to indicate CAN status

keelung-yang, 2019-12-27:
We want to use LED to indicate CAN controller status: _busy_, _idle_ and _error_

Since there is an API _can_get_status_info()_, we know when _error_ occurred. But how to determine _busy_ and _idle_? _msgs_to_tx_ and _msgs_to_rx_  in _can_status_info_t_ are not appropriate to handle _busy_ state since there are changed quickly then LED

Dazza0, 2020-03-13:
@keelung-yang 

> Also there is no interruption mentioned in CAN document.

The interrupts are handled by the driver, and not exposed to the users

> Since there is an API can_get_status_info(), we know when error occurred. But how to determine busy and idle? msgs_to_tx and msgs_to_rx in can_status_info_t are not appropriate to handle busy state since there are changed quickly, laged behind can_transmit() , and hard to capture.

The hardware does not provide an interrupt for when the CAN bus becomes busy or free. However, it has two readable bits to indicate whether the CAN controller is currently transmitting or receiving respectively. I can add an API to inquire these status bits. But this will mean the application level will need to periodically poll these status bits. Depending on the frequency of polling, the bus may be busy in between polling samples.

keelung-yang, 2020-04-03:
@Dazza0 I think this solution is bad than ISR in #4527

Dazza0, 2020-04-07:
> @Dazza0 I think this solution is bad than ISR in #4527

I'm not quite sure what you mean?

keelung-yang, 2020-04-07:
> > @Dazza0 I think this solution is bad than ISR in #4527
> 
> I'm not quite sure what you mean?

Polling is bad then ISR

keelung-yang, 2025-11-17:
@wanckl The timestamp is a generic and basic feature. So, it's better to be implemented in drivers but end users.

wanckl, 2025-12-31:
@keelung-yang Hi, As timestamp support is in doing now, and LED status is able to impl already, we can close this issue now, thanks

closed 2025-12-31 by Alvin1Zhang