espressif/esp-idf #4550: ESP32 Mdns with Android Chrome

Yuvaraj4041, 2019-12-27:
Hi,

I am trying http web server with mDNS. As per ESP-IDF Programming guide if we want use mDNS with windows browsers **Bonjour** has to be installed.  I followed the same and i am able to access [http://esp32.local/](url).

But in my case i want to open same [http://esp32.local/](url) in android browsers. Is there any way to achieve it.

negativekelvin, 2019-12-31:
No, people have been asking to have this feature on Android for a long time.  You have to make an app or use captive portal if you are in AP mode.

https://bugs.chromium.org/p/chromium/issues/detail?id=405925
https://issuetracker.google.com/issues/36932514
https://issuetracker.google.com/issues/140786115

DheeranYuvaraj, 2020-01-02:
Dear @negativekelvin 
Thanks for the information. Do you have any app source code?

closed 2020-01-10 by DheeranYuvaraj