espressif/esp-idf #4552: Compile error on verbose level

Reiner1210, 2019-12-27:
Log info to V
`
In file included from /home/reiner/data/programm/esp/esp-idf/components/soc/include/hal/hal_defs.h:17,
                 from /home/reiner/data/programm/esp/esp-idf/components/soc/src/hal/sdio_slave_hal.c:23:
/home/reiner/data/programm/esp/esp-idf/components/soc/src/hal/sdio_slave_hal.c: In function 'sdio_slave_hal_send_reset_counter':
/home/reiner/data/programm/esp/esp-idf/components/log/include/esp_log.h:281:13: error: 'len' may be used uninitialized in this function [-Werror=maybe-uninitialized]
             ets_printf(LOG_FORMAT(log_tag_letter, format), esp_log_timestamp(), tag, ##__VA_ARGS__); \
             ^~~~~~~~~~
/home/reiner/data/programm/esp/esp-idf/components/soc/src/hal/sdio_slave_hal.c:375:14: note: 'len' was declared here
`

masterd89, 2019-12-29:
Whatching this issue. Happening here as well

Reiner1210, 2019-12-29:
You only need to initialize the len with zero :-)

tritao, 2020-01-17:
Getting the same issue on the `hello_world` sample.

gshmakov, 2020-02-11:
Same issue. Still not fixed.

closed 2020-02-21 by espressif-bot

arnowi, 2020-03-09:
Hi,

I'm still getting this issue on branch release/v4.1

`In file included from /home/arno/Documents/projects/esp-idf/components/soc/include/hal/hal_defs.h:17,
                 from /home/arno/Documents/projects/esp-idf/components/soc/src/hal/sdio_slave_hal.c:23:
/home/arno/Documents/projects/esp-idf/components/soc/src/hal/sdio_slave_hal.c: In function 'sdio_slave_hal_send_reset_counter':
/home/arno/Documents/projects/esp-idf/components/log/include/esp_log.h:281:13: error: 'len' may be used uninitialized in this function [-Werror=maybe-uninitialized]
             ets_printf(LOG_FORMAT(log_tag_letter, format), esp_log_timestamp(), tag, ##__VA_ARGS__); \
             ^~~~~~~~~~
/home/arno/Documents/projects/esp-idf/components/soc/src/hal/sdio_slave_hal.c:375:14: note: 'len' was declared here
     uint32_t len;
              ^~~
`

igrr, 2020-03-09:
@arnowi Thanks for the notice, we will backport the fix to release/v4.1 as well