espressif/esp-idf #4554: LR mode does not work in WIFI_MODE_APSTA

Reiner1210, 2019-12-28:
When I create a station in BGN mode and an AP in LR mode no client can connect to the LR access point (AP). Here the output from the LR station
`
V (53163) esp_adapter: thread sem get: sem=0x3ffc8da8
D (55613) event: running post WIFI_EVENT:5 with handler 0x4011e650 on loop 0x3ffbe228
0x4011e650: wifi_default_action_sta_disconnected at /home/reiner/esp/esp-idf/components/esp_wifi/src/wifi_default.c:103

D (55613) esp_netif_handlers: esp_netif action disconnected with netif0x3ffbf158 from event_id=5
D (55623) esp_netif_lwip: check: remote, if=0x3ffbf158 fn=0x400d7eb8
0x400d7eb8: esp_netif_down_api at /home/reiner/esp/esp-idf/components/esp_netif/lwip/esp_netif_lwip.c:1070

D (55623) esp_netif_lwip: esp_netif_down_api esp_netif:0x3ffbf158
D (55633) esp_netif_lwip: esp_netif_start_ip_lost_timer esp_netif:0x3ffbf158
D (55643) esp_netif_lwip: if0x3ffbf158 start ip lost tmr: no need start because netif=0x3ffbf1d8 interval=120 ip=0
V (55653) esp_netif_objects: esp_netif_next_unsafe 0x0
V (55653) esp_netif_lwip: esp_netif_is_netif_up esp_netif:0x3ffbf49c
V (55663) esp_netif_objects: esp_netif_next_unsafe 0x3ffbf49c
V (55663) esp_netif_lwip: esp_netif_is_netif_up esp_netif:0x3ffbf158
V (55673) esp_netif_objects: esp_netif_next_unsafe 0x3ffbf158
V (55683) esp_netif_lwip: esp_netif_is_netif_up esp_netif:0x3ffbf49c
D (55683) esp_netif_lwip: call api in lwip: ret=0x0, give sem
D (55693) event: running post WIFI_EVENT:5 with handler 0x400d53b0 on loop 0x3ffbe228
0x400d53b0: wifi_event_handler at /home/reiner/esp/mycomponents/components/wifi_helper.c:460

I (55703) wifi_helper: WIFI_EVENT_STA_DISCONNECTED
I (55703) wifi_helper: Event 'Disconnected' arrived
I (55713) wifi_helper: Disconnected from:
I (55713) wifi_helper:     SSID          : LR_AP
I (55723) wifi_helper:     SSID length   : 5
I (55723) wifi_helper:     BSSID         : 00:00:00:00:00:00
I (55733) wifi_helper:     Reason        : CONNECTION_FAIL
I (55733) wifi_helper: Start wifi_reconnect_timer
I (65743) wifi_helper: wifi_reconnect_timer_func
V (65743) esp_adapter: thread sem get: sem=0x3ffc8da8
I (65863) wifi: new:<6,2>, old:<6,0>, ap:<6,2>, sta:<6,0>, prof:6
I (65863) wifi: state: init -> auth (b0)
I (66863) wifi: state: auth -> init (200)
I (66863) wifi: new:<6,0>, old:<6,2>, ap:<6,2>, sta:<6,0>, prof:6
D (66863) event: running post WIFI_EVENT:5 with handler 0x4011e650 on loop 0x3ffbe228
0x4011e650: wifi_default_action_sta_disconnected at /home/reiner/esp/esp-idf/components/esp_wifi/src/wifi_default.c:103

D (66873) esp_netif_handlers: esp_netif action disconnected with netif0x3ffbf158 from event_id=5
D (66883) esp_netif_lwip: check: remote, if=0x3ffbf158 fn=0x400d7eb8
0x400d7eb8: esp_netif_down_api at /home/reiner/esp/esp-idf/components/esp_netif/lwip/esp_netif_lwip.c:1070

D (66883) esp_netif_lwip: esp_netif_down_api esp_netif:0x3ffbf158
D (66893) esp_netif_lwip: esp_netif_start_ip_lost_timer esp_netif:0x3ffbf158
D (66903) esp_netif_lwip: if0x3ffbf158 start ip lost tmr: no need start because netif=0x3ffbf1d8 interval=120 ip=0
V (66913) esp_netif_objects: esp_netif_next_unsafe 0x0
V (66913) esp_netif_lwip: esp_netif_is_netif_up esp_netif:0x3ffbf49c
V (66923) esp_netif_objects: esp_netif_next_unsafe 0x3ffbf49c
V (66923) esp_netif_lwip: esp_netif_is_netif_up esp_netif:0x3ffbf158
V (66933) esp_netif_objects: esp_netif_next_unsafe 0x3ffbf158
V (66943) esp_netif_lwip: esp_netif_is_netif_up esp_netif:0x3ffbf49c
D (66943) esp_netif_lwip: call api in lwip: ret=0x0, give sem
D (66953) event: running post WIFI_EVENT:5 with handler 0x400d53b0 on loop 0x3ffbe228
0x400d53b0: wifi_event_handler at /home/reiner/esp/mycomponents/components/wifi_helper.c:460

I (66963) wifi_helper: WIFI_EVENT_STA_DISCONNECTED
I (66963) wifi_helper: Event 'Disconnected' arrived
I (66973) wifi_helper: Disconnected from:
I (66973) wifi_helper:     SSID          : LR_AP
I (66983) wifi_helper:     SSID length   : 5
I (66983) wifi_helper:     BSSID         : 30:ae:a4:7b:91:5d
I (66993) wifi_helper:     Reason        : AUTH_EXPIRE
I (67003) wifi_helper: Start wifi_reconnect_timer
I (77003) wifi_helper: wifi_reconnect_timer_func
V (77003) esp_adapter: thread sem get: sem=0x3ffc8da8
D (79453) event: running post WIFI_EVENT:5 with handler 0x4011e650 on loop 0x3ffbe228
0x4011e650: wifi_default_action_sta_disconnected at /home/reiner/esp/esp-idf/components/esp_wifi/src/wifi_default.c:103

[... 65 lines cut ...]
D (103303) esp_netif_lwip: check: remote, if=0x3ffbf158 fn=0x400d7eb8
0x400d7eb8: esp_netif_down_api at /home/reiner/esp/esp-idf/components/esp_netif/lwip/esp_netif_lwip.c:1070

D (103313) esp_netif_lwip: esp_netif_down_api esp_netif:0x3ffbf158
D (103323) esp_netif_lwip: esp_netif_start_ip_lost_timer esp_netif:0x3ffbf158
D (103323) esp_netif_lwip: if0x3ffbf158 start ip lost tmr: no need start because netif=0x3ffbf1d8 interval=120 ip=0
V (103333) esp_netif_objects: esp_netif_next_unsafe 0x0
V (103343) esp_netif_lwip: esp_netif_is_netif_up esp_netif:0x3ffbf49c
V (103343) esp_netif_objects: esp_netif_next_unsafe 0x3ffbf49c
V (103353) esp_netif_lwip: esp_netif_is_netif_up esp_netif:0x3ffbf158
V (103363) esp_netif_objects: esp_netif_next_unsafe 0x3ffbf158
V (103363) esp_netif_lwip: esp_netif_is_netif_up esp_netif:0x3ffbf49c
D (103373) esp_netif_lwip: call api in lwip: ret=0x0, give sem
D (103383) event: running post WIFI_EVENT:5 with handler 0x400d53b0 on loop 0x3ffbe228
0x400d53b0: wifi_event_handler at /home/reiner/esp/mycomponents/components/wifi_helper.c:460

I (103383) wifi_helper: WIFI_EVENT_STA_DISCONNECTED
I (103393) wifi_helper: Event 'Disconnected' arrived
I (103393) wifi_helper: Disconnected from:
I (103403) wifi_helper:     SSID          : LR_AP
I (103413) wifi_helper:     SSID length   : 5
I (103413) wifi_helper:     BSSID         : 00:00:00:00:00:00
I (103423) wifi_helper:     Reason        : CONNECTION_FAIL

`

projectgus, 2019-12-29:
Linking back to the matching email thread: https://esp32.com/viewtopic.php?f=13&t=13639

Reiner1210, 2019-12-30:
I fixed the problem. I must configure the LR mode also for the station of device 1. So the station has bgnlr as mode.

zhangyanjiaoesp, 2019-12-31:
@Reiner1210 
If you set the AP as LR mode, only the STA which supports LR mode can connect to the AP.
So when you set the STA as BGNLR mode, the STA can connect to the AP.

Reiner1210, 2019-12-31:
@zhangyanjiaoesp That's clear. But my problem is a little more complex. What I want is a ssh connection from laptop to WLAN hotspot. But distance is tool long. So the idea was to use 2 ESP32 which atr connected via LR mode.
Have a look here on the diagram [https://esp32.com/viewtopic.php?f=13&t=13639]

So device 1 will run in BGN as Station to connect to WLAN hotspot and run as AP in LR mode.
Device2 will run in LR mode as station (to connect to device 1) and run as AP in BGN mode (so the laptop can connect). All works fine if I do not use LR mode between the devices. 
When I use LR mode device 2 can not connect to device 1. 

I can fif this when setting for the station of device 1 also the LR mode. Now I have a connection between the 2 devices but the AP of device 2 is invisible (so I assume it also runs in LR mode).

Reiner

zhangyanjiaoesp, 2020-01-02:
@Reiner1210 
Thanks for your explanation, I have understood your problem. Yes, your are right, currently we don't support to set softAP and STA in LR mode seperately. So in your usage scenario, the AP of device2 will send the beacon packets in LR data rate, thus the laptop can't find the AP2.

Reiner1210, 2020-01-02:
@zhangyanjiaoesp ok. Will this be changed in the near future ?

closed 2020-01-02 by Reiner1210

zhangyanjiaoesp, 2020-01-02:
@Reiner1210 
We will update the documentation. Thanks for reporting this problem.

chamroeunbm, 2021-02-10:
My LR Mode Range is 60meters only. Any suggestion? I use WROM 32 and ESP32 Cam

zhangyanjiaoesp, 2021-02-18:
@chamroeunbm Is your scenario the same as the original question？The original issue has been closed, please create a new issue to describe your the details of your issue. Thanks!