espressif/esp-idf #4555: kconfig-frontends (menuconfig) rendered badly, barely usable in visual studio code integrated-terminal.

Devenda, 2019-12-29:
## Environment
- IDF version (run ``git describe --tags`` to find it): 
   v4.1-dev-1572-g30372f5a4
- Build System:         [CMake]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
   xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
- Operating System:     [Windows]

## Problem Description

After updating to the latest version of esp-idf the menuconfig renders badly in the vscode integrated terminal.
Arrow keys did not work in previous versions either. As a workaround the j and k keys do work.

This is most likely an issue with the vscode integrated terminal, but maybe a workaround/fix is known.
See microsoft/vscode#87838 for more details.
Issue #4067 is also equal to my issue, however using bash is not feasible.
Issue #1969 also seems to be relevant, however removing/cleaning the build dir does not help.

dobairoland, 2020-01-03:
I don't have this issue with the same ESP-IDF commit and VSCode:
```
Version: 1.41.1
Commit: 26076a4de974ead31f97692a0d32f90d735645c0
Date: 2019-12-20T11:11:01.597Z
Electron: 6.1.7
Chrome: 76.0.3809.146
Node.js: 12.4.0
V8: 7.6.303.31-electron.0
OS: Linux x64 5.4.7-arch1-1
```
and Python 3.8.1 and zsh 5.7.1.

I see in the upstream issue that you use the same VSCode but on Windows.

Please test if this helps in any way: `idf.py menuconfig --style=default`.

Devenda, 2020-01-03:
> I don't have this issue with the same ESP-IDF commit and VSCode:
>
> ```
> Version: 1.41.1
> Commit: 26076a4de974ead31f97692a0d32f90d735645c0
> Date: 2019-12-20T11:11:01.597Z
> Electron: 6.1.7
> Chrome: [...]

It seems that option only changes the colors:
[image]

Anyhow, given that it works on Linux, it must be a Windows issue as you noted in xtermjs/xterm.js#2657

ulfalizer, 2020-01-03:
Could you try this patch and see if it makes a difference? It forces the entire screen to be redrawn each time.

```diff
diff --git a/tools/kconfig_new/menuconfig.py b/tools/kconfig_new/menuconfig.py
index b26e2aa14..9d6a236b4 100644
--- a/tools/kconfig_new/menuconfig.py
+++ b/tools/kconfig_new/menuconfig.py
@@ -1374,7 +1374,7 @@ def _draw_main():
     # Update the symbol window
     #
 
-    _menu_win.erase()
+    _menu_win.clear()
 
     # Draw the _shown nodes starting from index _menu_scroll up to either as
     # many as fit in the window, or to the end of _shown
```

Wouldn't want to do that by default, because it causes annoying flicker. If it helps, it's working around a bug.

Devenda, 2020-01-03:
Thanks, but the patch made no difference sadly...

ulfalizer, 2020-01-03:
@Devenda 
Thanks for trying it out. Pretty sure this is a rendering bug on the xterm.js end at least. Looks like ends of lines get messed up.