espressif/esp-idf #4558: C++ missing libgcc library

paoloach, 2019-12-29:
----------------------------- Delete below -----------------------------

If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at esp32.com instead.

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [ ] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [ ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ ] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs on the monitor](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/idf-monitor.html#automatically-decoding-addresses) or the [coredump](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Provide more items under **Other items if possible** can help us better locate your problem.
5. Use markup (buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:     none
- Module or chip used:  ESP32-WROVER-I
- IDF version :   v4.1-dev-1572-g30372f5a4
- Build System:         CMake
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
- Operating System:    Linux 
- Power Supply:         USB

## Problem Description

With this repository version, my project, containing c++ files, don't more link
I got a list of undefined symbol as _Unwind_GetTextRelBase, _Unwind_GetRegionStart', etc...

The problem disappears if I add the libgcc to the link command:  
xtensa-esp32-elf-g++  -mlongcalls -Wno-frame-address -O3 -DNDEBUG  .... -lc  ** -lgcc ** 

As workaround I added this library to component.cmake so now I don't more manually execute the link command for adding the libgcc library
.....
438     list(REMOVE_ITEM common_reqs ${component_lib})
439    link_libraries(${common_reqs} **gcc**)
440
---
    
### Expected Behavior

It have to link without problem

### Actual Behavior

When use C++ code, I have undefined symbol because the lack of libgcc

### Steps to repropduce

I not be able to use the example to reproduce it using the example in c++. Can be my project contains some c++ features not present in the simple examples

igrr, 2019-12-29:
Can you please give a link to upload the project to reproduce this?

If you do idf.py fullclean and the idf.py build again, does the issue still occur?

paoloach, 2019-12-29:
Great. You are right.
It was enough recreate the build directory.
Thank a lot.

paoloach, 2019-12-29:
The issue doesn't exist

closed 2019-12-29 by paoloach