espressif/esp-idf #4559: How do I use esp32 as an external RAM for other microcontroller?

Mr-Techtron, 2019-12-29:
esp32 is a great micro-controller to use but one of the major downside of esp32 is that it has limited I/O pins. This might sounds quite stupid but I was wondering if esp32 can be used as an external RAM for other micro-controllers? Can someone please put light on this topic, is there any way I could achieve this? Thank you.

0xjakob, 2019-12-30:
Hi @Mr-Techtron. Technically it is an interesting question. There are, however, probably easier solutions. If the pins on the ESP32 aren't enough, you can use one or several GPIO multiplexers they normally connect via I2C (afaik).
For more general questions, it might be more convenient to post in the espressif forum (esp32.com) as github issues/feature requests are normally very specific.

Mr-Techtron, 2019-12-30:
Thank you @0xjakob for your answer. Yes, that is one of the viable solutions but I'm more interested if esp32 can be used as an external RAM for other microcontrollers and I apologize for posting my inconvenient question here.

Alvin1Zhang, 2019-12-30:
@Mr-Techtron Thanks for reporting, for how questions and third party issues, please help submit a post on our forum, thanks.

closed 2019-12-30 by Alvin1Zhang

Mr-Techtron, 2019-12-30:
@Alvin1Zhang Thanks you for your concern. I've posted my query in the esp32 forum and looking forward to validation of my query.