espressif/esp-idf #4560: ESP32添加自己生成的库文件(.a)到工程编译报错

lxdemon, 2019-12-30:
你好：
我有一个项目需要使用自己生成的库文件，但是添加之后编译不过去，报错信息如下：
/home/lixin/workspace/esp32/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/lixin/workspace/esp32/IR_extender_ESP32/components/ir_handle/ir_engine/libuird.a(irLearnLib.o): Relocations in generic ELF (EM: 40)
/home/lixin/workspace/esp32/IR_extender_ESP32/components/ir_handle/ir_engine/libuird.a: error adding symbols: File in wrong format
collect2: error: ld returned 1 exit status

component.mk添加的配置如下：
COMPONENT_ADD_LDFLAGS += $(COMPONENT_PATH)/ir_handle/ir_engine/libuird.a

生成库文件的环境：
OS: Ubuntu18.04
gcc：arm-none-eabi-gcc

望解答，感谢！

igrr, 2019-12-30:
The issue happens because the compiler you use to generate the library (arm-none-eabi-gcc) is for the different CPU architecture than the one used in ESP32. To generate a library compatible with the ESP32, you need to use `xtensa-esp32-elf-gcc` and pass `-mlongcalls` flag to it.

lxdemon, 2019-12-31:
Now it is working, thank you!

closed 2020-01-07 by Alvin1Zhang