espressif/esp-idf #4561: pthread_once: race condition with CONFIG_SPIRAM

szmodz, 2019-12-30:
pthread_once is implemented using uxPortCompareSetExtram

https://github.com/espressif/esp-idf/blob/30372f5a4ff2c0dfdaad95f544dc36bcdda30b75/components/pthread/pthread.c#L499

however, uxPortCompareSetExtram is not safe to use with interrupts enabled:

https://github.com/espressif/esp-idf/blob/30372f5a4ff2c0dfdaad95f544dc36bcdda30b75/components/freertos/port.c#L444

vPortCPUAcquireMutexIntsDisabled assumes interrupts are disabled.

Besides, uxPortCompareSetExtram will never work if extram_mux is itself in extram, so it should probably use vPortCPUAcquireMutexIntsDisabledInternal instead.

uLipe, 2020-01-06:
Hi @szmodz  thanks for reporting that.

The code section itself is safe to use even with interrupts enabled. extram_mux will never placed outside of IRAM (this is a must) ensuring the vPortCPUAcquireMutexIntsDisabled() become a vPortCPUAcquireMutexIntsDisabledInternal() which executes atomically.

Have you experienced some problem with race condition there? Could you please share it?

szmodz, 2020-01-07:
Hi @uLipe 

That's incorrect.

vPortCPUAcquireMutexIntsDisabled is a no-op with CONFIG_FREERTOS_UNICORE, so uxPortCompareSetExtram is completely unprotected in unicore mode (PSRAM is currently only usable in unicore mode). Multiple tasks using pthread_once will race with each other.

https://github.com/espressif/esp-idf/blob/826b9f63805ab9b955cd2129673117080c6d29ba/components/freertos/portmux_impl.h#L93

Races are by definition hard to reproduce. A lot of things can influence the timing. I don't have a minimal test case to share, but the race is there.

The situation doesn't look any better for multicore mode, though. I didn't look at vPortCPUAcquireMutexIntsDisabled all that much, but I know it's a spinlock of the recursive variety, so it can't be safe to use without disabling interrupts (or the scheduler) or some sort of priority inheritance (which is not there).

Consider this: A low priority task (with interrupts enabled) successfully acquires a lock using  vPortCPUAcquireMutexIntsDisabled, and starts running the protected code section. Then it gets preempted by a higher priority task, which tries to acquire the same lock. Two things can happen, depending on the spinlock implementation.

If the spinlock is recursive (can be acquired multiple times by the same core, this seems to be the case here), the situation is the same as in single core mode: the lock will offer no protection against multiple tasks running on the same core.

If, on the other hand, the spinlock is not recursive, the higher priority task will spin in an infinite loop, as the lower priority task will never resume execution, so it will never get a chance to release the lock.

No matter how you look at it, vPortCPUAcquireMutexIntsDisabled is not safe to use with interrupts enabled.

BTW. I also noticed stdatomic.h is incompatible with PSRAM, which can break valid, standard C code. It did cause problems for me.

szmodz, 2020-01-07:
Similarly, vPortCPUAcquireMutex / vPortCPUAcquireMutexTimeout don't make much sense to me.

vPortCPUAcquireMutex should only disable interrupts, while vPortCPUReleaseMutex should re-enable them.

The critical sections protected by a spinlock should always be running with interrupts disabled.

uLipe, 2020-01-07:
Hi, @szmodz you're right, the correct thing here is to disable any kind of preemption until the cmpexchg is executed.

I'll forward a fix for it, thanks for pointing that.

szmodz, 2020-01-07:
@uLipe  the correct thing to do is to disable preemption for the entire critical code path 

In case of uxPortCompareSetExtram that's
```
	prev=*addr;
	if (prev==compare) {
		*addr=*set;
	}
	*set=prev;
```

https://github.com/espressif/esp-idf/blob/30372f5a4ff2c0dfdaad95f544dc36bcdda30b75/components/freertos/port.c#L392

Those should be reviewed as well. They don't make any sense.

uLipe, 2020-01-07:
@szmodz agreed, it does not make sense a compare and exchange code section which its execution is not atomic.

szmodz, 2020-01-07:
Should probably look at this as well:

https://github.com/espressif/esp-idf/blob/826b9f63805ab9b955cd2129673117080c6d29ba/components/freertos/tasks.c#L2779

Even if that's not a bug (interrupts are disabled on entry), it's suboptimal.

uLipe, 2020-01-07:
@szmodz , the fix is currently under internal review and will be available shortly :)

szmodz, 2020-01-07:
@uLipe I hacked a local fix myself, so there's no rush. I understand this specific issue is probably rare in practice. I posted the bug report to make the issue known.

szmodz, 2020-01-07:
In my opinion, spinlocks should never be placed in PSRAM.

The current implementation looks pretty weird ( that's a technical term ;-) ): in order to acquire a spinlock located in PSRAM you first acquire a global spinlock in internal memory. Not only is that inefficient (global lock contention), it also makes my head hurt (mostly because of the global lock contention).

I understand that's because s32c1i (the xtensa compare-exchange instruction) can't be used on PSRAM addresses. However, I think it would be simpler, more efficient, and easier on the eyes to enforce placing spinlocks in internal memory.

My application is memory intensive: it absolutely requires the external RAM (lots of small-sized allocations). The default esp-idf allocation policy (use internal RAM by default, only use PSRAM as fallback) doesn't make much sense in this case, as the internal RAM pool gets exhausted very quickly, and that causes weird and difficult to find problems (see for example https://github.com/espressif/esp-idf/issues/4406, which is NOT currently solved, but I have a local patch for that, so I don't really care personally).

In my application, I opted for a different memory allocation approach: set SPIRAM_MALLOC_ALWAYSINTERNAL to 0 (use PSRAM by default) and place critical data in internal RAM explicitly (esp-idf uses the same approach for processor instructions: everything is placed in flash by default, except for functions explicitly marked with IRAM_ATTR, or specified in linker fragments).

For example, I patched pvPortMalloc to always use internal RAM (replaced malloc with heap_caps_malloc):
https://github.com/espressif/esp-idf/blob/826b9f63805ab9b955cd2129673117080c6d29ba/components/freertos/include/freertos/portable.h#L149

So in my application, the entire internal RAM is reserved for critical data (FreeRTOS structures, thread stacks, DMA buffers, etc), while regular application data (mostly Lua objects, http buffers, xml / json parser data) is placed in PSRAM.

Setting SPIRAM_MALLOC_ALWAYSINTERNAL to 0 exposed a number of issues with the esp-idf. This suggests to me the PSRAM support is simply not production ready. 

I hope that clarifies my issues somewhat.

szmodz, 2020-01-07:
My suggestion would be to remove SPIRAM_MALLOC_ALWAYSINTERNAL and use PSRAM by default, if available. Then make sure everything works in that setup.

closed 2020-01-10 by espressif-bot

szmodz, 2020-01-14:
On closer look, there's another race in pthread_once, event without CONFIG_SPIRAM.

https://linux.die.net/man/3/pthread_once
> On return from pthread_once(), init_routine shall have completed.

The current implementation doesn't guarantee that in case of multiple competing threads. It's possible that pthread_once will return on one thread before init_routine runs to completion on another thread.