espressif/esp-idf #4562: esp_ble_gattc_open failed when scanning

syu-source, 2019-12-30:
Hi all, 

I'm working on esp-idf v4.0-dev-1287-gd7e659df2, with esp32s dev board.
My application requires esp32 to connect to multiple BLE devices. So I need to keep esp32 scanning,  connecting to devices as discovered.

However I have an issue, my app cannot connect to BLE while scanning. 
While scanning,  after esp_ble_gattc_open, the ESP_GATTC_OPEN_EVT comes with open.status == 133(0x85), which is ESP_GATT_ERROR.

Currently I have to do following steps as walkaround:
esp_ble_gap_stop_scanning
esp_ble_gattc_open
if ( got ESP_GATTC_OPEN_EVT )esp_ble_gap_start_scanning

My question is:
Is there anything I missed or is it just normal?

GYC-Espressif, 2019-12-30:
Yes, the recommended action is to stop scanning while connecting and restart the scan after the connection is established.
If you have the print log in ESP_GAP_BLE_SCAN_RESULT_EVT, please try to reduce the log in this event. That might work too.

syu-source, 2019-12-31:
Why is log printing related... 
Thanks anyway.

GYC-Espressif, 2019-12-31:
That's because uart's printing speed is too slow and it may occupy too much time in the Bluetooth Stack. If your issue is resolved, can you close it? Thank you.

Alvin1Zhang, 2020-01-07:
@syu-source Thanks for reporting, would you please help share if any updates for the issue? Has the issue been resolved? Thanks.

syu-source, 2020-01-25:
Stop scanning, connect, restart scanning.
Then everything works. 
I guess scanning reports lots of result and callback may occupy too much time. 
Anyway, issue closed. Thanks.

closed 2020-01-25 by syu-source

Alvin1Zhang, 2020-01-29:
Thanks for sharing the updates.