espressif/esp-idf #4563: Secure boot crash

fpgamaster, 2019-12-30:
## Environment

- Development Kit:      ESP32-PICO-Kit
- Kit version PicoKit
- Module or chip used:  ESP32-PICO-D4
- IDF version (run ``git describe --tags`` to find it): v4.1-dev-1572-g30372f5a4
- Build System:         Make
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 5.2.0
- Operating System:     Linux
- Power Supply:         USB

## Problem Description
The bootloader crashes with secureboot enabled. Without secureboot and the same app (only re-burning the bootloader w/o secureboot) no problems.

--- idf_monitor on /dev/ttyUSB0 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
I (19) boot: ESP-IDF v4.1-dev-1572-g30372f5a4-dirty 2nd stage bootloader
I (20) boot: compile time 12:33:27
D (20) boot: Enabling RTCWDT(9000 ms)
I (23) boot: Enabling RNG early entropy source...
D (29) boot: magic e9
D (31) boot: segments 04
D (34) boot: spi_mode 02
D (36) boot: spi_speed 00
D (39) boot: spi_size 02
I (42) boot: SPI Speed      : 40MHz
I (46) boot: SPI Mode       : DIO
I (50) boot: SPI Flash Size : 4MB
D (54) bootloader_flash: mmu set paddr=00010000 count=1 size=c00 src_addr=10000 src_addr_aligned=10000
D (63) boot: mapped partition table 0x10000 at 0x3f400000
D (69) flash_parts: partition table verified, 7 entries
I (74) boot: Partition Table:
I (78) boot: ## Label            Usage          Type ST Offset   Length
D (85) boot: load partition table entry 0x3f400000
D (90) boot: type=1 subtype=2
I (93) boot:  0 nvs              WiFi data        01 02 00011000 00004000
D (100) boot: load partition table entry 0x3f400020
D (105) boot: type=1 subtype=0
I (108) boot:  1 otadata          OTA data         01 00 00015000 00002000
D (116) boot: load partition table entry 0x3f400040
D (121) boot: type=1 subtype=1
I (124) boot:  2 phy_init         RF data          01 01 00017000 00001000
D (131) boot: load partition table entry 0x3f400060
D (136) boot: type=1 subtype=99
I (140) boot:  3 eeprom           Unknown data     01 99 00018000 00008000
D (147) boot: load partition table entry 0x3f400080
D (152) boot: type=0 subtype=10
I (155) boot:  4 app0             OTA app          00 10 00020000 001f0000
D (163) boot: load partition table entry 0x3f4000a0
D (168) boot: type=0 subtype=11
I (171) boot:  5 app1             OTA app          00 11 00210000 001f0000
I (179) boot: End of partition table
D (183) boot: OTA data offset 0x15000
D (187) bootloader_flash: mmu set paddr=00010000 count=1 size=2000 src_addr=15000 src_addr_aligned=10000
D (196) boot: otadata[0]: sequence values 0x00000001
D (201) boot: otadata[1]: sequence values 0xffffffff
D (206) boot_comm: Only otadata[0] is valid
D (210) boot: Active otadata[0]
D (214) boot: Mapping seq 0 -> OTA slot 0
D (218) boot: Trying partition index 0 offs 0x20000 size 0x1f0000
D (224) esp_image: reading image header @ 0x20000
D (229) bootloader_flash: mmu set block paddr=0x00020000 (was 0xffffffff)
D (236) esp_image: image header: 0xe9 0x07 0x02 0x02 400813f0
I (241) boot_comm: chip revision: 1, min. application chip revision: 0
V (249) esp_image: loading segment header 0 at offset 0x20018
V (254) esp_image: segment data length 0x52380 data starts 0x20020
V (261) esp_image: segment 0 map_segment 1 segment_data_offs 0x20020 load_addr 0x3f400020
I (269) esp_image: segment 0: paddr=0x00020020 vaddr=0x3f400020 size=0x52380 (336768) map
D (278) esp_image: free data page_count 0x00000032
D (283) bootloader_flash: mmu set paddr=00020000 count=6 size=52380 src_addr=20020 src_addr_aligned=20000
[... 107 lines cut ...]
Identity fuses:
MAC                    Factory MAC Address
  = d8:a0:1d:64:fc:88 (CRC 0x52 OK) R/W
CHIP_VER_REV1          Silicon Revision 1                                = 1 R/W (0x1)
CHIP_VER_REV2          Silicon Revision 2                                = 0 R/W (0x0)
CHIP_VERSION           Reserved for future chip versions                 = 0 R/W (0x0)
CHIP_PACKAGE           Chip package identifier                           = 5 R/W (0x5)

Calibration fuses:
BLK3_PART_RESERVE      BLOCK3 partially served for ADC calibration data  = 0 R/W (0x0)
ADC_VREF               Voltage reference calibration                     = 1079 R/W (0x13)

Flash voltage (VDD_SDIO) determined by GPIO12 on reset (High for 1.8V, Low/NC for 3.3V).

Partition table offset 0x10000:
# Name,   Type, SubType, Offset,        Size,           Flags
nvs,      data, nvs,     0x11000,       0x4000,
otadata,  data, ota,     0x15000,       0x2000,
phy_init, data, phy,     0x17000,       0x1000,
eeprom,   data, 0x99,    0x18000,       0x8000,
app0,     app,  ota_0,   0x20000,       0x1f0000,
app1,     app,  ota_1,   0x210000,      0x1f0000,

### Steps to repropduce
Enable secure boot

fpgamaster, 2020-01-02:
With ESP-IDF v3.3.1 all is fine.

I (19) boot: ESP-IDF v3.3.1-dirty 2nd stage bootloader
I (19) boot: compile time 11:42:15
D (19) boot: Enabling RTCWDT(9000 ms)
I (21) boot: Enabling RNG early entropy source...
D (27) boot: magic e9
D (29) boot: segments 04
D (32) boot: spi_mode 02
D (34) boot: spi_speed 00
D (37) boot: spi_size 02
I (40) boot: SPI Speed      : 40MHz
I (44) boot: SPI Mode       : DIO
I (48) boot: SPI Flash Size : 4MB
D (52) bootloader_flash: mmu set paddr=00010000 count=1 size=c00 src_addr=10000 src_addr_aligned=10000
D (61) boot: mapped partition table 0x10000 at 0x3f400000
D (67) flash_parts: partition table verified, 7 entries
I (72) boot: Partition Table:
I (75) boot: ## Label            Usage          Type ST Offset   Length
D (83) boot: load partition table entry 0x3f400000
D (88) boot: type=1 subtype=2
I (91) boot:  0 nvs              WiFi data        01 02 00011000 00004000
D (98) boot: load partition table entry 0x3f400020
D (103) boot: type=1 subtype=0
I (106) boot:  1 otadata          OTA data         01 00 00015000 00002000
D (114) boot: load partition table entry 0x3f400040
D (119) boot: type=1 subtype=1
I (122) boot:  2 phy_init         RF data          01 01 00017000 00001000
D (129) boot: load partition table entry 0x3f400060
D (134) boot: type=1 subtype=99
I (137) boot:  3 eeprom           Unknown data     01 99 00018000 00008000
D (145) boot: load partition table entry 0x3f400080
D (150) boot: type=0 subtype=10
I (153) boot:  4 app0             OTA app          00 10 00020000 001f0000
D (161) boot: load partition table entry 0x3f4000a0
D (166) boot: type=0 subtype=11
I (169) boot:  5 app1             OTA app          00 11 00210000 001f0000
I (176) boot: End of partition table
D (181) boot: OTA data offset 0x15000
D (184) bootloader_flash: mmu set paddr=00010000 count=1 size=2000 src_addr=15000 src_addr_aligned=10000
D (194) boot: otadata[0]: sequence values 0xffffffff
D (199) boot: otadata[1]: sequence values 0xffffffff
D (204) boot: OTA sequence numbers both empty (all-0xFF) or partition table does not have bootable ota_apps (app_count=2)
I (215) boot: No factory image, trying OTA 0
D (220) boot: Trying partition index 0 offs 0x20000 size 0x1f0000
D (226) esp_image: reading image header @ 0x20000
D (231) bootloader_flash: mmu set block paddr=0x00020000 (was 0xffffffff)
D (238) esp_image: image header: 0xe9 0x07 0x02 0x02 40081534
I (244) boot_comm: chip revision: 1, min. application chip revision: 0
V (251) esp_image: loading segment header 0 at offset 0x20018
V (257) esp_image: segment data length 0x4f258 data starts 0x20020
V (263) esp_image: segment 0 map_segment 1 segment_data_offs 0x20020 load_addr 0x3f400020
I (271) esp_image: segment 0: paddr=0x00020020 vaddr=0x3f400020 size=0x4f258 (324184) map
D (280) esp_image: free data page_count 0x00000032
D (285) bootloader_flash: mmu set paddr=00020000 count=5 size=4f258 src_addr=20020 src_addr_aligned=20000
V (408) esp_image: loading segment header 1 at offset 0x6f278
D (409) bootloader_flash: mmu set block paddr=0x00060000 (was 0xffffffff)
V (410) esp_image: segment data length 0xd90 data starts 0x6f280
V (416) esp_image: segment 1 map_segment 0 segment_data_offs 0x6f280 load_addr 0x3ffbdb60
I (424) esp_image: segment 1: paddr=0x0006f280 vaddr=0x3ffbdb60 size=0x00d90 (  3472) load
D (433) esp_image: free data page_count 0x00000032
D (438) bootloader_flash: mmu set paddr=00060000 count=2 size=d90 src_addr=6f280 src_addr_aligned=60000
V (449) esp_image: loading segment header 2 at offset 0x70010
D (453) bootloader_flash: mmu set block paddr=0x00070000 (was 0xffffffff)
V (460) esp_image: segment data length 0xf7dc0 data starts 0x70018
V (466) esp_image: segment 2 map_segment 1 segment_data_offs 0x70018 load_addr 0x400d0018
0x400d0018: _stext at ??:?

I (475) esp_image: segment 2: paddr=0x00070018 vaddr=0x400d0018 size=0xf7dc0 (1015232) map
0x400d0018: _stext at ??:?
[... 83 lines cut ...]
D (1852) secure_boot: write iv+digest to flash
I (1852) secure_boot: Digest generation complete.
I (1852) boot: Checking secure boot...
I (1856) secure_boot: blowing secure boot efuse...
D (1862) secure_boot: before updating, EFUSE_BLK0_RDATA6 4
I (1867) secure_boot: Disable JTAG...
I (1871) secure_boot: Disable ROM BASIC interpreter fallback...
D (1890) secure_boot: after updating, EFUSE_BLK0_RDATA6 54
I (1890) secure_boot: secure boot is now enabled for bootloader image
I (1893) boot: Disabling RNG early entropy source...
D (1899) boot: Mapping segment 0 as DROM
D (1903) boot: Mapping segment 2 as IROM
D (1907) boot: calling set_cache_and_start_app
D (1911) boot: configure drom and irom and start
V (1916) boot: d mmu set paddr=00020000 vaddr=3f400000 size=324184 n=5
V (1922) boot: rc=0
V (1924) boot: rc=0
V (1927) boot: i mmu set paddr=00070000 vaddr=400d0000 size=1015232 n=16
V (1933) boot: rc=0
V (1936) boot: rc=0
D (1938) boot: start: 0x40081534
0x40081534: call_start_cpu0 at /home/krassi/esp/esp-idf/components/esp32/cpu_start.c:123

I (1941) cpu_start: Pro cpu up.
...

projectgus, 2020-03-31:
Hi @fpgamaster,

I'm really sorry noone got back to you about this earlier, this was an unacceptably long time to wait for a response. Your patience is greatly appreciated.

There was a regression in the master branch verison of the bootloader which causes a crash like the one you are seeing. It was fixed in commit 2bf329eb2c258f08871a411bf5a69b004e2c7dd0 for master branch (v4.2-dev) and commit 546cf10c5d63edad104cc48463000b1f92c86db6  for release/v4.1 branch (v4.1-dev) - which was included in the [v4.1-beta1 version](https://github.com/espressif/esp-idf/releases/tag/v4.1-beta1) and will be included in the v4.1 final release.

If you check out the latest prerelease code for either v4.1 or v4.2 (master), it should work again. Please let us know if you're still seeing the issue.

Angus

projectgus, 2020-04-30:
Closing as this is believed to be fixed.

closed 2020-04-30 by projectgus