espressif/esp-idf #4564: select call on stdin not working as expected

ajaybhargav, 2019-12-30:
## Environment

- Development Kit:      ESP32-DevKitC
- Kit version (for WroverKit/PicoKit/DevKitC): v4
- Module or chip used:  ESP32-WROOM-32U
- IDF version (run ``git describe --tags`` to find it): 
- Build System:  PlatformIO, Framework (espidf)
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
- Operating System:     Windows
- Power Supply:         USB

## Problem Description

I am trying to read STDIN using select function but I am getting uart driver error instead and return 0 (timeout). But never read anything. Any help would be appreciated.

### Expected Behavior

Select call should return >0 if bytes present
0 for timeout
-1 if error

### Actual Behavior

Select ret: 0
E (243542) uart: uart_get_buffered_data_len(1253): uart driver error

### Steps to repropduce

### Code to reproduce this issue

Following is code I am running.
```

		timeout.tv_sec = 1;
		timeout.tv_usec = 0;

		FD_ZERO(&set); /* clear the set */
		FD_SET(STDIN_FILENO, &set); /* add our file descriptor to the set */
		ret = select(STDIN_FILENO + 1, &set, NULL, NULL, &timeout);
		debug(DBG_OFF, "Select ret: %d\n", ret);
		if (ret > 0) {
			ret = 0;
			read(STDIN_FILENO, &ret, 1);
			break;
		}

```
## Debug Logs

```
Select ret: 0
E (328242) uart: uart_get_buffered_data_len(1253): uart driver error
Select ret: 0
E (329342) uart: uart_get_buffered_data_len(1253): uart driver error
Select ret: 0
E (330442) uart: uart_get_buffered_data_len(1253): uart driver error
Select ret: 0
E (331542) uart: uart_get_buffered_data_len(1253): uart driver error
Select ret: 0
E (332642) uart: uart_get_buffered_data_len(1253): uart driver error
```

## Other items if possible

dobairoland, 2019-12-30:
Hello @ajaybhargav.

Please take a look how it is done in the example: https://github.com/espressif/esp-idf/tree/master/examples/peripherals/uart/uart_select

Did you called `esp_vfs_dev_uart_use_driver()` like it is done here: https://github.com/espressif/esp-idf/blob/master/examples/peripherals/uart/uart_select/main/uart_select_example_main.c#L46

Another example which could be helpful: https://github.com/espressif/esp-idf/tree/master/examples/system/select

ajaybhargav, 2019-12-31:
I have not called esp_vfs_dev_uart_use_driver() I believe driver must be already setup since standard IO should be setup by system itself. I may be wrong, but I will still try with calling this function and update.

ajaybhargav, 2019-12-31:
I think a proper stdin/out initialization missing in documentation. If anyone else stumble upon the same issue then please init your standard IO as follows:

```
setvbuf(stdin, NULL, _IONBF, 0);
setvbuf(stdout, NULL, _IONBF, 0);
ESP_ERROR_CHECK(uart_driver_install((uart_port_t)CONFIG_CONSOLE_UART_NUM,
						256, 0, 0, NULL, 0));
esp_vfs_dev_uart_use_driver(CONFIG_CONSOLE_UART_NUM);
esp_vfs_dev_uart_set_rx_line_endings(ESP_LINE_ENDINGS_CR);
/* Move the caret to the beginning of the next line on '\n' */
esp_vfs_dev_uart_set_tx_line_endings(ESP_LINE_ENDINGS_CRLF);
```
I thought the above would be done automatically since printf was working fine. Anyways, Problem is now resolved.

closed 2019-12-31 by ajaybhargav