espressif/esp-idf #4565: error in running the basic ethernet example

manish5897, 2019-12-31:
<h3>Environment</h3>

    Module or chip used: [ESP32-WROOM-32]
    IDF version : v4.0-beta2-303-gb0d6c01b0
    Build System: [Make]
    Compiler version : 8.2.0
    Operating System: [Linux]
    Power Supply: [USB]

<h3>Problem Description</h3>
E (556) emac_esp32: emac_esp32_init(308): reset timeout

E (556) esp_eth: esp_eth_driver_install(191): init mac failed

<h3>Code to reproduce this issue</h3>
examples: basic ethernet

<h3>Debug Logs</h3>

I (354) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
E (556) emac_esp32: emac_esp32_init(308): reset timeout
E (556) esp_eth: esp_eth_driver_install(191): init mac failed
ESP_ERROR_CHECK failed: esp_err_t 0xffffffff (ESP_FAIL) at 0x40086038
0x40086038: _esp_error_check_failed at /home/manish/esp/esp-idf/components/esp32/panic.c:726

file: "/home/manish/esp/esp-idf/examples/ethernet/basic/main/ethernet_example_main.c" line 120
func: app_main
expression: esp_eth_driver_install(&config, &eth_handle)

ELF file SHA256: 06cccc389bee449a4eefc83d013d19c35c6931f2a866a4f40a4434494940fc3e

Backtrace: 0x40085aa5:0x3ffb5f40 0x4008603b:0x3ffb5f60 0x400d3903:0x3ffb5f80 0x400d1c7a:0x3ffb5fe0 0x40087f81:0x3ffb6000
0x40085aa5: invoke_abort at /home/manish/esp/esp-idf/components/esp32/panic.c:155

0x4008603b: _esp_error_check_failed at /home/manish/esp/esp-idf/components/esp32/panic.c:727

0x400d3903: app_main at /home/manish/esp/esp-idf/examples/ethernet/basic/main/ethernet_example_main.c:120 (discriminator 1)

0x400d1c7a: main_task at /home/manish/esp/esp-idf/components/esp32/cpu_start.c:554

0x40087f81: vPortTaskWrapper at /home/manish/esp/esp-idf/components/freertos/port.c:143

suda-morris, 2019-12-31:
The default config is enough if you're running on the official Ethernet board from Espressif.
If not, you need to tell how you configure your Ethernet board (attach your sdkconfig can be better).
In your case, ESP32 doesn't get a valid RMII clock (50MHz, usually sourced from outside on GPIO0)

manish5897, 2019-12-31:
I'm using using olimex ESP32-PoE-ISO board. 

SDKconfig:

#
# Automatically generated file. DO NOT EDIT.
# Espressif IoT Development Framework (ESP-IDF) Project Configuration
#
CONFIG_IDF_TARGET_ESP32=y
CONFIG_IDF_TARGET="esp32"
CONFIG_IDF_FIRMWARE_CHIP_ID=0x0000

#
# SDK tool configuration
#
CONFIG_SDK_TOOLPREFIX="xtensa-esp32-elf-"
CONFIG_SDK_PYTHON="python"
CONFIG_SDK_MAKE_WARN_UNDEFINED_VARIABLES=y
CONFIG_APP_COMPILE_TIME_DATE=y
# CONFIG_APP_EXCLUDE_PROJECT_VER_VAR is not set
# CONFIG_APP_EXCLUDE_PROJECT_NAME_VAR is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_NONE is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_ERROR is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_WARN is not set
CONFIG_BOOTLOADER_LOG_LEVEL_INFO=y
# CONFIG_BOOTLOADER_LOG_LEVEL_DEBUG is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_VERBOSE is not set
CONFIG_BOOTLOADER_LOG_LEVEL=3
# CONFIG_BOOTLOADER_VDDSDIO_BOOST_1_8V is not set
CONFIG_BOOTLOADER_VDDSDIO_BOOST_1_9V=y
# CONFIG_BOOTLOADER_FACTORY_RESET is not set
# CONFIG_BOOTLOADER_APP_TEST is not set
CONFIG_BOOTLOADER_WDT_ENABLE=y
# CONFIG_BOOTLOADER_WDT_DISABLE_IN_USER_CODE is not set
CONFIG_BOOTLOADER_WDT_TIME_MS=9000
# CONFIG_BOOTLOADER_APP_ROLLBACK_ENABLE is not set
# CONFIG_SECURE_SIGNED_APPS_NO_SECURE_BOOT is not set
# CONFIG_SECURE_BOOT_ENABLED is not set
# CONFIG_SECURE_FLASH_ENC_ENABLED is not set
CONFIG_ESPTOOLPY_PORT="/dev/ttyUSB0"
CONFIG_ESPTOOLPY_BAUD_115200B=y
# CONFIG_ESPTOOLPY_BAUD_230400B is not set
# CONFIG_ESPTOOLPY_BAUD_921600B is not set
# CONFIG_ESPTOOLPY_BAUD_2MB is not set
# CONFIG_ESPTOOLPY_BAUD_OTHER is not set
CONFIG_ESPTOOLPY_BAUD_OTHER_VAL=115200
CONFIG_ESPTOOLPY_BAUD=115200
CONFIG_ESPTOOLPY_COMPRESSED=y
# CONFIG_ESPTOOLPY_FLASHMODE_QIO is not set
# CONFIG_ESPTOOLPY_FLASHMODE_QOUT is not set
CONFIG_ESPTOOLPY_FLASHMODE_DIO=y
# CONFIG_ESPTOOLPY_FLASHMODE_DOUT is not set
CONFIG_ESPTOOLPY_FLASHMODE="dio"
# CONFIG_ESPTOOLPY_FLASHFREQ_80M is not set
CONFIG_ESPTOOLPY_FLASHFREQ_40M=y
# CONFIG_ESPTOOLPY_FLASHFREQ_26M is not set
# CONFIG_ESPTOOLPY_FLASHFREQ_20M is not set
CONFIG_ESPTOOLPY_FLASHFREQ="40m"
# CONFIG_ESPTOOLPY_FLASHSIZE_1MB is not set
CONFIG_ESPTOOLPY_FLASHSIZE_2MB=y
# CONFIG_ESPTOOLPY_FLASHSIZE_4MB is not set
# CONFIG_ESPTOOLPY_FLASHSIZE_8MB is not set
# CONFIG_ESPTOOLPY_FLASHSIZE_16MB is not set
CONFIG_ESPTOOLPY_FLASHSIZE="2MB"
CONFIG_ESPTOOLPY_FLASHSIZE_DETECT=y
CONFIG_ESPTOOLPY_BEFORE_RESET=y
# CONFIG_ESPTOOLPY_BEFORE_NORESET is not set
CONFIG_ESPTOOLPY_BEFORE="default_reset"
CONFIG_ESPTOOLPY_AFTER_RESET=y
# CONFIG_ESPTOOLPY_AFTER_NORESET is not set
[... 600 lines cut ...]
CONFIG_TCP_OVERSIZE_MSS=y
# CONFIG_TCP_OVERSIZE_QUARTER_MSS is not set
# CONFIG_TCP_OVERSIZE_DISABLE is not set
CONFIG_UDP_RECVMBOX_SIZE=6
CONFIG_TCPIP_TASK_STACK_SIZE=3072
CONFIG_TCPIP_TASK_AFFINITY_NO_AFFINITY=y
# CONFIG_TCPIP_TASK_AFFINITY_CPU0 is not set
# CONFIG_TCPIP_TASK_AFFINITY_CPU1 is not set
CONFIG_TCPIP_TASK_AFFINITY=0x7FFFFFFF
# CONFIG_PPP_SUPPORT is not set
CONFIG_ESP32_PTHREAD_TASK_PRIO_DEFAULT=5
CONFIG_ESP32_PTHREAD_TASK_STACK_SIZE_DEFAULT=3072
CONFIG_ESP32_PTHREAD_STACK_MIN=768
CONFIG_ESP32_DEFAULT_PTHREAD_CORE_NO_AFFINITY=y
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_0 is not set
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_1 is not set
CONFIG_ESP32_PTHREAD_TASK_CORE_DEFAULT=-1
CONFIG_ESP32_PTHREAD_TASK_NAME_DEFAULT="pthread"
CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ABORTS=y
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_FAILS is not set
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ALLOWED is not set
CONFIG_IP_LOST_TIMER_INTERVAL=120
CONFIG_SUPPRESS_SELECT_DEBUG_OUTPUT=y
CONFIG_SUPPORT_TERMIOS=y
# End of deprecated options

manish5897, 2019-12-31:
I'm using Olimex ESP32-PoE-ISO board.
please find the sdkconfig:

[sdkconfig.pdf](https://github.com/espressif/esp-idf/files/4012096/sdkconfig.pdf)

suda-morris, 2019-12-31:
OK, now things get easier to debug. 

In your case (Olimex ESP32-PoE-ISO), the RMII clock is generated from ESP32 (on GPIO17).
What you configured in the Example Configuration is right, but you also need to configure a little bit in `components->Ethernet`. See more docs [here](https://github.com/espressif/esp-idf/tree/master/examples/ethernet)

Try these two options
```
CONFIG_ETH_RMII_CLK_OUTPUT=y
CONFIG_ETH_RMII_CLK_OUT_GPIO=17
```

manish5897, 2019-12-31:
<h3>Error Logs</h3>

firstly I get an error like this:

E (371) lan8720: lan8720_pwrctl(332): power up failed
E (381) lan8720: lan8720_init(368): power control failed
E (381) esp_eth: esp_eth_driver_install(192): init phy failed
ESP_ERROR_CHECK failed: esp_err_t 0xffffffff (ESP_FAIL) at 0x400861a8
0x400861a8: _esp_error_check_failed at /home/manish/esp/esp-idf/components/esp32/panic.c:726

file: "/home/manish/Desktop/ethernet_example/main/ethernet_example_main.c" line 89
func: app_main
expression: esp_eth_driver_install(&config, &eth_handle)

ELF file SHA256: 7b6e8216d89c66906ac4ff60a6f35e87c857a57d9085f9c5b66d8872e513c210

Backtrace: 0x40085c15:0x3ffb5f40 0x400861ab:0x3ffb5f60 0x400d3908:0x3ffb5f80 0x400d1c82:0x3ffb5fe0 0x400880f1:0x3ffb6000
0x40085c15: invoke_abort at /home/manish/esp/esp-idf/components/esp32/panic.c:155

0x400861ab: _esp_error_check_failed at /home/manish/esp/esp-idf/components/esp32/panic.c:727

0x400d3908: app_main at /home/manish/Desktop/ethernet_example/main/ethernet_example_main.c:89 (discriminator 1)

0x400d1c82: main_task at /home/manish/esp/esp-idf/components/esp32/cpu_start.c:554

0x400880f1: vPortTaskWrapper at /home/manish/esp/esp-idf/components/freertos/port.c:143

<h3>Error Logs</h3>

Second time system restarted and got the IP address:

I (362) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (392) eth_example: Ethernet Started
I (4392) eth_example: Ethernet Link Up
I (4392) eth_example: Ethernet HW Addr 24:0a:c4:25:e3:4f
I (5362) tcpip_adapter: eth ip: 192.168.1.144, mask: 255.255.255.0, gw: 192.168.1.1
I (5362) eth_example: Ethernet Got IP Address
I (5362) eth_example: ~~~~~~~~~~~
I (5362) eth_example: ETHIP:192.168.1.144
I (5372) eth_example: ETHMASK:255.255.255.0
I (5372) eth_example: ETHGW:192.168.1.1
I (5382) eth_example: ~~~~~~~~~~~

szmodz, 2020-01-03:
I was seeing something similar. I solved it by inserting a delay (500ms) before initialising ethernet. Slow oscillator startup? In my case the clock is input on GPIO0 using the LAN8720A built-in oscillator as source (REFCLKO).

szmodz, 2020-01-03:
Seems to work without the additional delay after increasing the reset timeout in my case. Perhaps the default timeouts should be increased?

suda-morris, 2020-01-07:
Congratulations, regarding to "increase reset timeout", we could mention it in the doc, since the default reset time varies between boards, we can't set a large value to cover all situations.

closed 2020-03-16 by suda-morris

Maldus512, 2020-10-11:
I apologize for resuming an old issue, but I have the exact same problem. I'm running the default eth example modified for the Olimex-ESP32-POE demo board (https://github.com/OLIMEX/ESP32-POE/tree/master/SOFTWARE/ESP-IDF/ESP32_PoE_Ethernet_IDFv4.0); it works, but only the second time I attempt to initialize the ethernet interface. 

The first time, no matter what delay I use, the initialization procedure fails with the following error:

```
I (4448) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
E (4448) lan8720: lan8720_pwrctl(333): power up failed
E (4448) lan8720: lan8720_init(369): power control failed
E (4458) esp_eth: esp_eth_driver_install(197): init phy failed
```

The example then panics, resets and tries again, this time successfully. Even without resetting, simply trying again will succeed.

I tried using the `sw_reset_timeout_ms` and `reset_timeout_ms` fields of the mac and phy config, but to no avail. Of course, trying again on failure is a very acceptable workaround. Regardless, any idea of what the problem might be?

rony91jp, 2020-11-24:
I apologize for re-open it. 
I am now using ESP32-POE board to create an opc ua server.  i was able to connect the OPC UA server through wifi but i got error when i tried to connect via LAN.  Error content is the same as this issue and i tried to solve it following the instruction above but still the error are same. Please check the below log file.
Using Python in C:\Python38\
Python 3.8.2
Using Git in C:\Program Files\Git\cmd\
git version 2.21.0.windows.1
Setting IDF_PATH: C:\Users\saha_ronykumer\esp-idf

Adding ESP-IDF tools to PATH...
    C:\Users\saha_ronykumer\.espressif\tools\xtensa-esp32s2-elf\esp-2020r3-8.4.0\xtensa-esp32s2-elf\bin
    C:\Users\saha_ronykumer\.espressif\tools\xtensa-esp32s3-elf\esp-2020r3-8.4.0\xtensa-esp32s3-elf\bin
    C:\Users\saha_ronykumer\.espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin
    C:\Users\saha_ronykumer\.espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin
    C:\Users\saha_ronykumer\.espressif\tools\idf-exe\1.0.1\
    C:\Users\saha_ronykumer\.espressif\tools\ccache\3.7\
    C:\Users\saha_ronykumer\.espressif\tools\dfu-util\0.9\dfu-util-0.9-win64
    C:\Users\saha_ronykumer\.espressif\python_env\idf4.3_py3.8_env\Scripts

Checking if Python packages are up to date...
Python requirements from C:\Users\saha_ronykumer\esp-idf\requirements.txt are satisfied.

Done! You can now compile ESP-IDF projects.
Go to the project directory and run:

  idf.py build

C:\Users\saha_ronykumer\esp-idf>cd examples

C:\Users\saha_ronykumer\esp-idf\examples>cd opcua^esp32
指定されたパスが見つかりません。

C:\Users\saha_ronykumer\esp-idf\examples>cd opcua-esp32

C:\Users\saha_ronykumer\esp-idf\examples\opcua-esp32>idf.py build
Executing action: all (aliases: build)
Running ninja in directory c:\users\saha_ronykumer\esp-idf\examples\opcua-esp32\build
Executing "ninja all"...
[1/10] Performing build step for 'bootloader'
ninja: no work to do.
[8/8] Generating binary image from built executable
esptool.py v3.0-dev
Generated C:/Users/saha_ronykumer/esp-idf/examples/opcua-esp32/build/opcua_esp32.bin

Project build complete. To flash, run this command:
C:\Users\saha_ronykumer\.espressif\python_env\idf4.3_py3.8_env\Scripts\python.exe ..\..\components\esptool_py\esptool\esptool.py -p (PORT) -b 460800 --before default_reset --after hard_reset --chip esp32  write_flash --flash_mode dio --flash_size detect --flash_freq 40m 0x1000 build\bootloader\bootloader.bin 0x8000 build\partition_table\partition-table.bin 0x10000 build\opcua_esp32.bin
or run 'idf.py -p (PORT) flash'

C:\Users\saha_ronykumer\esp-idf\examples\opcua-esp32>idf.py build
Executing action: all (aliases: build)
Running ninja in directory c:\users\saha_ronykumer\esp-idf\examples\opcua-esp32\build
Executing "ninja all"...
[1/10] Performing build step for 'bootloader'
ninja: no work to do.
[8/8] Generating binary image from built executable
esptool.py v3.0-dev
Generated C:/Users/saha_ronykumer/esp-idf/examples/opcua-esp32/build/opcua_esp32.bin

Project build complete. To flash, run this command:
C:\Users\saha_ronykumer\.espressif\python_env\idf4.3_py3.8_env\Scripts\python.exe ..\..\components\esptool_py\esptool\esptool.py -p (PORT) -b 460800 --before default_reset --after hard_reset --chip esp32  write_flash --flash_mode dio --flash_size detect --flash_freq 40m 0x1000 build\bootloader\bootloader.bin 0x8000 build\partition_table\partition-table.bin 0x10000 build\opcua_esp32.bin
or run 'idf.py -p (PORT) flash'

C:\Users\saha_ronykumer\esp-idf\examples\opcua-esp32>idf.py -p COM4 flash
Executing action: flash
Running ninja in directory c:\users\saha_ronykumer\esp-idf\examples\opcua-esp32\build
Executing "ninja flash"...
[1/4] Performing build step for 'bootloader'
ninja: no work to do.
[1/2] cmd.exe /C "cd /D C:\Users\saha_ronykumer\esp-idf\components\esptool_py && C...build" -P C:/Users/saha_ronykumer/esp-idf/components/esptool_py/run_esptool.cmake"
esptool.py --chip esp32 -p COM4 -b 460800 --before=default_reset --after=hard_reset write_flash --flash_mode dio --flash_freq 40m --flash_size 2MB 0x8000 partition_table/partition-table.bin 0x1000 bootloader/bootloader.bin 0x10000 opcua_esp32.bin
esptool.py v3.0-dev
[... 159 lines cut ...]
0x4008a62d: vPortTaskWrapper at C:/Users/saha_ronykumer/esp-idf/components/freertos/xtensa/port.c:170

ELF file SHA256: ff0d2b91bdef90c7

Entering gdb stub now.
$T0b#e6GNU gdb (crosstool-NG esp-2020r3) 8.1.0.20180627-git
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "--host=x86_64-host_w64-mingw32 --target=xtensa-esp32-elf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from c:\users\saha_ronykumer\esp-idf\examples\opcua-esp32\build\opcua_esp32.elf...done.
Remote debugging using \\.\COM4
warning: unrecognized item "T0b" in "qSupported" response
0x4008772a in panic_abort (details=0x3ffbc6fb "abort() was called at PC 0x40086687 on core 0") at C:/Users/saha_ronykumer/esp-idf/components/esp_system/panic.c:360
360         *((int *) 0) = 0; // NOLINT(clang-analyzer-core.NullDereference) should be an invalid operation on targets
(gdb)

suda-morris, 2020-11-25:
Ethernet initialization code has a great relationship with your hardware design. The default configuration is only guarantee on the official Ethernet board. You might have to tuned the configuration based on your board. More information could be seen here: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_eth.html#configure-mac-and-phy

rony91jp, 2020-11-26:
Hello,

Check the log file below and i am getting nothing after that. Where to change? 
I (29) boot: ESP-IDF v4.3-dev-1720-g494a124d9-dirty 2nd stage bootloader
I (29) boot: compile time 11:31:41
I (30) boot: chip revision: 1
I (34) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (41) boot.esp32: SPI Speed      : 40MHz
I (45) boot.esp32: SPI Mode       : DIO
I (50) boot.esp32: SPI Flash Size : 2MB
I (54) boot: Enabling RNG early entropy source...
I (60) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (78) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (86) boot:  2 factory          factory app      00 00 00010000 00100000
I (93) boot: End of partition table
I (97) boot_comm: chip revision: 1, min. application chip revision: 0
I (104) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x0e4c8 ( 58568) map
I (136) esp_image: segment 1: paddr=0x0001e4f0 vaddr=0x3ffb0000 size=0x01b28 (  6952) load
I (139) esp_image: segment 2: paddr=0x00020020 vaddr=0x400d0020 size=0x2fd34 (195892) map
0x400d0020: _stext at ??:?

I (217) esp_image: segment 3: paddr=0x0004fd5c vaddr=0x3ffb1b28 size=0x00e60 (  3680) load
I (219) esp_image: segment 4: paddr=0x00050bc4 vaddr=0x40080000 size=0x00404 (  1028) load
0x40080000: _WindowOverflow4 at C:/Users/saha_ronykumer/esp-idf/components/freertos/xtensa/xtensa_vectors.S:1730

I (224) esp_image: segment 5: paddr=0x00050fd0 vaddr=0x40080404 size=0x0b4d4 ( 46292) load
I (259) boot: Loaded app from partition at offset 0x10000
I (260) boot: Disabling RNG early entropy source...
I (271) cpu_start: Pro cpu up.
I (271) cpu_start: Starting app cpu, entry point is 0x40081104
0x40081104: call_start_cpu1 at C:/Users/saha_ronykumer/esp-idf/components/esp_system/port/cpu_start.c:124

I (0) cpu_start: App cpu up.
I (285) cpu_start: Pro cpu start user code
I (285) cpu_start: cpu freq: 160000000
I (286) cpu_start: Application information:
I (290) cpu_start: Project name:     ethernet_basic
I (296) cpu_start: App version:      v4.3-dev-1720-g494a124d9-dirty
I (302) cpu_start: Compile time:     Nov 26 2020 11:31:16
I (309) cpu_start: ELF file SHA256:  845361f89b6dece1...
I (315) cpu_start: ESP-IDF:          v4.3-dev-1720-g494a124d9-dirty
I (322) heap_init: Initializing. RAM available for dynamic allocation:
I (329) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (335) heap_init: At 3FFB4150 len 0002BEB0 (175 KiB): DRAM
I (341) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (347) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (354) heap_init: At 4008B8D8 len 00014728 (81 KiB): IRAM
I (361) spi_flash: detected chip: generic
I (365) spi_flash: flash io: dio
W (369) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (383) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (403) system_api: Base MAC address is not set
I (403) system_api: read default base MAC address from EFUSE
I (913) esp_eth.netif.glue: 98:f4:ab:79:ae:fb
I (913) esp_eth.netif.glue: ethernet attached to netif
I (923) eth_example: Ethernet Started
I (4923) eth_example: Ethernet Link Up
I (4923) eth_example: Ethernet HW Addr 98:f4:ab:79:ae:fb

suda-morris, 2020-11-26:
What's GPIO you used for RMII clock, it looks to me is till the RMII clock issue, the clock phase might have a large difference between MAC and PHY.

rony91jp, 2020-11-26:
@suda-morris 

RMII clock GPIO at 17. Please check the configuration capture.
[image]

suda-morris, 2020-11-26:
Can you try GPIO16 instead to see if it works for you?

rony91jp, 2020-11-26:
I tried GPIO16 but here is the following error. 
[image]

Please check the Below Example configuration and I am using ESP32-POE board.
[image]

suda-morris, 2020-11-26:
OK, I though you were using ESP32 and connecting to  LAN8720 break out board by jump wires. In your case, it seems like GPIO17 is the fixed IO that can be used as RMII clock. 
If  you have also set the PHY address correct, then I have no other clue, maybe you should go to the forum to see if some one using the same board has met the same issue.

rony-91, 2020-11-28:
Hello@suda-morris,

Regarding the above issue, I was able to get the static IP address using the Arduino ethernet example.  I used the below command to assign my static IP.
ETH.begin();

ETH.config(IPAddress(192, 168, 1, 101),IPAddress(192, 168, 1, 1),IPAddress(255, 255, 255, 0),IPAddress(192, 168, 1, 1), IPAddress(192, 168, 1, 1));

Output:
[image]

My question is, I want to disable the DHCP and want to assign the static IP above the same way in the esp-idf example. Where should I change in the esp-idf ethernet basic example program? 

Than you

francisconavy, 2021-01-17:
Hello, I'm trying to use ethernet basic example of esp-idf on esp32 and LAN8720.
The errors are these:
E (514) lan8720: lan8720_pwrctl(359): power up failed
E (514) lan8720: lan8720_init(421): power control failed
E (514) esp_eth: esp_eth_driver_install(205): init phy failed

I already saw Frank Sautter @sauttefk [tutorial](https://sautter.com/blog/ethernet-on-esp32-using-lan8720/), tryied to increase sw reset timeout to 500ms and my local esp-idf repository is up to date.

My conections are exactly like this:
(the only difference is that I'm using two parallel 10k resistors as pullup and pulldown)
[image]

The oscilator enable is connected to NC:
[image]

 And my configs are these in the moment:
[image]
[image]

I think did everything I could find on internet to solve this, but I must be missing out some point. Does anyone have a sugestion? I really want to make this work.